robotology/icub-main #628: getRefPosition/s of IPositionDirect doesn't work on iCub_SIM

Nicogene, 2019-11-26:
Here how to reproduce it:

```bash
$ iCub_SIM
$ yarp rpc /icubSim/left_arm/rpc:i
$>>get dpos
Response: [is] dpos (0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0) [tsta] 0 1574779368.71414 [ok]
$>>get poss
Response: [is] poss (-24.9999892856126 19.9999914284901 0.0 49.9999785712252 0.0 0.0 0.0 58.9999747140457 19.9999914284901 19.9999914284901 19.9999914284901 9.99999571424504 9.99999571424504 9.99999571424504 9.99999571424504 9.99999571424504) [tsta] 2 1574779383.96386 [ok]
``` 

No matter in which control mode the joints are set. And same thing for the "single-joint" requests, it seems that the `getRefPosition/s` are broken for the `IPositionDirect` interface.

cc @vtikha @pattacini @randaz81

Nicogene, 2019-11-26:
Checking quickly the `iCub_SIM` implementation of `IPositionDirect::getRefPosition/s`, the actual joint position is updated **after** that the first Position direct command is executed.

This explain why it return all zeros if the robot it never being moved in position direct.

The question now is, it is a bug, or an intended implementation?

On the real robot it gets the info from the encoders right?

pattacini, 2019-11-26:
I'm not into it; investigation required :shipit:

randaz81, 2019-11-28:
Uhm, if the robot never received a reference, why getReference() should contain a value? which value by the way? Maybe zero is not an appropriate answer, not-a-number is more suitable?

Nicogene, 2019-11-28:
Chatting f2f with @randaz81 , probably the implementation is correct, I am asking a reference frame and if no reference frame is set, IPositionDirect doesn't have any information.

Maybe it could return nan(I personally don't like it very much) or the request should fail returning false.

BTW probably the best way to know the state of the joints of the robot is using the `iEncoders` interface.

barbalberto, 2019-11-28:
My two cents: whatever you choose (function returning false, values to nan, both) I suggest to to the same on real robot and gazebo, for coherence.

Which leads to the question:
What happens on the real robot if you try the same test?

Nicogene, 2019-11-28:
> What happens on the real robot if you try the same test?

Good question indeed :thinking:

traversaro, 2019-11-28:
Just to understand, if `getRefPosition/s` only the reference positions explicitly set by `setRefPosition/s` what is the proper way to get the reference positions that the low-level position PIDs uses when it is switched to the POSITION_DIRECT mode, before the first `setRefPosition/s` is send?
For a lot of use cases, it is quite important to be able to initialize positions filters with the exact initial reference, to avoid discontinuities in the reference signal. 
For example, when using KUKA's industrial robot RSI interface, sending as new reference the measured position in some configurations may result in a controller emergency stop, as the difference between reference position and actual position due to the static error is big enough to create a large acceleration that exceeds the safety limits. (feel free to ask @AlbertoRemusIIT and @AngeloRendiniello more details on this : ) ).

randaz81, 2019-12-05:
> Just to understand, if getRefPosition/s only the reference positions explicitly set by setRefPosition/s what is the proper way to get the reference positions that the low-level position PIDs uses when it is switched to the POSITION_DIRECT mode, before the first setRefPosition/s is send?

You need to use this one:

```yarp::dev::IPidControl::getPidReference (const PidControlTypeEnum &pidtype, int j, double *ref)=0```

Indeed I refactored this API some time ago, to properly distinguish the PID internal reference from user high level reference (trajectory or position direct).

traversaro, 2019-12-05:
cc @GiulioRomualdi @S-Dafarra

S-Dafarra, 2019-12-05:
Maybe is off-topic. Does this mean that https://github.com/robotology/icub-main/issues/475 was solved?

randaz81, 2019-12-10:
> does this mean that #475 was solved?

I tried to read the discussion in #475, but I fear that most of the information there are outdated.
I think that if you are interested in performance of these methods, the best way is to _open a new issue_, add a reference to the old one and restart the discussion from scratch. In this respect, a new experimental measurement of the delays is needed.

> If not, I was wondering whether it could make sense for getTargetPosition/getRefPosition to call the corresponding IPidControl::getPidReference to fix both performance issues and the problem arose in this issue.

No, as mentioned above, these methods have different meanings, they do not share a common implementation and they are used to obtain data from different sources. So they cannot be used for this purpose.

closed 2019-12-10 by randaz81