robotology/icub-main #631: SkinPrototype Error when compiling icub-main v1.14.0

dcam0050, 2019-12-18:
Tried compiling icub-main v1.14.0 and received the following error during compilation:

```
Scanning dependencies of target skinprototype
[ 17%] Built target skinWrapper
[ 17%] Building CXX object src/libraries/icubmod/skinprototype/CMakeFiles/skinprototype.dir/skinprototype.cpp.o
[ 17%] Built target canBusAnalogSensor
[ 17%] Building CXX object src/libraries/icubmod/skinprototype/CMakeFiles/skinprototype.dir/yarp_plugin_icubmod_skinprototype.cpp.o
In file included from /usr/local/src/robot/icub-main/src/libraries/icubmod/skinprototype/skinprototype.cpp:7:
/usr/local/src/robot/icub-main/src/libraries/icubmod/skinprototype/skinprototype.h: In constructor ‘SkinPrototype::SkinPrototype(int)’:
/usr/local/src/robot/icub-main/src/libraries/icubmod/skinprototype/skinprototype.h:40:74: error: class ‘SkinPrototype’ does not have any field named ‘mutex’
     SkinPrototype(int period=20) : PeriodicThread((double)period/1000.0),mutex(1)
                                                                          ^~~~~
In file included from /usr/local/src/robot/icub-main/build-pc104/src/libraries/icubmod/skinprototype/yarp_plugin_icubmod_skinprototype.cpp:13:
/usr/local/src/robot/icub-main/src/libraries/icubmod/skinprototype/skinprototype.h: In constructor ‘SkinPrototype::SkinPrototype(int)’:
/usr/local/src/robot/icub-main/src/libraries/icubmod/skinprototype/skinprototype.h:40:74: error: class ‘SkinPrototype’ does not have any field named ‘mutex’
     SkinPrototype(int period=20) : PeriodicThread((double)period/1000.0),mutex(1)
                                                                          ^~~~~
make[2]: *** [src/libraries/icubmod/skinprototype/CMakeFiles/skinprototype.dir/build.make:76: src/libraries/icubmod/skinprototype/CMakeFiles/skinprototype.dir/yarp_plugin_icubmod_skinprototype.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[2]: *** [src/libraries/icubmod/skinprototype/CMakeFiles/skinprototype.dir/build.make:63: src/libraries/icubmod/skinprototype/CMakeFiles/skinprototype.dir/skinprototype.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:2009: src/libraries/icubmod/skinprototype/CMakeFiles/skinprototype.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 18%] Built target canBusDoubleFTSensor
make: *** [Makefile:130: all] Error 2
```

Any pointers? Thanks.

dcam0050, 2019-12-18:
Deleting 'mutex(1)' from the initialiser seems to work. Please advise on repercussions?

pattacini, 2019-12-18:
Hi @dcam0050 hope you're doing well 😄 

Thanks for pointing out this compilation bug that we didn't catch with our CI.
`skinprototype` is a deprecated device so you can safely switch it off at CMake configuration time.

We're also going to patch this in the distro.

pattacini, 2019-12-18:
By the way, I assume you're upgrading your own machine and not the robot `iCubSheffield01` since it is not eligible for receiving the distro `2019.12.feat-01`.

pattacini, 2019-12-18:
@dcam0050 we've just produced the tag [`v1.14.1`][1], which has yielded the distro `2019.12.feat-02`.

[1]: https://github.com/robotology/icub-main/releases/tag/v1.14.1

closed 2019-12-18 by pattacini

dcam0050, 2020-01-21:
@pattacini sorry for the silence. Can you advise which is the latest distro compatible with iCubSheffield01?