robotology/icub-main #634: Support compilation against OpenCV 4

traversaro, 2020-01-13:
I tried to compile icub-main with OpenCV 4, and this is the error message with icub-main: 
~~~
2020-01-13T16:59:39.2925890Z Scanning dependencies of target stereoCalib
2020-01-13T16:59:39.3277190Z [ 74%] Building CXX object src/tools/stereoCalib/CMakeFiles/stereoCalib.dir/src/main.cpp.o
2020-01-13T16:59:41.9945860Z [ 74%] Building CXX object src/tools/stereoCalib/CMakeFiles/stereoCalib.dir/src/stereoCalibModule.cpp.o
2020-01-13T16:59:44.6748180Z [ 74%] Building CXX object src/tools/stereoCalib/CMakeFiles/stereoCalib.dir/src/stereoCalibThread.cpp.o
2020-01-13T16:59:46.4877660Z /Users/runner/runners/2.163.1/work/robotology-superbuild/robotology-superbuild/robotology/ICUB/src/tools/stereoCalib/src/stereoCalibThread.cpp:190:80: error: use of undeclared identifier 'CV_CALIB_CB_ADAPTIVE_THRESH'; did you mean 'CALIB_CB_ADAPTIVE_THRESH'?
2020-01-13T16:59:46.4877920Z                     foundL = findChessboardCorners(Left, boardSize, pointbufL, CV_CALIB_CB_ADAPTIVE_THRESH | CV_CALIB_CB_FAST_CHECK | CV_CALIB_CB_NORMALIZE_IMAGE);
2020-01-13T16:59:46.4878060Z                                                                                ^~~~~~~~~~~~~~~~~~~~~~~~~~~
2020-01-13T16:59:46.4878170Z                                                                                CALIB_CB_ADAPTIVE_THRESH
2020-01-13T16:59:46.4881460Z /usr/local/Cellar/opencv/4.2.0_1/include/opencv4/opencv2/calib3d.hpp:255:8: note: 'CALIB_CB_ADAPTIVE_THRESH' declared here
2020-01-13T16:59:46.4881600Z enum { CALIB_CB_ADAPTIVE_THRESH = 1,
2020-01-13T16:59:46.4881690Z        ^
2020-01-13T16:59:46.4882280Z /Users/runner/runners/2.163.1/work/robotology-superbuild/robotology-superbuild/robotology/ICUB/src/tools/stereoCalib/src/stereoCalibThread.cpp:190:110: error: use of undeclared identifier 'CV_CALIB_CB_FAST_CHECK'; did you mean 'CALIB_CB_FAST_CHECK'?
2020-01-13T16:59:46.4882450Z                     foundL = findChessboardCorners(Left, boardSize, pointbufL, CV_CALIB_CB_ADAPTIVE_THRESH | CV_CALIB_CB_FAST_CHECK | CV_CALIB_CB_NORMALIZE_IMAGE);
2020-01-13T16:59:46.4882580Z                                                                                                              ^~~~~~~~~~~~~~~~~~~~~~
2020-01-13T16:59:46.4883000Z                                                                                                              CALIB_CB_FAST_CHECK
2020-01-13T16:59:46.4883540Z /usr/local/Cellar/opencv/4.2.0_1/include/opencv4/opencv2/calib3d.hpp:258:8: note: 'CALIB_CB_FAST_CHECK' declared here
2020-01-13T16:59:46.4883650Z        CALIB_CB_FAST_CHECK      = 8,
2020-01-13T16:59:46.4883730Z        ^
2020-01-13T16:59:46.4884350Z /Users/runner/runners/2.163.1/work/robotology-superbuild/robotology-superbuild/robotology/ICUB/src/tools/stereoCalib/src/stereoCalibThread.cpp:190:135: error: use of undeclared identifier 'CV_CALIB_CB_NORMALIZE_IMAGE'; did you mean 'CALIB_CB_NORMALIZE_IMAGE'?
2020-01-13T16:59:46.4884480Z                     foundL = findChessboardCorners(Left, boardSize, pointbufL, CV_CALIB_CB_ADAPTIVE_THRESH | CV_CALIB_CB_FAST_CHECK | CV_CALIB_CB_NORMALIZE_IMAGE);
2020-01-13T16:59:46.4884610Z                                                                                                                                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~
2020-01-13T16:59:46.4884710Z                                                                                                                                       CALIB_CB_NORMALIZE_IMAGE
2020-01-13T16:59:46.4885200Z /usr/local/Cellar/opencv/4.2.0_1/include/opencv4/opencv2/calib3d.hpp:256:8: note: 'CALIB_CB_NORMALIZE_IMAGE' declared here
2020-01-13T16:59:46.4885300Z        CALIB_CB_NORMALIZE_IMAGE = 2,
2020-01-13T16:59:46.4885380Z        ^
2020-01-13T16:59:46.5302790Z /Users/runner/runners/2.163.1/work/robotology-superbuild/robotology-superbuild/robotology/ICUB/src/tools/stereoCalib/src/stereoCalibThread.cpp:191:81: error: use of undeclared identifier 'CV_CALIB_CB_ADAPTIVE_THRESH'; did you mean 'CALIB_CB_ADAPTIVE_THRESH'?
2020-01-13T16:59:46.5303450Z                     foundR = findChessboardCorners(Right, boardSize, pointbufR, CV_CALIB_CB_ADAPTIVE_THRESH | CV_CALIB_CB_FAST_CHECK | CV_CALIB_CB_NORMALIZE_IMAGE);
2020-01-13T16:59:46.5303580Z                                                                                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~
2020-01-13T16:59:46.5303670Z                                                                                 CALIB_CB_ADAPTIVE_THRESH
2020-01-13T16:59:46.5304200Z /usr/local/Cellar/opencv/4.2.0_1/include/opencv4/opencv2/calib3d.hpp:255:8: note: 'CALIB_CB_ADAPTIVE_THRESH' declared here
2020-01-13T16:59:46.5304310Z enum { CALIB_CB_ADAPTIVE_THRESH = 1,
2020-01-13T16:59:46.5304390Z        ^
2020-01-13T16:59:46.5304930Z /Users/runner/runners/2.163.1/work/robotology-superbuild/robotology-superbuild/robotology/ICUB/src/tools/stereoCalib/src/stereoCalibThread.cpp:191:111: error: use of undeclared identifier 'CV_CALIB_CB_FAST_CHECK'; did you mean 'CALIB_CB_FAST_CHECK'?
2020-01-13T16:59:46.5305070Z                     foundR = findChessboardCorners(Right, boardSize, pointbufR, CV_CALIB_CB_ADAPTIVE_THRESH | CV_CALIB_CB_FAST_CHECK | CV_CALIB_CB_NORMALIZE_IMAGE);
2020-01-13T16:59:46.5305320Z                                                                                                               ^~~~~~~~~~~~~~~~~~~~~~
2020-01-13T16:59:46.5305420Z                                                                                                               CALIB_CB_FAST_CHECK
2020-01-13T16:59:46.5305900Z /usr/local/Cellar/opencv/4.2.0_1/include/opencv4/opencv2/calib3d.hpp:258:8: note: 'CALIB_CB_FAST_CHECK' declared here
2020-01-13T16:59:46.5306000Z        CALIB_CB_FAST_CHECK      = 8,
2020-01-13T16:59:46.5306060Z        ^
2020-01-13T16:59:46.5306610Z /Users/runner/runners/2.163.1/work/robotology-superbuild/robotology-superbuild/robotology/ICUB/src/tools/stereoCalib/src/stereoCalibThread.cpp:191:136: error: use of undeclared identifier 'CV_CALIB_CB_NORMALIZE_IMAGE'; did you mean 'CALIB_CB_NORMALIZE_IMAGE'?
2020-01-13T16:59:46.5306750Z                     foundR = findChessboardCorners(Right, boardSize, pointbufR, CV_CALIB_CB_ADAPTIVE_THRESH | CV_CALIB_CB_FAST_CHECK | CV_CALIB_CB_NORMALIZE_IMAGE);
2020-01-13T16:59:46.5307040Z                                                                                                                                        ^~~~~~~~~~~~~~~~~~~~~~~~~~~
2020-01-13T16:59:46.5307170Z                                                                                                                                        CALIB_CB_NORMALIZE_IMAGE
2020-01-13T16:59:46.5307710Z /usr/local/Cellar/opencv/4.2.0_1/include/opencv4/opencv2/calib3d.hpp:256:8: note: 'CALIB_CB_NORMALIZE_IMAGE' declared here
2020-01-13T16:59:46.5307800Z        CALIB_CB_NORMALIZE_IMAGE = 2,
2020-01-13T16:59:46.5307870Z        ^
2020-01-13T16:59:46.5778150Z /Users/runner/runners/2.163.1/work/robotology-superbuild/robotology-superbuild/robotology/ICUB/src/tools/stereoCalib/src/stereoCalibThread.cpp:312:80: error: use of undeclared identifier 'CV_CALIB_CB_ADAPTIVE_THRESH'; did you mean 'CALIB_CB_ADAPTIVE_THRESH'?
2020-01-13T16:59:46.5778450Z                     foundL = findChessboardCorners(Left, boardSize, pointbufL, CV_CALIB_CB_ADAPTIVE_THRESH | CV_CALIB_CB_FAST_CHECK | CV_CALIB_CB_NORMALIZE_IMAGE);
2020-01-13T16:59:46.5778630Z                                                                                ^~~~~~~~~~~~~~~~~~~~~~~~~~~
2020-01-13T16:59:46.5778730Z                                                                                CALIB_CB_ADAPTIVE_THRESH
2020-01-13T16:59:46.5779300Z /usr/local/Cellar/opencv/4.2.0_1/include/opencv4/opencv2/calib3d.hpp:255:8: note: 'CALIB_CB_ADAPTIVE_THRESH' declared here
2020-01-13T16:59:46.5779630Z enum { CALIB_CB_ADAPTIVE_THRESH = 1,
2020-01-13T16:59:46.5789830Z        ^
2020-01-13T16:59:46.5790750Z /Users/runner/runners/2.163.1/work/robotology-superbuild/robotology-superbuild/robotology/ICUB/src/tools/stereoCalib/src/stereoCalibThread.cpp:312:110: error: use of undeclared identifier 'CV_CALIB_CB_FAST_CHECK'; did you mean 'CALIB_CB_FAST_CHECK'?
2020-01-13T16:59:46.5791360Z                     foundL = findChessboardCorners(Left, boardSize, pointbufL, CV_CALIB_CB_ADAPTIVE_THRESH | CV_CALIB_CB_FAST_CHECK | CV_CALIB_CB_NORMALIZE_IMAGE);
[... cut ...]
2020-01-13T16:59:46.7956150Z make[4]: *** [src/tools/stereoCalib/CMakeFiles/stereoCalib.dir/all] Error 2
2020-01-13T16:59:46.7963400Z make[3]: *** [all] Error 2
2020-01-13T16:59:46.7978370Z make[2]: *** [robotology/ICUB/CMakeFiles/YCMStamp/ICUB-build] Error 2
2020-01-13T16:59:46.7981270Z make[1]: *** [CMakeFiles/ICUB.dir/all] Error 2
2020-01-13T16:59:46.7989190Z make: *** [all] Error 2
2020-01-13T16:59:46.8018810Z ##[error]Process completed with exit code 2.
2020-01-13T16:59:46.8107470Z Cleaning up orphan processes
~~~

pattacini, 2020-01-13:
I think @xEnVrE made it compile on his fork.
@xEnVrE would you be willing to open up a PR to address this?

xEnVrE, 2020-01-13:
> I think @xEnVrE made it compile on his fork.

Unfortunately, I had no time and just disabled the component of `icub-main` that caused the errors becuase I needed to use  `yarp` + `icub-main (w/o ICUB_USE_OpenCV)` with `OpenCV 4` for some experiments.

Nicogene, 2020-01-14:
I can handle it, in the same way I handled https://github.com/robotology/yarp/pull/2162

pattacini, 2020-01-14:
@Nicogene thanks!

This shall go to `devel`.

However, given that we have lots of repos to update (icub-main, icub-basic-demos, icub-tutorials, iol, segmentation...), we definitely need to coordinate the overall effort.

We could take on this tomorrow at the dedicated meeting.

Nicogene, 2020-01-14:
I noticed also that compiling `icub-main` with opencv 4 there is a linking problem

```
CMakeFiles/stereoCalib.dir/src/main.cpp.o: In function `cvflann::anyimpl::big_any_policy<cv::String>::static_delete(void**)':
main.cpp:(.text._ZN7cvflann7anyimpl14big_any_policyIN2cv6StringEE13static_deleteEPPv[_ZN7cvflann7anyimpl14big_any_policyIN2cv6StringEE13static_deleteEPPv]+0x15): undefined reference to `cv::String::deallocate()'
CMakeFiles/stereoCalib.dir/src/main.cpp.o: In function `cvflann::anyimpl::big_any_policy<cv::String>::move(void* const*, void**)':
main.cpp:(.text._ZN7cvflann7anyimpl14big_any_policyIN2cv6StringEE4moveEPKPvPS5_[_ZN7cvflann7anyimpl14big_any_policyIN2cv6StringEE4moveEPKPvPS5_]+0x10): undefined reference to `cv::String::deallocate()'
main.cpp:(.text._ZN7cvflann7anyimpl14big_any_policyIN2cv6StringEE4moveEPKPvPS5_[_ZN7cvflann7anyimpl14big_any_policyIN2cv6StringEE4moveEPKPvPS5_]+0x24): undefined reference to `cv::String::deallocate()'
CMakeFiles/stereoCalib.dir/src/stereoCalibThread.cpp.o: In function `cv::FileStorage& cv::operator<< <cv::Mat>(cv::FileStorage&, cv::Mat const&) [clone .constprop.201]':
stereoCalibThread.cpp:(.text+0x179): undefined reference to `cv::write(cv::FileStorage&, cv::String const&, cv::Mat const&)'
stereoCalibThread.cpp:(.text+0x1cd): undefined reference to `cv::String::allocate(unsigned long)'
stereoCalibThread.cpp:(.text+0x227): undefined reference to `cv::error(int, cv::String const&, char const*, char const*, int)'
stereoCalibThread.cpp:(.text+0x22f): undefined reference to `cv::String::deallocate()'
stereoCalibThread.cpp:(.text+0x244): undefined reference to `cv::String::deallocate()'

``` 

This is due to the fact that the content of `${OpenCV_LIBRARIES}` changed in opencv 4.

## OpenCV 3
```
opencv_calib3d
opencv_core
opencv_features2d
opencv_flann
opencv_highgui
opencv_imgcodecs
opencv_imgproc
opencv_ml
opencv_objdetect
opencv_photo
opencv_shape
opencv_stitching
opencv_superres
opencv_video
opencv_videoio
opencv_videostab
opencv_viz
opencv_aruco
opencv_bgsegm
opencv_bioinspired
opencv_ccalib
opencv_datasets
opencv_dpm
opencv_face
opencv_freetype
opencv_fuzzy
opencv_hdf
opencv_line_descriptor
opencv_optflow
opencv_phase_unwrapping
opencv_plot
opencv_reg
opencv_rgbd
opencv_saliency
opencv_stereo
opencv_structured_light
opencv_surface_matching
opencv_text
opencv_ximgproc
opencv_xobjdetect
opencv_xphoto
opencv_calib3d
opencv_core
opencv_features2d
opencv_flann
opencv_highgui
opencv_imgcodecs
opencv_imgproc
opencv_ml
opencv_objdetect
opencv_photo
opencv_shape
opencv_stitching
opencv_superres
opencv_video
opencv_videoio
opencv_videostab
opencv_viz
opencv_aruco
opencv_bgsegm
opencv_bioinspired
opencv_ccalib
opencv_datasets
opencv_dpm
opencv_face
opencv_freetype
opencv_fuzzy
opencv_hdf
opencv_line_descriptor
opencv_optflow
opencv_phase_unwrapping
opencv_plot
opencv_reg
opencv_rgbd
opencv_saliency
opencv_stereo
opencv_structured_light
opencv_surface_matching
opencv_text
opencv_ximgproc
opencv_xobjdetect
opencv_xphoto
```

## OpenCV 4
```
opencv_video
opencv_imgproc
opencv_features2d
opencv_videoio
opencv_stitching
opencv_imgcodecs
opencv_calib3d
opencv_core
opencv_photo
opencv_highgui
opencv_dnn
opencv_gapi
opencv_ml
opencv_flann
opencv_objdetect
opencv_bioinspired
opencv_ximgproc
opencv_hfs
opencv_stereo
opencv_dpm
opencv_phase_unwrapping
opencv_videostab
opencv_face
opencv_tracking
opencv_hdf
opencv_img_hash
opencv_shape
opencv_superres
opencv_text
opencv_xfeatures2d
opencv_freetype
opencv_structured_light
opencv_reg
opencv_datasets
opencv_surface_matching
opencv_sfm
opencv_dnn_objdetect
opencv_saliency
opencv_xobjdetect
opencv_bgsegm
opencv_xphoto
opencv_optflow
opencv_cvv
opencv_line_descriptor
opencv_ccalib
opencv_aruco
opencv_fuzzy
opencv_rgbd
opencv_plot

```

Nicogene, 2020-01-14:
This issue seems more fishy than the one encountered in robotology/yarp#2162

Nicogene, 2020-01-14:
Another not-trivial problem is that `cvInitUndistortMap` seems to be moved to the impl headers, then is not anymore reachable.

It is used in all the camera calibration modules, we have to find a function that replaces it.
Maybe it has to be done in the refactoring of camera calibration modules

Nicogene, 2020-01-22:
Now the support is available in `devel`

PR #636 (2020-01-22): Support OpenCV4
This PR drops the support of OpenCV 2 and add the support to OpenCV 4.

TODO:

- [x] Compile with `opencv` 3.
- [x] Compile with `opencv` 4.
- [x] Test `camCalib`s.

It fixes #634
closed 2020-01-22 by Nicogene