robotology/icub-main #637: iCub_SIM crashes in some circumstances

pattacini, 2020-01-20:
I've been observing systematic crashes of `iCub_SIM` when it gets launched within the https://github.com/vvv-school/tutorial_gaze-interface application.

Below a Gitpod workspace that helps verify this quickly:

[image]

I suspect that the modifications introduced by https://github.com/robotology/icub-main/pull/615 could cause the problem.

pattacini, 2020-02-08:
@Nicogene carried out an extensive analysis and managed to trace back the bug:
```
Thread 4 "iCub_SIM" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7ff11f5d4700 (LWP 17513)]
0x00007ff1203084e0 in ?? ()
(gdb) bt
#0  0x00007ff1203084e0 in ?? ()
#1  0x00007ff12cda3ed4 in yarp::os::impl::BottleImpl::synch (this=0x7ff1200937e0)
    at /home/ngenesio/robotology/robotology-superbuild/robotology/YARP/src/libYARP_os/src/yarp/os/impl/BottleImpl.cpp:539
#2  0x00007ff12cda41a5 in yarp::os::impl::BottleImpl::synch (this=0x7ff1200937e0)
    at /home/ngenesio/robotology/robotology-superbuild/robotology/YARP/src/libYARP_os/src/yarp/os/impl/BottleImpl.cpp:515
#3  0x00007ff12cda426f in yarp::os::impl::BottleImpl::write (this=0x7ff1200937e0, writer=...)
    at /home/ngenesio/robotology/robotology-superbuild/robotology/YARP/src/libYARP_os/src/yarp/os/impl/BottleImpl.cpp:426
#4  0x00007ff12cd210d1 in yarp::os::Bottle::write (
    this=0x7ff12d085290 <yarp::os::impl::LogForwarder::forward(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)::b>, writer=...) at /home/ngenesio/robotology/robotology-superbuild/robotology/YARP/src/libYARP_os/src/yarp/os/Bottle.cpp:231
#5  0x00007ff12ce024de in yarp::os::impl::PortCoreOutputUnit::sendHelper (this=0x7ff114002e30)
    at /home/ngenesio/robotology/robotology-superbuild/robotology/YARP/src/libYARP_os/src/yarp/os/impl/PortCoreOutputUnit.cpp:295
#6  0x00007ff12ce01ee5 in yarp::os::impl::PortCoreOutputUnit::run (this=0x7ff114002e30)
    at /home/ngenesio/robotology/robotology-superbuild/robotology/YARP/src/libYARP_os/src/yarp/os/impl/PortCoreOutputUnit.cpp:99
#7  0x00007ff12ce14e41 in theExecutiveBranch (args=0x7ff114002e30)
    at /home/ngenesio/robotology/robotology-superbuild/robotology/YARP/src/libYARP_os/src/yarp/os/impl/ThreadImpl.cpp:80
#8  0x00007ff12ce16762 in std::__invoke_impl<void, void (*)(void*), void*> (__f=@0x7ff12007a2e0: 0x7ff12ce14d70 <theExecutiveBranch(void*)>, 
    __args=@0x7ff12007a2d8: 0x7ff114002e30) at /usr/bin/../lib/gcc/x86_64-linux-gnu/7.5.0/../../../../include/c++/7.5.0/bits/invoke.h:60
#9  0x00007ff12ce166b2 in std::__invoke<void (*)(void*), void*> (__fn=@0x7ff12007a2e0: 0x7ff12ce14d70 <theExecutiveBranch(void*)>, 
    __args=@0x7ff12007a2d8: 0x7ff114002e30) at /usr/bin/../lib/gcc/x86_64-linux-gnu/7.5.0/../../../../include/c++/7.5.0/bits/invoke.h:95
---Type <return> to continue, or q <return> to quit---
#10 0x00007ff12ce16672 in std::thread::_Invoker<std::tuple<void (*)(void*), void*> >::_M_invoke<0ul, 1ul> (this=0x7ff12007a2d8)
    at /usr/bin/../lib/gcc/x86_64-linux-gnu/7.5.0/../../../../include/c++/7.5.0/thread:234
#11 0x00007ff12ce16625 in std::thread::_Invoker<std::tuple<void (*)(void*), void*> >::operator() (this=0x7ff12007a2d8)
    at /usr/bin/../lib/gcc/x86_64-linux-gnu/7.5.0/../../../../include/c++/7.5.0/thread:243
#12 0x00007ff12ce163e9 in std::thread::_State_impl<std::thread::_Invoker<std::tuple<void (*)(void*), void*> > >::_M_run (this=0x7ff12007a2d0)
    at /usr/bin/../lib/gcc/x86_64-linux-gnu/7.5.0/../../../../include/c++/7.5.0/thread:186
#13 0x00007ff12c9e36ef in ?? () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#14 0x00007ff12af926db in start_thread (arg=0x7ff11f5d4700) at pthread_create.c:463
#15 0x00007ff12c43e88f in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95
```

In a nutshell, `iCub_SIM` crashes only when launched from the `yarpmanager` with the env variable `YARP_FORWARD_LOG_ENABLE` enabled. The latter variable allows for redirecting the output of `LogStream` directives to the `yarplogger`.

Those log directives are called in the simulator quite intensively when dealing with world operations as it happens for the tutorial at hand. As a workaround, https://github.com/robotology/icub-main/commit/5f1e0be0bf47a193462288af1eac3f56fe9007df disables those printouts, which actually are quite redundant and are meant for debugging purpose only.

It might be useful to investigate this further to try to understand whether:
- is there any race condition in the way log directives are called in the simulator?
- are log directives inherently thread-safe?

closed 2020-02-08 by pattacini

Nicogene, 2020-02-10:
> - are log directives inherently thread-safe?

I'd move this issue/question on YARP :thinking: