robotology/icub-main #640: Crash of wholeBodyDynamics on iCubGenova01

traversaro, 2020-01-31:
With the latest devel branch, running the wholeBodyDynamics on iCubGenova01 with `--autoconnect --no_legs --headV2` results in a crash in the following block (**observerThread.cpp**):

```
    // DYN/SKIN CONTACTS
   dynContacts = icub->upperTorso->leftSensor->getContactList();
    const dynContactList& contactListR = icub->upperTorso->rightSensor->getContactList();
    dynContacts.insert(dynContacts.begin(), contactListR.begin(), contactListR.end());
    // for each dynContact find the related skinContact (if any) and set the wrench in it
    unsigned long cId;
    bool contactFound=false;
    for(unsigned int i=0; i<dynContacts.size(); i++)
    {
        cId = dynContacts[i].getId();
        for(unsigned int j=0; j<skinContacts.size(); j++)
        {
            if(cId == skinContacts[j].getId())
            {
				yDebug() << i << dynContacts[i].getForceMoment().toString();
                skinContacts[j].setForceMoment( dynContacts[i].getForceMoment() );
                contactFound = true;
                j = skinContacts.size();    // break from the inside for loop
            }
        }
        // if there is no associated skin contact, create one
        if(!contactFound)
            skinContacts.push_back(skinContact(dynContacts[i]));
        contactFound = false;
    }
```

Nicogene, 2020-02-04:
Digging into it I found out that it is not robot-related, I suspect that the cause is the introduction of [move semantics in yarp::sig::Vector.](https://github.com/robotology/yarp/pull/1974)

I suspect it because I had a old yarp on my pc and it was not crashing, after updating yarp it crashed also on my pc.

Running it with gdb I have this output:

[image]

Nicogene, 2020-02-04:
The strange thing is that the segmentation fault happens at this line:

```c++
char* buf = new char[len];
```
wher len is !=0.

Nicogene, 2020-02-04:
Running `wholeBodyDynamics` with the latest yarp master with Valgrind I have:
```
==11076== 
==11076== HEAP SUMMARY:
==11076==     in use at exit: 289 bytes in 2 blocks
==11076==   total heap usage: 2,111,778 allocs, 2,111,776 frees, 156,247,902 bytes allocated
==11076== 
==11076== LEAK SUMMARY:
==11076==    definitely lost: 0 bytes in 0 blocks
==11076==    indirectly lost: 0 bytes in 0 blocks
==11076==      possibly lost: 0 bytes in 0 blocks
==11076==    still reachable: 289 bytes in 2 blocks
==11076==         suppressed: 0 bytes in 0 blocks
==11076== Rerun with --leak-check=full to see details of leaked memory
==11076== 
==11076== For counts of detected and suppressed errors, rerun with: -v
==11076== Use --track-origins=yes to see where uninitialised values come from
==11076== ERROR SUMMARY: 1707 errors from 11 contexts (suppressed: 0 from 0)
```
It doesn't crash, probably because valgrind slows it down  a lot.

Reverting the commit that add yarp::sig::Vector move semantics **it doesn't** crash both with and without valgrind and I have this output:

```
==16126== 
==16126== HEAP SUMMARY:
==16126==     in use at exit: 289 bytes in 2 blocks
==16126==   total heap usage: 2,376,095 allocs, 2,376,093 frees, 177,525,007 bytes allocated
==16126== 
==16126== LEAK SUMMARY:
==16126==    definitely lost: 0 bytes in 0 blocks
==16126==    indirectly lost: 0 bytes in 0 blocks
==16126==      possibly lost: 0 bytes in 0 blocks
==16126==    still reachable: 289 bytes in 2 blocks
==16126==         suppressed: 0 bytes in 0 blocks
==16126== Rerun with --leak-check=full to see details of leaked memory
==16126== 
==16126== For counts of detected and suppressed errors, rerun with: -v
==16126== ERROR SUMMARY: 0 errors from 0 contexts (suppressed: 0 from 0)
```

pattacini, 2020-02-04:
Unless we will fix this up soonish, I recommend that we roll back this modification in `yarp/master`.
Moreover, we could grab the opportunity to use smart pointers underneath to avoid manual disposal of the allocated memory.

cc @drdanz