robotology/icub-main #643: Don't build on Windows 10, VIsual Studio 19

KhomutovEvgeniy, 2020-02-21:
I generated and installed YARP on Windows 10 with Visual Studio 19 by following the instruction https://www.yarp.it/install_yarp_windows.html from sources. Yarp is working.
But if I try to generate files iCub from iCub Sources https://github.com/robotology/icub-main.git I get: Error in cinfiguration process, project files may be invalid in CMake.

Files from http://wiki.icub.org/wiki/Downloads - icub-mai-src-1.12.0.zip is generated in CMake. But next step: in Debug configuration build the target INSTALL (Visual Studio 2019) I get a Errors
[image]

What am I doing wrong?

pattacini, 2020-02-21:
Hi @KhomutovEvgeniy 

We are currently in the process of applying a major overhaul to the way we deliver software.
An announcement that gives you already an overview can be browsed at https://github.com/robotology/QA/issues/377.

Essentially, you ought to visit our [**Versioning Table**][1] and act according to the brief instructions you'll find therein. You'll get better pointers to the latest software you can clone locally.

[1]: http://wiki.icub.org/wiki/Software_Versioning_Table

pattacini, 2020-02-27:
Closing, feel free to open it up again if needed.

closed 2020-02-27 by pattacini