robotology/icub-main #644: What is the skeletal structure format of skeletonV3.ini?

linzdg, 2020-02-25:
The skeletal structure format of skeletonV3.ini have something in common with BVH (BioVision Hierarchical). But the BVH  defines each bone using keyword such as CHANNELS, OFFSET. In skeletonV3.ini defines each bone using keyword such as RPY_XYZ , DH . Is it possible to get the BVH format of icub skeleton.

pattacini, 2020-02-25:
I presume you're referring to [`iCubGui`][1] conf files.

Maybe @ale-git can chime in.

[1]: ./app/iCubGui

ale-git, 2020-02-25:
Yes, the iCubGui file format has been derived from the BVH format and adapted to our needs since the iCubGui was derived from a open source simple animation software. The main difference is that our format supports Denavit Hartenberg representation. We don't have the BVH format of the iCub model because the iCubGui was developed to read the already existing DH iCub model. BTW you can derive the BVH format from the iCub configuration file but it requires some effort.

linzdg, 2020-02-26:
Hi , @ale-git 
I have some doubt that if DH( -15 -10.6 -90.0 90.0) ，then OFFSET(-15 0 -10.6).
if RPY_XYZ( 0.0 0.0 -90.0 -21.6 -73.65 -60.5), then  OFFSET (-21.6 -73.65 -60.5 0 -90 0).
Is it correct to rewrite like this?

ale-git, 2020-02-26:
Hi @linzdg , I'm not an expert about bvh files, but i think that the OFFSET parameters represent cartesian displacements (links). There ar no explicit rotation offsets because the rotation angles are stored apart in the MOTION section. Here a document that explains the BVH format:

http://www.dcs.shef.ac.uk/intranet/research/public/resmes/CS0111.pdf

Hope this helps

pattacini, 2020-03-02:
Closing, feel free to open it up again if needed.

linzdg, 2020-03-12:
Now I have obtained the bvh file of iCub. And after retarget the human motion,  I obtained the MOTION section of icub, is there any way to apply to the true robot and let it move？The Motion section consisit of euler angle of each joint,  but Cartesian Interface of the positionMove needs the encoder data.

ale-git, 2020-03-12:
The Dynamic Interaction Control group https://dic.iit.it/ is working on mimic, maybe they have some modules that translates from human body measures to robot joint space?

pattacini, 2020-03-12:
Hi @linzdg 

What exactly do you mean when saying "Euler angle of each joint"? A single revolute joint can only be commanded by asking it to move to a specified standard angle. Euler angles serve to determine orientations in 3D of a structure like a kinematic chain (for example the robot arm), not a joint.

The Cartesian Interface is a component that lets you control the pose of the end-point of a kinematic chain (for example the hand of the robot arm), thus comprising its orientation. The Cartesian Interface doesn't resort to `positionMove` of the joint as it commands the whole structure in the operational space. Therefore, the joints are moved by the algorithm itself.

To sum up, you ought to work in the joint space and come up with a suitable sequence of joint positions moving over time according to the format that can be consumed by [`yarpdatadumper`][1]. Once done with that, to replay the sequences on the robot you can make use of the dual component [`yarpdataplayer`][2].

[1]: http://yarp.it/yarpdatadumper.html
[2]: http://yarp.it/yarpdataplayer.html

linzdg, 2020-03-13:
BVH MOTION arrays provide Zrotation Xrotation Yrotation values which are Euler angles in degrees corresponding to the skeleton information defined above. 
For example, I want to use this Zrotation Xrotation Yrotation values to make left leg move. But no method was found to apply to icub.
Here's the data I got from retargetting.
HIERARCHY
ROOT Hips
{
	OFFSET 0.000000 0.000000 0.000000
	CHANNELS 6 Xposition Yposition Zposition Zrotation Yrotation Xrotation 
	JOINT base_to_left_leg_1
	{
		OFFSET 0.000000 0.000000 0.000000
		CHANNELS 3 Zrotation Yrotation Xrotation
		JOINT 0_left_leg_fwd_2
		{
			OFFSET 6.810000 -11.990000 0.000000
			CHANNELS 3 Zrotation Yrotation Xrotation
			JOINT 1_left_leg_spread_3
			{
				OFFSET 0.000000 0.000000 0.000000
				CHANNELS 3 Zrotation Yrotation Xrotation
				JOINT 2_left_leg_rot_4
				{
					OFFSET 0.000000 -24.000000 0.000000
					CHANNELS 3 Zrotation Yrotation Xrotation
					JOINT 3_left_knee_fold_5
					{
						OFFSET 0.000000 -22.000000 0.000000
						CHANNELS 3 Zrotation Yrotation Xrotation
						JOINT 4_left_ankle_pitch_6
						{
							OFFSET 0.000000 0.000000 0.000000
							CHANNELS 3 Zrotation Yrotation Xrotation
							JOINT 5_left_ankle_roll_7
							{
								OFFSET 0.000000 -4.100000 0.000000
								CHANNELS 3 Zrotation Yrotation Xrotation
								End Site
								{
									OFFSET 0.000000 0.000000 0.000000
								}
							}
						}
					}
				}
			}
		}
	}
MOTION
Frames: 408
Frame Time: 0.008333
1.264031 65.080997 -8.960823 3.084827 13.962112 -7.686169 3.755391 -3.438108 -3.234098 4.267096 -2.626119 -1.939569 4.448896 -2.948539 -2.282182 -16.827582 -5.937031 16.566146 21.777096 14.325650 13.662276 24.228061 -1.332077 13.248087 0.000000 0.000000 0.000000
(The first three are Xposition Yposition Zposition)

pattacini, 2020-03-17:
As said previously, you can command iCub either in joint space by streaming out joints set-points or in the operational space through the use of specific IK components like the Cartesian and Gaze interfaces.

closed 2020-03-21 by pattacini