robotology/icub-main #645: Python bindings not working due to recent mutex changes

diegoferigo, 2020-02-29:
I suspect that https://github.com/robotology/icub-main/pull/618 caused problem to the Python bindings. You can find in the bottom of this comment the compiler output. It seems that the swig autogenerated code tries to copy a recursive mutex, and the type is not copiable.

If I comment all the skin bindings ([here](https://github.com/robotology/icub-main/blob/7ee911aa6020c66107806b6547a62824b9b6a51f/bindings/icub.i#L43-L51), [here](https://github.com/robotology/icub-main/blob/7ee911aa6020c66107806b6547a62824b9b6a51f/bindings/icub.i#L93-L101) and [here](https://github.com/robotology/icub-main/blob/7ee911aa6020c66107806b6547a62824b9b6a51f/bindings/icub.i#L142-L164)), swig is happy again.

cc @pattacini 

<details>
<summary>Build error</summary>

```
/iit/sources/robotology-superbuild/build/robotology/ICUB/bindings/icubPYTHON_wrap.cxx:17373:3: warning: delete called on 'iCub::iKin::iKinIterateCallback' that is abstract but has non-virtual destructor [-Wdelete-abstract-non-virtual-dtor]
  delete arg1;
  ^
/iit/sources/robotology-superbuild/build/robotology/ICUB/bindings/icubPYTHON_wrap.cxx:27481:3: warning: delete called on 'iCub::ctrl::ff2LayNN' that is abstract but has non-virtual destructor [-Wdelete-abstract-non-virtual-dtor]
  delete arg1;
  ^
/iit/sources/robotology-superbuild/build/robotology/ICUB/bindings/icubPYTHON_wrap.cxx:32412:12: error: overload resolution selected deleted operator '='
      arg2 = *temp;
      ~~~~ ^ ~~~~~
/usr/bin/../lib/gcc/x86_64-linux-gnu/9/../../../../include/c++/9/mutex:101:22: note: candidate function has been explicitly deleted
    recursive_mutex& operator=(const recursive_mutex&) = delete;
                     ^
/iit/sources/robotology-superbuild/build/robotology/ICUB/bindings/icubPYTHON_wrap.cxx:32416:35: error: overload resolution selected deleted operator '='
  if (arg1) (arg1)->recursive_mtx = arg2;
            ~~~~~~~~~~~~~~~~~~~~~ ^ ~~~~
/usr/bin/../lib/gcc/x86_64-linux-gnu/9/../../../../include/c++/9/mutex:101:22: note: candidate function has been explicitly deleted
    recursive_mutex& operator=(const recursive_mutex&) = delete;
                     ^
/iit/sources/robotology-superbuild/build/robotology/ICUB/bindings/icubPYTHON_wrap.cxx:32438:10: error: overload resolution selected deleted operator '='
  result =  ((arg1)->recursive_mtx);
  ~~~~~~ ^  ~~~~~~~~~~~~~~~~~~~~~~~
/usr/bin/../lib/gcc/x86_64-linux-gnu/9/../../../../include/c++/9/mutex:101:22: note: candidate function has been explicitly deleted
    recursive_mutex& operator=(const recursive_mutex&) = delete;
                     ^
/iit/sources/robotology-superbuild/build/robotology/ICUB/bindings/icubPYTHON_wrap.cxx:32439:39: error: call to deleted constructor of 'std::recursive_mutex'
  resultobj = SWIG_NewPointerObj((new std::recursive_mutex(static_cast< const std::recursive_mutex& >(result))), SWIGTYPE_p_std__recursive_mutex, SWIG_POINTER_OWN |  0 );
                                      ^                    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/iit/sources/robotology-superbuild/build/robotology/ICUB/bindings/icubPYTHON_wrap.cxx:1187:89: note: expanded from macro 'SWIG_NewPointerObj'
#define SWIG_NewPointerObj(ptr, type, flags)            SWIG_Python_NewPointerObj(NULL, ptr, type, flags)
                                                                                        ^~~
/usr/bin/../lib/gcc/x86_64-linux-gnu/9/../../../../include/c++/9/mutex:100:5: note: 'recursive_mutex' has been explicitly marked deleted here
    recursive_mutex(const recursive_mutex&) = delete;
    ^
/iit/sources/robotology-superbuild/build/robotology/ICUB/bindings/icubPYTHON_wrap.cxx:41883:13: warning: 'Mutex' is deprecated: Use std::mutex instead [-Wdeprecated-declarations]
  yarp::os::Mutex *arg2 = (yarp::os::Mutex *) 0 ;
            ^
/iit/sources/robotology-superbuild/build/install/include/yarp/os/Mutex.h:34:7: note: 'Mutex' has been explicitly marked deprecated here
class YARP_os_DEPRECATED_API_MSG("Use std::mutex instead") Mutex
      ^
/iit/sources/robotology-superbuild/build/install/include/yarp/os/api.h:23:43: note: expanded from macro 'YARP_os_DEPRECATED_API_MSG'
#    define YARP_os_DEPRECATED_API_MSG(X) YARP_DEPRECATED_API_MSG(X)
                                          ^
/iit/sources/robotology-superbuild/build/install/include/yarp/conf/api.h:131:38: note: expanded from macro 'YARP_DEPRECATED_API_MSG'
#  define YARP_DEPRECATED_API_MSG(X) YARP_DEPRECATED_COMPILER_SPECIFIC_MSG(X) YARP_os_API
                                     ^
/iit/sources/robotology-superbuild/build/install/include/yarp/conf/api.h:117:69: note: expanded from macro 'YARP_DEPRECATED_COMPILER_SPECIFIC_MSG'
#  define YARP_DEPRECATED_COMPILER_SPECIFIC_MSG(MSG) __attribute__((__deprecated__(MSG)))
                                                                    ^
/iit/sources/robotology-superbuild/build/robotology/ICUB/bindings/icubPYTHON_wrap.cxx:41883:38: warning: 'Mutex' is deprecated: Use std::mutex instead [-Wdeprecated-declarations]
  yarp::os::Mutex *arg2 = (yarp::os::Mutex *) 0 ;
                                     ^
/iit/sources/robotology-superbuild/build/install/include/yarp/os/Mutex.h:34:7: note: 'Mutex' has been explicitly marked deprecated here
class YARP_os_DEPRECATED_API_MSG("Use std::mutex instead") Mutex
      ^
/iit/sources/robotology-superbuild/build/install/include/yarp/os/api.h:23:43: note: expanded from macro 'YARP_os_DEPRECATED_API_MSG'
#    define YARP_os_DEPRECATED_API_MSG(X) YARP_DEPRECATED_API_MSG(X)
                                          ^
/iit/sources/robotology-superbuild/build/install/include/yarp/conf/api.h:131:38: note: expanded from macro 'YARP_DEPRECATED_API_MSG'
#  define YARP_DEPRECATED_API_MSG(X) YARP_DEPRECATED_COMPILER_SPECIFIC_MSG(X) YARP_os_API
                                     ^
[... 111 lines cut ...]
/iit/sources/robotology-superbuild/build/robotology/ICUB/bindings/icubPYTHON_wrap.cxx:46094:63: warning: 'RateThreadWrapper' is deprecated [-Wdeprecated-declarations]
    return (void *)((yarp::os::PeriodicThread *)  ((yarp::os::RateThreadWrapper *) x));
                                                              ^
/iit/sources/robotology-superbuild/build/install/include/yarp/os/RateThread.h:243:7: note: 'RateThreadWrapper' has been explicitly marked deprecated here
class YARP_os_DEPRECATED_API RateThreadWrapper : public PeriodicThread
      ^
/iit/sources/robotology-superbuild/build/install/include/yarp/os/api.h:22:36: note: expanded from macro 'YARP_os_DEPRECATED_API'
#    define YARP_os_DEPRECATED_API YARP_DEPRECATED_API
                                   ^
/iit/sources/robotology-superbuild/build/install/include/yarp/conf/api.h:130:31: note: expanded from macro 'YARP_DEPRECATED_API'
#  define YARP_DEPRECATED_API YARP_DEPRECATED_COMPILER_SPECIFIC YARP_os_API
                              ^
/iit/sources/robotology-superbuild/build/install/include/yarp/conf/api.h:116:60: note: expanded from macro 'YARP_DEPRECATED_COMPILER_SPECIFIC'
#  define YARP_DEPRECATED_COMPILER_SPECIFIC __attribute__((__deprecated__))
                                                           ^
13 warnings and 4 errors generated.
bindings/CMakeFiles/icub.dir/build.make:62: recipe for target 'bindings/CMakeFiles/icub.dir/icubPYTHON_wrap.cxx.o' failed
make[2]: *** [bindings/CMakeFiles/icub.dir/icubPYTHON_wrap.cxx.o] Error 1
CMakeFiles/Makefile2:4401: recipe for target 'bindings/CMakeFiles/icub.dir/all' failed
make[1]: *** [bindings/CMakeFiles/icub.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2
```

</details>

diegoferigo, 2020-02-29:
cc @Nicogene

pattacini, 2020-02-29:
The involved PR doesn't seem to be #618 but rather #616, which dates back to **4 months ago**.
@diegoferigo can you confirm that you were working on a system 4 months old at least before updating? This would help to narrow down the glitch.

We're not currently checking bindings in our CI system yet. Probably, it would be worth doing that as well.

diegoferigo, 2020-02-29:
No it's the first time I try using the Python bindings of this repo. I saw that in the past there were few persons using them. Thanks, I'll have a look to the CI.

pattacini, 2020-03-01:
In essence, I've checked that [`skinPartBase::recursive_mtx`][1] needs actually to be recursive since some methods of the derived class `skinPart` call each other, which are protected by the mutex.

I'm not a SWIG user but I tried out variations of [this solution][2] unsuccessfully.

[1]: https://github.com/robotology/icub-main/blob/master/src/libraries/skinDynLib/include/iCub/skinDynLib/skinPart.h#L63
[2]: https://stackoverflow.com/questions/59023574/problem-wrapping-a-c-class-with-a-mutex-in-a-member-using-swig

pattacini, 2020-03-01:
The problem should have been present also with the `yarp::os::RecursiveMutex` as it simply a wrapper over `std::recursive_mutex`.

traversaro, 2020-03-01:
Unless someone has a strict need for them, we can just disable the bindings for skinDynLib .

pattacini, 2020-03-01:
> As a general thing I think mutual exclusion should be dealt at the thread level rather than at the data structure level exactly for avoiding these issues

I fully concur with this approach 👍 

> explicitly deleting (with `= delete`) at the C++ the copy constructor and copy operator of the affected classes would be enough.

Uhm, those classes explicitly define copy constructors and copy operators so I'm afraid this is not doable. Instead, we should fiddle with the SWIG configuration file and I don't know how to do that.

traversaro, 2020-03-01:
> > explicitly deleting (with `= delete`) at the C++ the copy constructor and copy operator of the affected classes would be enough.
> 
> Uhm, those classes explicitly define copy constructors and copy operators so I'm afraid this is not doable. Instead, we should fiddle with the SWIG configuration file and I don't know how to do that. The solution I came across and posted above goes in this direction.

Sorry, then I missed something... @diegoferigo on which methods SWIG if failing? If the copy operator is correctly handled, perhaps we just need to correctly define the copy constructor?

pattacini, 2020-03-01:
There are the lines in question:
https://github.com/robotology/icub-main/blob/9340b2345f10b598899fa980b49c85f48fee417f/src/libraries/skinDynLib/src/skinPart.cpp#L11-L28

I've made an attempt to modify the copy constructor to only copy all the data members but the recursive mutex. This approach didn't come off either.

guidoschillaci, 2020-04-23:
Hi, do you maybe have any update about this issue? Otherwise, which version with working python bindings would you suggest me to use?

Thanks!
Guido

pattacini, 2020-04-23:
Hi @guidoschillaci 

We'll do our best to deliver a fix along with our next distro 2020.05 scheduled this May. Sorry for the inconvenience.

At any rate, you may consider trying out the workaround reported by @diegoferigo in his opening post above, where he suggested commenting out the following few lines:

> If I comment all the skin bindings ([here](https://github.com/robotology/icub-main/blob/7ee911aa6020c66107806b6547a62824b9b6a51f/bindings/icub.i#L43-L51), [...]

guidoschillaci, 2020-04-24:
Hi @pattacini 

thanks a lot for the quick reply, I look forward to the new release!

As for the moment, unfortunately the suggested workaround would not work for me, as I am actually interested in using the skin with the python bindings. I will try with icub-main v1.14.1, or would you suggest me a more recent commit?

pattacini, 2020-04-24:
My hunch is that any release delivered earlier than #612 (merged Oct 28, 2019) could be a good candidate.

Skin bindings have been introduced with #544 and are clashing with SWIG because of the recursive mutexes (introduced with https://github.com/robotology/icub-main/commit/6f799216f98b43b77aa834de2b98a7c4ba6a3d27), which are in turn not copyable. Anyway, #544 came after the recursive mutexes and – assuming that #544 was correct – I tend to consider the refactoring we applied to migrate to the use of `std::mutex` (see #612) what is now causing the problem.

Nicogene, 2020-04-29:
This issue should be fixed now, pull the latest `devel` and try.

Closing for now, reopening in case of further problems

PR #655 (2020-04-29): Fix compilation of the bindings after mutexes cleanup
This PR fixes #645 .

The compilation should be fine my only concern is that maybe there could be runtime issues, but I don't know how to test it. I might need help on it.
commit 23916719 (2020-04-29): Bindings: fix compilation

It fixes #645
closed 2020-04-29 by Nicogene