robotology/icub-main #651: Install relocatable CMake config files

traversaro, 2020-04-03:
As of version v1.15.1 , the `ICUB` CMake package install CMake config files that contains absolute references to the `CMAKE_INSTALL_PREFIX` of the machine in which they were generated, making the installation artifacts non-relocatable (see https://cmake.org/cmake/help/v3.3/manual/cmake-packages.7.html#creating-relocatable-packages). 

This is problematic for two reasons:
* Windows binary installers (the current workaround is to force the installation to be in the same exact directory used for the install in the build machine) 
* Cache for CI (see https://github.com/robotology/whole-body-estimators/issues/60) 

It would be ideal if icub-main could install relocatable CMake config files. A possible strategy is to use YCM's `InstallBasicPackageFiles` as we do in https://github.com/robotology/how-to-export-cpp-library, removing the old hard coded logic for generating the CMake config  files contained in this repo.

pattacini, 2020-04-04:
This would be a nice improvement indeed.
Is anyone eager to take on it?

I'm not sure I'll have time for it.
cc @Nicogene @traversaro

traversaro, 2020-04-04:
I may work on it in the next months, but I do not have a precise schedule yet. Obviously if there are other volunteers that would be great.

Nicogene, 2020-04-06:
I may handle it, but since icub-main is slightly more articulated than repos i worked on (cmake-side) i would need support

traversaro, 2020-04-06:
I would be happy to provide support.

Nicogene, 2020-06-23:
Fixed by #652

PR #652 (2020-06-23): Add relocatable CMakeConfig files.
This PR introduce a deep refactor on how the Config.cmake are generated inside `icub-main`.

With this PR all the libraries are handled as components so:
`find_package(ICUB COMPONENTS iKin)` and `find_package(iKin)` are equivalent.

If no components are specify here is the list of the components automatically added:
```
-- Found ICUB 1.16.0 (Components: iCubDev ctrlLib skinDynLib iKin iDyn learningMachine perceptiveModels actionPrimitives optimization)
```
(depends also if you have compiled some of them that are optional like `optimization`, `perceptiveModels [... cut]
closed 2020-06-23 by Nicogene