robotology/icub-main #661: Avoid to raise error message on iCub CAN-based robots

traversaro, 2020-06-19:
On some robots (such as iCubGenova01) the default configuration files raise errors, even if the yarprobotinterface then works fine, see for example:

~~~
1198	176,758495	ERROR	yarp.devices.ControlBoardWrapper	Parameter networks use deprecated syntax
1199	176,758645	ERROR	yarp.devices.ControlBoardWrapper	Parameter networks use deprecated syntax
1207	176,819181	ERROR	yarp.devices.ControlBoardWrapper	Parameter networks use deprecated syntax
1208	176,819221	ERROR	yarp.devices.ControlBoardWrapper	Parameter networks use deprecated syntax
1216	176,869759	ERROR	yarp.devices.ControlBoardWrapper	Parameter networks use deprecated syntax
1224	176,930320	ERROR	yarp.devices.ControlBoardWrapper	Parameter networks use deprecated syntax
1237	177,042019	ERROR		motorPwmLimit not found
1246	177,042479	ERROR		motorPwmLimit not found
1288	177,498879	ERROR		motorPwmLimit not found
1297	177,499182	ERROR		motorPwmLimit not found
1339	177,781739	ERROR		motorPwmLimit not found
1348	177,781888	ERROR		motorPwmLimit not found
1392	178,064507	ERROR		motorPwmLimit not found
1401	178,064881	ERROR		motorPwmLimit not found
1444	178,338372	ERROR		motorPwmLimit not found
1458	178,348179	ERROR		motorPwmLimit not found
1502	178,531273	ERROR		motorPwmLimit not found
1516	178,540652	ERROR		motorPwmLimit not found
1687	180,505567	ERROR	yarp.os.Port	Port /diagnostics/skin/errors:o failed to activate at tcp://10.0.0.2:10055/ (address conflict)
1692	181,043584	ERROR	yarp.os.Port	Port /diagnostics/skin/errors:o failed to activate at tcp://10.0.0.2:10055/ (address conflict)
2353	203,711748	ERROR		HeadV2_Calibrator : checkGoneToZeroThreshold: joint  4  Timeout while going to zero!
2354	203,711824	ERROR		HeadV2_Calibrator : set 3 : some axis got timeout while reaching zero position... disabling this set of axes
2357	203,721850	ERROR		sharedcan [0] board 3 FAULT OVERLOAD AXIS 0
2358	203,721894	ERROR		sharedcan [0] board 3 FAULT OVERLOAD AXIS 1
2614	238,799462	ERROR		getCurrentRangeRaw not yet implemented for CanBusMotionControl
2615	238,930830	ERROR		getCurrentRangeRaw not yet implemented for CanBusMotionControl
2616	239,062208	ERROR		getCurrentRangeRaw not yet implemented for CanBusMotionControl
2617	239,213590	ERROR		getCurrentRangeRaw not yet implemented for CanBusMotionControl
2618	239,345092	ERROR		getCurrentRangeRaw not yet implemented for CanBusMotionControl
2620	239,476537	ERROR		getCurrentRangeRaw not yet implemented for CanBusMotionControl
2626	239,759320	ERROR		getCurrentRangeRaw not yet implemented for CanBusMotionControl
2627	239,900462	ERROR		getCurrentRangeRaw not yet implemented for CanBusMotionControl
2628	240,031586	ERROR		getCurrentRangeRaw not yet implemented for CanBusMotionControl
2629	240,192956	ERROR		getCurrentRangeRaw not yet implemented for CanBusMotionControl
2630	240,414934	ERROR		getCurrentRangeRaw not yet implemented for CanBusMotionControl
2632	240,616902	ERROR		getCurrentRangeRaw not yet implemented for CanBusMotionControl
2634	240,748177	ERROR		getCurrentRangeRaw not yet implemented for CanBusMotionControl
2635	240,920024	ERROR		getCurrentRangeRaw not yet implemented for CanBusMotionControl
2637	241,121864	ERROR		getCurrentRangeRaw not yet implemented for CanBusMotionControl
2638	241,313842	ERROR		getCurrentRangeRaw not yet implemented for CanBusMotionControl
2639	241,455321	ERROR		getCurrentRangeRaw not yet implemented for CanBusMotionControl
2641	241,586453	ERROR		getCurrentRangeRaw not yet implemented for CanBusMotionControl
2643	241,717591	ERROR		getCurrentRangeRaw not yet implemented for CanBusMotionControl
2644	241,848728	ERROR		getCurrentRangeRaw not yet implemented for CanBusMotionControl
2645	241,990096	ERROR		getCurrentRangeRaw not yet implemented for CanBusMotionControl
2646	242,121280	ERROR		getCurrentRangeRaw not yet implemented for CanBusMotionControl
2657	243,646166	ERROR		getCurrentRangeRaw not yet implemented for CanBusMotionControl
2662	243,777229	ERROR		getCurrentRangeRaw not yet implemented for CanBusMotionControl
2663	243,938740	ERROR		getCurrentRangeRaw not yet implemented for CanBusMotionControl
2667	244,100179	ERROR		getCurrentRangeRaw not yet implemented for CanBusMotionControl
2668	244,292001	ERROR		getCurrentRangeRaw not yet implemented for CanBusMotionControl
2672	244,423103	ERROR		getCurrentRangeRaw not yet implemented for CanBusMotionControl
2677	244,554081	ERROR		getCurrentRangeRaw not yet implemented for CanBusMotionControl
2678	244,675101	ERROR		getCurrentRangeRaw not yet implemented for CanBusMotionControl
2684	244,806337	ERROR		getCurrentRangeRaw not yet implemented for CanBusMotionControl
2685	244,978087	ERROR		getCurrentRangeRaw not yet implemented for CanBusMotionControl
2686	245,210371	ERROR		getCurrentRangeRaw not yet implemented for CanBusMotionControl
2687	245,442460	ERROR		getCurrentRangeRaw not yet implemented for CanBusMotionControl
2689	245,674653	ERROR		getCurrentRangeRaw not yet implemented for CanBusMotionControl
2691	245,856355	ERROR		getCurrentRangeRaw not yet implemented for CanBusMotionControl
2692	246,078623	ERROR		getCurrentRangeRaw not yet implemented for CanBusMotionControl
2693	246,240335	ERROR		getCurrentRangeRaw not yet implemented for CanBusMotionControl
2700	247,511770	ERROR		getCurrentRangeRaw not yet implemented for CanBusMotionControl
2702	247,662890	ERROR		getCurrentRangeRaw not yet implemented for CanBusMotionControl
2703	247,804164	ERROR		getCurrentRangeRaw not yet implemented for CanBusMotionControl
~~~

traversaro, 2020-06-19:
I think it is important to avoid raising errors unless strictly, as otherwise we train the users to ignore the error messages, and probably the will ignore them even when they are critical  and should be instead reported to https://github.com/robotology/icub-tech-support . 

To get an  actionable definition of done, I think a reasonable target for 1.17/2020.08 is to have no error message on the iCubGenova01 on a run that is working as intended.

traversaro, 2020-06-19:
A first step in this direction is https://github.com/robotology/icub-main/pull/662 .

pattacini, 2020-06-19:
I fully agree with this approach.

traversaro, 2020-06-19:
Second related PR: https://github.com/robotology/icub-main/pull/663 .

pattacini, 2020-06-22:
Both PR have been merged 👍

closed 2020-06-22 by pattacini