robotology/icub-main #665: iCubGenova07 - Controlling the eyelids

dariopasquali, 2020-06-24:
Hi @valegagge 
we can't control the eyelids through the FacialExpression application.

## Description of the failure
Using the emotionInterface module from FacialExpression application is not possible to move the eyelids. For example, if I send "set all sad" everything changes except the eyelids.

The only way we can move the eyelids is through the motorgui.
Referring to issue robotology/icub-tech-support#775 , it is not clear if the feature is developed or not.

My supposition is that the Robot Facial Expression board either misses a connection to eyelids joint or does not control it at all.

So could you please develop the eyelids control integrated with FacialExpression application?

Thank you,
Dario

Nicogene, 2020-06-26:
First step of implementation: https://github.com/robotology/icub-main/pull/668

dariopasquali, 2020-06-30:
Hi!
Thank you for the update!
I just added a few comments in the #668 pull request regarding the `.ini` file and the `setRaw` method.

Reading robotology/funny-things#15 code, It would be great if you could implement the same RAW control also in the `emotionInterface` `setRaw` method.

Nicogene, 2020-07-02:
Hi @dariopasquali , the `setRaw` method in my opinion doesn't suit very well a module like `iCubBliker` that elaborates high-level commands.

On the other hand for me it makes sense to have it in `emotioninterface`.

Thanks for the comments in #668, all your requests should be satisfied now.

Nicogene, 2020-07-03:
The PRs have been merged, the new eyelids are slower than the old ones, probably you will notice a different behaviour, but software side we are done.

Closing for now, feel free to reopen

closed 2020-07-03 by Nicogene