robotology/icub-main #666: icub-main devel configuration fail on Debian Stable

traversaro, 2020-06-25:
See https://github.com/robotology/robotology-superbuild/runs/805730684 .  

Error log; 
~~~
2020-06-25T02:12:04.4164767Z CMake Error at src/libraries/iCubDev/CMakeLists.txt:29 (install):
2020-06-25T02:12:04.4164887Z   install TARGETS given no ARCHIVE DESTINATION for static library target
2020-06-25T02:12:04.4164996Z   "iCubDev".
2020-06-25T02:12:04.4165059Z 
2020-06-25T02:12:04.4165105Z 
2020-06-25T02:12:04.4165434Z CMake Error at /__w/robotology-superbuild/robotology-superbuild/build/install/share/YCM/modules/InstallBasicPackageFiles.cmake:670 (export):
2020-06-25T02:12:04.4165565Z   export Export set "iCubDev" not found.
2020-06-25T02:12:04.4165671Z Call Stack (most recent call first):
2020-06-25T02:12:04.4165784Z   conf/iCubInstallBasicPackageFiles.cmake:38 (install_basic_package_files)
2020-06-25T02:12:04.4165906Z   src/libraries/iCubDev/CMakeLists.txt:37 (icub_install_basic_package_files)
2020-06-25T02:12:04.4166022Z 
2020-06-25T02:12:04.4166067Z 
2020-06-25T02:12:04.4166176Z CMake Error at src/libraries/ctrlLib/CMakeLists.txt:62 (install):
2020-06-25T02:12:04.4166294Z   install TARGETS given no ARCHIVE DESTINATION for static library target
2020-06-25T02:12:04.4166404Z   "ctrlLib".
2020-06-25T02:12:04.4166464Z 
2020-06-25T02:12:04.4166508Z 
2020-06-25T02:12:04.4166836Z CMake Error at /__w/robotology-superbuild/robotology-superbuild/build/install/share/YCM/modules/InstallBasicPackageFiles.cmake:670 (export):
2020-06-25T02:12:04.4166964Z   export Export set "ctrlLib" not found.
2020-06-25T02:12:04.4167069Z Call Stack (most recent call first):
2020-06-25T02:12:04.4167181Z   conf/iCubInstallBasicPackageFiles.cmake:38 (install_basic_package_files)
2020-06-25T02:12:04.4167286Z   src/libraries/ctrlLib/CMakeLists.txt:70 (icub_install_basic_package_files)
2020-06-25T02:12:04.4167362Z 
2020-06-25T02:12:04.4167406Z 
2020-06-25T02:12:04.4167514Z CMake Error at src/libraries/skinDynLib/CMakeLists.txt:41 (install):
2020-06-25T02:12:04.4167677Z   install TARGETS given no ARCHIVE DESTINATION for static library target
2020-06-25T02:12:04.4167793Z   "skinDynLib".
2020-06-25T02:12:04.4167841Z 
2020-06-25T02:12:04.4167902Z 
2020-06-25T02:12:04.4168230Z CMake Error at /__w/robotology-superbuild/robotology-superbuild/build/install/share/YCM/modules/InstallBasicPackageFiles.cmake:670 (export):
2020-06-25T02:12:04.4168354Z   export Export set "skinDynLib" not found.
2020-06-25T02:12:04.4168460Z Call Stack (most recent call first):
2020-06-25T02:12:04.4168557Z   conf/iCubInstallBasicPackageFiles.cmake:38 (install_basic_package_files)
2020-06-25T02:12:04.4168679Z   src/libraries/skinDynLib/CMakeLists.txt:49 (icub_install_basic_package_files)
2020-06-25T02:12:04.4168758Z 
2020-06-25T02:12:04.4168804Z 
2020-06-25T02:12:04.4168908Z CMake Error at src/libraries/iKin/CMakeLists.txt:60 (install):
2020-06-25T02:12:04.4169023Z   install TARGETS given no ARCHIVE DESTINATION for static library target
2020-06-25T02:12:04.4169135Z   "iKin".
2020-06-25T02:12:04.4169182Z 
2020-06-25T02:12:04.4169241Z 
2020-06-25T02:12:04.4169566Z CMake Error at /__w/robotology-superbuild/robotology-superbuild/build/install/share/YCM/modules/InstallBasicPackageFiles.cmake:670 (export):
2020-06-25T02:12:04.4169693Z   export Export set "iKin" not found.
2020-06-25T02:12:04.4169785Z Call Stack (most recent call first):
2020-06-25T02:12:04.4169899Z   conf/iCubInstallBasicPackageFiles.cmake:38 (install_basic_package_files)
2020-06-25T02:12:04.4170020Z   src/libraries/iKin/CMakeLists.txt:68 (icub_install_basic_package_files)
2020-06-25T02:12:04.4170093Z 
2020-06-25T02:12:04.4170139Z 
2020-06-25T02:12:04.4170245Z CMake Error at src/libraries/iDyn/CMakeLists.txt:32 (install):
2020-06-25T02:12:04.4170359Z   install TARGETS given no ARCHIVE DESTINATION for static library target
2020-06-25T02:12:04.4170502Z   "iDyn".
2020-06-25T02:12:04.4170551Z 
2020-06-25T02:12:04.4170610Z 
2020-06-25T02:12:04.4170940Z CMake Error at /__w/robotology-superbuild/robotology-superbuild/build/install/share/YCM/modules/InstallBasicPackageFiles.cmake:670 (export):
2020-06-25T02:12:04.4171057Z   export Export set "iDyn" not found.
2020-06-25T02:12:04.4171161Z Call Stack (most recent call first):
2020-06-25T02:12:04.4171273Z   conf/iCubInstallBasicPackageFiles.cmake:38 (install_basic_package_files)
2020-06-25T02:12:04.4171394Z   src/libraries/iDyn/CMakeLists.txt:40 (icub_install_basic_package_files)
2020-06-25T02:12:04.4171467Z 
2020-06-25T02:12:04.4171512Z 
2020-06-25T02:12:04.4171845Z -- Found YARP: /__w/robotology-superbuild/robotology-superbuild/build/install/lib/cmake/YARP (found version "3.3.103+20200623.4+git1ae7811f3")
2020-06-25T02:12:04.4171979Z CMake Error at src/libraries/learningMachine/CMakeLists.txt:75 (install):
2020-06-25T02:12:04.4172100Z   install TARGETS given no ARCHIVE DESTINATION for static library target
2020-06-25T02:12:04.4172210Z   "learningMachine".
2020-06-25T02:12:04.4172261Z 
2020-06-25T02:12:04.4172382Z 
[... 74 lines cut ...]
2020-06-25T02:12:04.4232494Z 
2020-06-25T02:12:04.4232830Z CMake Error at /__w/robotology-superbuild/robotology-superbuild/build/install/share/YCM/modules/InstallBasicPackageFiles.cmake:670 (export):
2020-06-25T02:12:04.4232963Z   export Export set "optimization" not found.
2020-06-25T02:12:04.4233068Z Call Stack (most recent call first):
2020-06-25T02:12:04.4233169Z   conf/iCubInstallBasicPackageFiles.cmake:38 (install_basic_package_files)
2020-06-25T02:12:04.4233292Z   src/libraries/optimization/CMakeLists.txt:46 (icub_install_basic_package_files)
2020-06-25T02:12:04.4233374Z 
2020-06-25T02:12:04.4233420Z 
2020-06-25T02:12:04.4233640Z --  +++ tool compiling canLoaderLib
2020-06-25T02:12:04.4233869Z --  +++ tool compiling canLoader-console
2020-06-25T02:12:04.4234093Z --  +++ tool compiling ethLoaderLib
2020-06-25T02:12:04.4234318Z --  +++ tool compiling strainCalibLib
2020-06-25T02:12:04.4234657Z -- Found YARP: /__w/robotology-superbuild/robotology-superbuild/build/install/lib/cmake/YARP (found version "3.3.103+20200623.4+git1ae7811f3")
2020-06-25T02:12:04.4235010Z -- Found YARP: /__w/robotology-superbuild/robotology-superbuild/build/install/lib/cmake/YARP (found version "3.3.103+20200623.4+git1ae7811f3")
2020-06-25T02:12:04.4235361Z -- Found YARP: /__w/robotology-superbuild/robotology-superbuild/build/install/lib/cmake/YARP (found version "3.3.103+20200623.4+git1ae7811f3")
2020-06-25T02:12:04.4235690Z -- Found YARP: /__w/robotology-superbuild/robotology-superbuild/build/install/lib/cmake/YARP (found version "3.3.103+20200623.4+git1ae7811f3")
2020-06-25T02:12:04.4236194Z -- Found YARP: /__w/robotology-superbuild/robotology-superbuild/build/install/lib/cmake/YARP (found version "3.3.103+20200623.4+git1ae7811f3")
2020-06-25T02:12:04.4236539Z -- Found YARP: /__w/robotology-superbuild/robotology-superbuild/build/install/lib/cmake/YARP (found version "3.3.103+20200623.4+git1ae7811f3")
2020-06-25T02:12:04.4236891Z -- Found YARP: /__w/robotology-superbuild/robotology-superbuild/build/install/lib/cmake/YARP (found version "3.3.103+20200623.4+git1ae7811f3")
2020-06-25T02:12:04.4237242Z -- Found YARP: /__w/robotology-superbuild/robotology-superbuild/build/install/lib/cmake/YARP (found version "3.3.103+20200623.4+git1ae7811f3")
2020-06-25T02:12:04.4237475Z -- Installing cmake scripts
2020-06-25T02:12:04.4237709Z -- Configuring incomplete, errors occurred!
~~~

On Debian Buster the CMake  used is 3.13 (see https://packages.debian.org/buster/cmake) so this could be relate to that.

traversaro, 2020-06-25:
The problem is probably in all the `install` invocation in https://github.com/robotology/icub-main/pull/652/files#diff-743c289c8bd401436bb8cf793fb84942R62 that lack specification for `RUNTIME DESTINATION` and `ARCHIVE DESTINATION` (see https://github.com/robotology/how-to-export-cpp-library/blob/master/src/LibTemplateCMake/CMakeLists.txt#L86). This is not a problem in CMake >= 3.14 as in that case a default for that argument has been specified (see https://cmake.org/cmake/help/v3.14/command/install.html#installing-targets), but it is compulsory to set  `RUNTIME DESTINATION` and `ARCHIVE DESTINATION` to remain compatible with CMake 3.13 .

pattacini, 2020-06-25:
@Nicogene fixed this in #667 👍

PR #667 (2020-06-25): Fix components on debian buster and better handling of internal dependencies
This PR fixes #666 

Moreover the internal dependencies have been handled in a better way.

Successfully tested on Ubuntu 20.04, on both build and install tree.

Please review code.
commit 95f34df8 (2020-06-25): COMPONENTS: fix the configuration with CMake < 3.14

It fixes #666
closed 2020-06-25 by pattacini