robotology/icub-main #678: icub-main fails to compile on Windows 10 using Visual Studio 16 2019

S-Dafarra, 2020-08-31:
I tried to install it on a clean Windows 10 installation using Visual Studio 16 2019
Here the errors:
```
Severity	Code	Description	Project	File	Line	Suppression State
Error	LNK2019	unresolved external symbol main referenced in function "int __cdecl invoke_main(void)" (?invoke_main@@YAHXZ)	joystickCheck	C:\robotology-superbuild\build\robotology\ICUB\src\tools\joystickCheck\MSVCRTD.lib(exe_main.obj)	1	
Error	LNK1120	1 unresolved externals	joystickCheck	C:\robotology-superbuild\build\robotology\ICUB\bin\Debug\joystickCheck.exe	1	
Error	LNK2019	unresolved external symbol main referenced in function "int __cdecl invoke_main(void)" (?invoke_main@@YAHXZ)	joystickCtrl	C:\robotology-superbuild\build\robotology\ICUB\src\tools\joystickCtrl\MSVCRTD.lib(exe_main.obj)	1	
Error	LNK1120	1 unresolved externals	joystickCtrl	C:\robotology-superbuild\build\robotology\ICUB\bin\Debug\joystickCtrl.exe	1	
Error	LNK2019	unresolved external symbol MAIN__ referenced in function main	iCub_SIM	C:\robotology-superbuild\build\robotology\ICUB\src\simulators\iCubSimulation\libifcoremdd.lib(for_main.obj)	1	
Error	LNK1120	1 unresolved externals	iCub_SIM	C:\robotology-superbuild\build\robotology\ICUB\bin\Debug\iCub_SIM.exe	1	
```

Here the complete build:
```
1>------ Build started: Project: ZERO_CHECK, Configuration: Debug x64 ------
1>Checking Build System
2>------ Build started: Project: ctrlLib, Configuration: Debug x64 ------
3>------ Build started: Project: learningMachine, Configuration: Debug x64 ------
4>------ Build started: Project: iCubDev, Configuration: Debug x64 ------
5>------ Build started: Project: QtICubSkinGuiPlugin, Configuration: Debug x64 ------
6>------ Build started: Project: camCalib, Configuration: Debug x64 ------
7>------ Build started: Project: camCalibWithPose, Configuration: Debug x64 ------
8>------ Build started: Project: cartesianInterfaceControl, Configuration: Debug x64 ------
9>------ Build started: Project: dualCamCalib, Configuration: Debug x64 ------
10>------ Build started: Project: emotionInterface, Configuration: Debug x64 ------
11>------ Build started: Project: frameGrabberGui2, Configuration: Debug x64 ------
12>------ Build started: Project: gazecontrollerclient, Configuration: Debug x64 ------
13>------ Build started: Project: iCubSkinGui, Configuration: Debug x64 ------
5>Automatic MOC for target QtICubSkinGuiPlugin
3>Building Custom Rule C:/robotology-superbuild/robotology/ICUB/src/libraries/learningMachine/CMakeLists.txt
2>Building Custom Rule C:/robotology-superbuild/robotology/ICUB/src/libraries/ctrlLib/CMakeLists.txt
4>Building Custom Rule C:/robotology-superbuild/robotology/ICUB/src/libraries/iCubDev/CMakeLists.txt
7>Building Custom Rule C:/robotology-superbuild/robotology/ICUB/src/modules/camCalibWithPose/CMakeLists.txt
6>Building Custom Rule C:/robotology-superbuild/robotology/ICUB/src/modules/camCalib/CMakeLists.txt
8>Building Custom Rule C:/robotology-superbuild/robotology/ICUB/src/modules/cartesianInterfaceControl/CMakeLists.txt
11>Automatic MOC for target frameGrabberGui2
9>Building Custom Rule C:/robotology-superbuild/robotology/ICUB/src/modules/dualCamCalib/CMakeLists.txt
13>Automatic MOC for target iCubSkinGui
12>Building Custom Rule C:/robotology-superbuild/robotology/ICUB/src/libraries/icubmod/gazeController/CMakeLists.txt
10>Building Custom Rule C:/robotology-superbuild/robotology/ICUB/src/core/emotionInterface/CMakeLists.txt
7>main.cpp
8>main.cpp
6>main.cpp
4>DebugInterfaces.cpp
2>math.cpp
3>DatasetRecorder.cpp
9>main.cpp
10>emotionInterfaceModule.cpp
8>C:\robotology-superbuild\build\install\include\yarp/dev/CartesianControl.h(49,21): warning C4251: 'yarp::dev::CartesianEventParameters::type': class 'std::basic_string<char,std::char_traits<char>,std::allocator<char>>' needs to have dll-interface to be used by clients of struct 'yarp::dev::CartesianEventParameters'
8>C:\Program Files (x86)\Microsoft Visual Studio\2019\Enterprise\VC\Tools\MSVC\14.26.28801\include\xstring(4575): message : see declaration of 'std::basic_string<char,std::char_traits<char>,std::allocator<char>>'
8>C:\robotology-superbuild\build\install\include\yarp/dev/CartesianControl.h(71,21): warning C4251: 'yarp::dev::CartesianEventVariables::type': class 'std::basic_string<char,std::char_traits<char>,std::allocator<char>>' needs to have dll-interface to be used by clients of struct 'yarp::dev::CartesianEventVariables'
8>C:\Program Files (x86)\Microsoft Visual Studio\2019\Enterprise\VC\Tools\MSVC\14.26.28801\include\xstring(4575): message : see declaration of 'std::basic_string<char,std::char_traits<char>,std::allocator<char>>'
13>Generating qrc_qml.cpp
12>ClientGazeController.cpp
3>C:\robotology-superbuild\robotology\ICUB\src\libraries\learningMachine\include\iCub/learningMachine/Prediction.h(98,1): warning C4267: 'return': conversion from 'size_t' to 'unsigned int', possible loss of data
2>filters.cpp
13>Building Custom Rule C:/robotology-superbuild/robotology/ICUB/src/tools/iCubSkinGui/src/CMakeLists.txt
3>DummyLearner.cpp
11>Generating qrc_res.cpp
5>Building Custom Rule C:/robotology-superbuild/robotology/ICUB/src/tools/iCubSkinGui/plugin/CMakeLists.txt
4>iCubDev.vcxproj -> C:\robotology-superbuild\build\robotology\ICUB\lib\Debug\iCubDevd.lib
10>C:\robotology-superbuild\robotology\ICUB\src\core\emotionInterface\emotionInterfaceModule.cpp(263,71): warning C4267: 'argument': conversion from 'size_t' to 'const int', possible loss of data
14>------ Build started: Project: controlBoardDumper, Configuration: Debug x64 ------
11>Generating ui_dc1394slider.h
10>C:\robotology-superbuild\robotology\ICUB\src\core\emotionInterface\emotionInterfaceModule.cpp(265,66): warning C4267: 'argument': conversion from 'size_t' to 'int', possible loss of data
10>C:\robotology-superbuild\robotology\ICUB\src\core\emotionInterface\emotionInterfaceModule.cpp(269,57): warning C4267: 'argument': conversion from 'size_t' to 'int', possible loss of data
10>C:\robotology-superbuild\robotology\ICUB\src\core\emotionInterface\emotionInterfaceModule.cpp(270,93): warning C4267: 'argument': conversion from 'size_t' to 'int', possible loss of data
10>C:\robotology-superbuild\robotology\ICUB\src\core\emotionInterface\emotionInterfaceModule.cpp(555,60): warning C4267: 'argument': conversion from 'size_t' to 'int', possible loss of data
10>C:\robotology-superbuild\robotology\ICUB\src\core\emotionInterface\emotionInterfaceModule.cpp(583,60): warning C4267: 'argument': conversion from 'size_t' to 'int', possible loss of data
2>kalman.cpp
[... 1308 lines cut ...]
44>libipoptD.lib(dorgtr.obj) : warning LNK4099: PDB 'vc140.pdb' was not found with 'libipoptD.lib(dorgtr.obj)' or at 'C:\robotology-superbuild\build\robotology\ICUB\bin\Debug\vc140.pdb'; linking object as if no debug info
44>libipoptD.lib(dsteqr.obj) : warning LNK4099: PDB 'vc140.pdb' was not found with 'libipoptD.lib(dsteqr.obj)' or at 'C:\robotology-superbuild\build\robotology\ICUB\bin\Debug\vc140.pdb'; linking object as if no debug info
44>libipoptD.lib(dgetf2.obj) : warning LNK4099: PDB 'vc140.pdb' was not found with 'libipoptD.lib(dgetf2.obj)' or at 'C:\robotology-superbuild\build\robotology\ICUB\bin\Debug\vc140.pdb'; linking object as if no debug info
44>libipoptD.lib(dlaswp.obj) : warning LNK4099: PDB 'vc140.pdb' was not found with 'libipoptD.lib(dlaswp.obj)' or at 'C:\robotology-superbuild\build\robotology\ICUB\bin\Debug\vc140.pdb'; linking object as if no debug info
44>libipoptD.lib(dpptrf.obj) : warning LNK4099: PDB 'vc140.pdb' was not found with 'libipoptD.lib(dpptrf.obj)' or at 'C:\robotology-superbuild\build\robotology\ICUB\bin\Debug\vc140.pdb'; linking object as if no debug info
44>libipoptD.lib(dpptrs.obj) : warning LNK4099: PDB 'vc140.pdb' was not found with 'libipoptD.lib(dpptrs.obj)' or at 'C:\robotology-superbuild\build\robotology\ICUB\bin\Debug\vc140.pdb'; linking object as if no debug info
44>libipoptD.lib(mumps_sol_es.obj) : warning LNK4099: PDB 'vc140.pdb' was not found with 'libipoptD.lib(mumps_sol_es.obj)' or at 'C:\robotology-superbuild\build\robotology\ICUB\bin\Debug\vc140.pdb'; linking object as if no debug info
44>libipoptD.lib(dtrsv.obj) : warning LNK4099: PDB 'vc140.pdb' was not found with 'libipoptD.lib(dtrsv.obj)' or at 'C:\robotology-superbuild\build\robotology\ICUB\bin\Debug\vc140.pdb'; linking object as if no debug info
[... cut ...]
44>libipoptD.lib(dlaset.obj) : warning LNK4099: PDB 'vc140.pdb' was not found with 'libipoptD.lib(dlaset.obj)' or at 'C:\robotology-superbuild\build\robotology\ICUB\bin\Debug\vc140.pdb'; linking object as if no debug info
44>libipoptD.lib(dlaev2.obj) : warning LNK4099: PDB 'vc140.pdb' was not found with 'libipoptD.lib(dlaev2.obj)' or at 'C:\robotology-superbuild\build\robotology\ICUB\bin\Debug\vc140.pdb'; linking object as if no debug info
44>libipoptD.lib(dlasr.obj) : warning LNK4099: PDB 'vc140.pdb' was not found with 'libipoptD.lib(dlasr.obj)' or at 'C:\robotology-superbuild\build\robotology

traversaro, 2020-08-31:
I remember having a similar issue in https://github.com/robotology/robotology-superbuild/issues/128#issuecomment-433695438 . Can you paste the `CMakeCache.txt` of `ICUB`?

S-Dafarra, 2020-08-31:
It's a 1528 lines long file. So I rather attach it as a file.
Here it is! [CMakeCache.txt](https://github.com/robotology/icub-main/files/5150276/CMakeCache.txt)

traversaro, 2020-08-31:
> 
> 
> It's a 1528 lines long file. So I rather attach it as a file.
> Here it is! [CMakeCache.txt](https://github.com/robotology/icub-main/files/5150276/CMakeCache.txt)

The found SDL seems to be indeed the vcpkg one, that is working fine in CI (see https://github.com/robotology/robotology-superbuild/runs/1049179020). If you don't need those tools you can simply set `ICUB_USE_SDL` to `FALSE` . Otherwise, we need to understand what is going on. I guess there is some SDL library that is providing the `main` symbol, and for some reason it is not linked.

S-Dafarra, 2020-09-02:
After setting ``ICUB_USE_SDL`` to ``FALSE``, it compiles successfully.

S-Dafarra, 2020-09-02:
This was probably due to a mistake of mine in dealing with the dependencies as I did not follow the [README](https://github.com/robotology/robotology-superbuild-dependencies-vcpkg#install-or-remove-vcpkg-ports) correctly. By reinstalling the dependencies and the ``robotology-superbuild`` I don't have this issue anymore.

closed 2020-09-02 by S-Dafarra