robotology/icub-main #685: Remove firewirebusreset utility?

traversaro, 2020-10-10:
While trying to build icub-main on conda-forge (https://conda-forge.org/) and by doing a cleanup of the apt dependencies of the superbuild (see https://github.com/robotology/robotology-superbuild/pull/483), I noticed that the `dragonfly2` device does not depend only on the [`libdc1394` library](https://damien.douxchamps.net/ieee1394/libdc1394/), as I always assumed and documented in the robotology-superbuild, but also on the much more lower level [libraw1394](https://ieee1394.wiki.kernel.org/index.php/Libraries). However, `libraw1394` is not actually used in the `dragonfly2` nor in the `dragonfly2raw` device, but rather just in a `firewirebusreset` executable tool. 

For this, I wanted to ask: 
* Is the  `firewirebusreset` tool used at all?  I tag a few users that may be aware of this: @pattacini @ale-git @davidetome  @julijenv @violadelbono @vtikha @randaz81 
* If the `firewirebusreset` tool is not used, could we remove it from the repo and drop the dependency on `libraw1394` ? This would simplify the compilation of the `dragonfly2` device.

pattacini, 2020-10-10:
The author is @ale-git (see https://github.com/robotology/icub-main/blame/master/src/libraries/icubmod/dragonfly2/linux/fwBusReset_main.cpp).

ale-git, 2020-10-12:
It is a very very old relic, written maybe by Lorenzo Natale many years ago. It can be deleted in my opinion.

violadelbono, 2020-10-12:
@traversaro we don't use that tool and I don't think we need it, so also for me it can be removed.

traversaro, 2020-10-12:
Thanks a lot for the feedback @violadelbono @ale-git !

closed 2020-10-12 by pattacini