robotology/icub-main #691: Unable to load ecan YARP device on Windows

traversaro, 2020-11-19:
**Describe the bug**
The `ecan` plugin installed by the ICUB option `ENABLE_icubmod_ecan` does not work correctly, and it is not shown correctly by the `yarpdev --list` on Windows.

**To Reproduce**
Install ICUB and enable the usbCamera device. Then, run `yarpdev --list`, you get:
~~~
2020-11-19T17:00:54.3549673Z [ERROR] |yarp.os.YarpPluginSettings| Error while opening C:/robotology/robotology/lib/iCub/esdCan.dll:
2020-11-19T17:00:54.3551267Z   The specified module could not be found.
2020-11-19T17:00:54.3551959Z 
2020-11-19T17:00:54.3552366Z 
~~~

**Expected behavior**
I would expect `yarpdev --list`  to work correctly as it does for all the other YARP devices installed by ICUB.

**Configuration (please complete the following information):**
 - OS: Windows 10
 - ICUB version: both master, devel and 1.17.0 release 
-  compiler: Visual Studio 2019

**Additional context**
I discovered this in https://github.com/robotology/robotology-superbuild/pull/521/checks?check_run_id=1425069998 .

Similar issues: 
* https://github.com/robotology/wearables/issues/91
* https://github.com/robotology/yarp/issues/2414

randaz81, 2020-11-19:
My first question is: is both the SDK and the windows device driver (.dll) correctly installed? 
The message `Error while opening C:/robotology/robotology/lib/iCub/esdCan.dll` may be misleading, because the file may exists, but I think the dll will probably try open internally the dll of the device driver, which needs to be in the system path or in `C:\Windows\System32` or equivalent.

traversaro, 2020-11-19:
The windows device driver is installed and on the path, on the other hand the SDK for sure is not installed in the CI, I can try on my PC where the device is installed to check if it works.