robotology/icub-main #694: [iKinFwd] Possible wrong handling of *_hand_finger (finger adduction/abduction) readings in getChainJoints()

xEnVrE, 2020-11-27:
I would like to rise an issue regarding how the readings of the joint `*_hand_finger` (i.e. hand finger adduction/abduction) are handled inside the `iKinFwd::iCubFinger::getChainJoints()` methods.

Namely, given the fact that on a real iCub robot setting this joint to `0` means moving the fingers apart and moving it to `60.0` means closing the fingers I think that the following lines might be wrong:

https://github.com/robotology/icub-main/blob/c42639c003c28b4a0c9b746a9b4d1bfbf3f90376/src/libraries/iKin/src/iKinFwd.cpp#L2117

https://github.com/robotology/icub-main/blob/c42639c003c28b4a0c9b746a9b4d1bfbf3f90376/src/libraries/iKin/src/iKinFwd.cpp#L2132

and the associated ones for the alternative method `bool iCubFinger::getChainJoints(const Vector &motorEncoders,
                                const Vector &jointEncoders,
                                Vector &chainJoints,
                                const Matrix &jointEncodersBounds)`.

Ideally, when the fingers are closed, i.e. `motorEncoders[offs+0] == 60.0` (or the maximum value achievable on the robot), the joint angle should be `chainJoints[0] = (60.0 - motorEncoders[offs+0]) / 3.0 = (60.0 - 60.0) / 3.0 = 0.0`. On the other side, when the fingers are far apart, i.e. `motorEncoders[offs+0] == 0`, we would like to have `chainJoints[0] = (60.0 - motorEncoders[offs+0]) / 3.0 = (60.0 - 0.0) / 3.0 = 20.0`. However, it seems that the code is missing the `60 degrees` offset.

Using the **current `iKinFwd`** and superimposing the hand meshes over the image plane we get (please ignore that some distal joints are wrong, this depend probably on hall effect sensors with a wrong calibration)

[image]

Instead, if we put an offset of `20 degrees` (why not 60.0 you might ask, I answer to this later) we get this more coherent result:

[image]

The issue here is that, although the ideal maximum limit of the joint is `60.0` degress, the actual one we might have on each robot depends on the actual mechanics. For example,  on `iCubGenova01` the software limit is set to `60.0` degrees but the joint cannot go beyond `20.0` degress, hence we should put that as the offset. Ideally, we might decide to 1) fix the joint limits in the robot configuration according to the actual range of motion of the robot 2) recruit that limit inside `iKinFwd` in order to decide the maximum offset we need to compensate for.

I would like to know your opinion on that :smile: 

@pattacini

pattacini, 2020-11-27:
So, to recap:
1. the login open/close is inverted in `iKin` (that's a bug)
2. the bound of the interval does not correspond to the real physical range

Is that correct?

If so, we can fix 1 in `iKin`, whereas for 2 the correct place is to update the conf files.

xEnVrE, 2020-11-27:
> Is that correct?

Yes, that is correct.

As discussed f2f, I will proceed by opening a PR. We use the bounds from the robot (available via the `iKinLink::getMin()/getMax()` methods) to find the proper offset. These limits will be coherent with those of robot if the user align to them using the `iCubFinger::alignJointBounds()` method beforehand.

xEnVrE, 2020-12-01:
Today I assisted during the calibration procedure of the joint responsible for adduction/abduction. Even if I can't find any document specifying that the actual range of motion of this group of joints is 60.0 degrees, the calibration procedure is such that
- the reading 60.0 is assigned to the configuration where the fingers are all closed
- the reading 0.0 is assigned to the configuration where the fingers are totally far apart

Given this, I must re-consider what I wrote in the first commit. Namely, given that 60.0 is considered as the `zero` reference position, the code should in any case be 

```
chainJoints[0] = (60.0 - motorEncoders[offs+0]) / 3.0 
```

independently of the soft limits set in the robot configuration.

On the other hand, once we agree on the fact that the canonical range of motion is 60.0 degrees, the actual limits of the 0-th link of fingers `index`, `ring` and `little` set in the method `iCubFinger::alignJointsBounds` should be changed to take into account possible soft limits set in the robot configuration. 

I would change from 

https://github.com/robotology/icub-main/blob/e030379c948d347f96a942058c4f5269234a4116/src/libraries/iKin/src/iKinFwd.cpp#L2042-L2044

to 

```
 (*this)[0].setMin(CTRL_DEG2RAD*(60.0-max)/3.0);
 (*this)[0].setMax(CTRL_DEG2RAD*(60.0-min)/3.0);
```
where the associations set**Min** -> 60.0-**max** and set**Max** -> 60.0 - **min** depend on the reversed open/close logic.

Tests on an iCub robot, calibrated using the aforementioned convention, follows:

[Video showing the behavior with the current iKin](https://youtu.be/g2Mu9jLGqVI)

[Video showing the behavior with the proposed fix](https://youtu.be/qZtxrSx55gk)
> In the second video it is visible the fact that dividing the reading by 3 is not necessarily accurate when the fingers are too far apart (especially for ring and little).

pattacini, 2020-12-02:
After a quick T2T, we agreed on doing a further investigation to see if the hard limit of 60 deg can be removed from iKin, which is not aware of any difference between hard/soft bounds on its own. By contrast, the latter difference should be located at the conf files of the specific robot.

xEnVrE, 2020-12-02:
We agreed on the following solution:
1. Given that the minimum `m` and maximum `M` for `*_hand_finger` are available (either from `iCubFinger::alignJointsBounds()` or using  the standard limits of the chain specified in `iCubFinger::allocate()`) we assume that the fingers are totally closed when the encoder reading == M and that they are not closed when the encoder reading == m (i.e. the angle between the index and the little fingers is (M-m))
2. As consequence the actual limits of the 0-th link of fingers index, ring and little are set in the method `iCubFinger::alignJointsBounds` as
   ```
    (*this)[0].setMin(0.0);
    (*this)[0].setMax(CTRL_DEG2RAD*(M-m)/3.0);
   ```
3. And the reverted reading to be used in `getChainJoints()` methods is as follows:
   ```
   chainJoints[0] = (M - motorEncoders[offs+0]) / 3.0 
   ```

closed 2020-12-02 by pattacini