robotology/icub-main #712: embObjMotionControl: issue in CONTROL configuration parsing

lrapetti, 2021-01-07:
While changing the parameters in the `CONTROL` group configuration I found the following issue which might lead to problem not easy to spot (as it was in my experience).

The `CONTROL` group usually is something like:
```xml
<group name="CONTROLS">
        <param name="positionControl">          POS_PID_DEFAULT_1     POS_PID_DEFAULT_2 </param>
        <param name="velocityControl">          POS_PID_DEFAULT_1     POS_PID_DEFAULT_2 </param>
        <param name="mixedControl">             POS_PID_DEFAULT_1     POS_PID_DEFAULT_2 </param>
        <param name="torqueControl">            TRQ_PID_DEFAULT_1     TRQ_PID_DEFAULT_2 </param>
        <param name="currentPid">               2FOC_CUR_CONTROL_1    2FOC_CUR_CONTROL_2 </param>
        <param name="speedPid">                 2FOC_VEL_CONTROL_1    2FOC_VEL_CONTROL_2 </param>
</group>
```
this means that we are defining a set of gains for each joint (each column) as defined in the given `POS_PID_DEFAULT_{i}` group.
The `POS_PID_DEFAULT_{i}` group is something like:
```xml
group name="POS_PID_DEFAULT_{i}">
        <param name="controlLaw">           minjerk             </param>
        <param name="outputType">           pwm                 </param>
        <param name="fbkControlUnits">      metric_units        </param>
        <param name="outputControlUnits">   machine_units       </param>
        <param name="kp">                   711.11      1066.66 </param>
        <param name="kd">                   0.00        0.00 </param>
        <param name="ki">                   7111.09     10666.64 </param>
        <param name="maxOutput">            8000        8000 </param>
        <param name="maxInt">               200         200 </param>
        <param name="stictionUp">           0           0 </param>
        <param name="stictionDown">         0           0 </param>
        <param name="kff">                  0           0 </param>
</group>
```
as you can see, it contains the gains for both the two joints and depending on which column of the `CONTROLS` group this configuration is used, the gains of the corresponding column will be used.

Up to now everything is fine. However, you can see that some of the configuration parameters (e.g. `controlLaw`, `outputType`, `fbkControlUnits`) are defined only once for all the joints. What happen if the value of those parameter is different in the different columns?
What I found it is that the one of the first configuration is going to be used (in this case `POS_PID_DEFAULT_1`). It follows that for example if the first configuration uses `machine_units` and the second in `metric_units`, the gains for the joint 2 will be anyway considered being in `machine_units`, and indeed they would be wrong.

pattacini, 2021-01-07:
My 2 cents on this.

Properties like `controlLaw`, `outputType`... need to be shared homogeneously within the same `POS_PID_DEFAULT_{i}` group. Therefore, if you aim to have heterogeneous quantities among different joints, you should make each joint point to a specific (dedicated/different) group.

Did you try creating dedicated groups for different joints?

If this understanding is correct, it ain't very much clear or emphasized anyway, I don't know though.
@randaz81 can chime in and give us heads-up.

lrapetti, 2021-01-07:
> Did you try creating dedicated groups for different joints?

That is actually what I did, if a dedicated group is defined for each column, the common parameter (`controlLaw`, `outputType`, `fbkControlUnits`) of the first group are used anyway for all the groups.

pattacini, 2021-01-07:
Sorry, then I didn't get it correctly in the first place.

So, to recap, if the conf file is something like the following

```xml
<group name="CONTROLS">
        <param name="positionControl">POS_PID_DEFAULT_1 POS_PID_DEFAULT_2</param>
        ...
</group>
group name="POS_PID_DEFAULT_1">
        ...
        <param name="fbkControlUnits"> metric_units </param>
        ...
</group>

group name="POS_PID_DEFAULT_2">
        ...
        <param name="fbkControlUnits"> machine_units </param>
        ...
</group>
```

then, the second joint's gains will be still in metric units instead of machine units?

lrapetti, 2021-01-07:
> Sorry, then I didn't get it correctly in the first place.
> 
> So, to recap, if the conf file is something like the following
> 
> ```
> <group name="CONTROLS">
>   <param name="positionControl">POS_PID_DEFAULT_1 POS_PID_DEFAULT_2</param>
>    ...
> </group>
> 
> <group name="POS_PID_DEFAULT_1">
>    ...
>   <param name="fbkControlUnits"> metric_units </param>
>    ...
> </group>
> 
> <group name="POS_PID_DEFAULT_2">
>   ...
>   <param name="fbkControlUnits"> machine_units </param>
>   ...
> </group>
> ```
> 
> then, the second joint's gains will be still in metric units instead of machine units?

Yes, exactly, the second joint's gains will be assume to be in `metric_units`.
Sorry if it was not clear.

pattacini, 2021-01-07:
>Sorry if it was not clear.

Never mind!
So, we'd need to investigate it.
@randaz81 any clue, in the meantime?

randaz81, 2021-01-14:
Indeed. I do not remember the reason nor the details, but it is something that is explicitly forbidden in the code. 
Assuming that you modify the parser where I put the comment here:
```
tpids[i].pid = torqueAlgo_ptr->getPID(i);
tpids[i].fbk_PidUnits = torqueAlgo_ptr->fbk_PidUnits;    //you can notice here that you do not have a [i] index for the units
```
then, later, you have a sanity check:
```
if(tpids[firstjoint].fbk_PidUnits != tpids[i].fbk_PidUnits || tpids[firstjoint].out_PidUnits != tpids[i].out_PidUnits)
{
    yError() << "embObjMC BOARD " << _boardname << "all joints with torque enabled should have same controlunits type. Joint " << firstjoint << " differs from joint " << i;
    return false;
}
```
I do not remember if this parameter is used only just by yarp and the high level motioncontroller or if it is propagated to the  hardware (the control board). The latter case is obviously much more complicated, and any modification will require a revision of the communication protocol. @ale-git what do you think?

lrapetti, 2021-01-14:
> Indeed. I do not remember the reason nor the details, but it is something that is explicitly forbidden in the code.
> Assuming that you modify the parser where I put the comment here:
>
> ``` [...]

I think the reason why it is forbidden might be related to the presence of the coupled joints.

The easiest fix, if not possible to change it, it would be to trough an error in case a configuration with different units (like the one I tested) is parsed.

Otherwise, in case different unit of measurements are present, the gains values should be changed in order to be expressed accordingly to the first configuration unit of measurement.

randaz81, 2021-01-14:
But my question is:
generally speaking, you should not use machine units at all. What is  your use case?
Because these parameters are used mainly for backward compatibility and (ideally) should be removed in the long term.
removing -> no need to change anything in the parser.

In this hypothesis, I'd not change anything in the current parser. 
Maybe I'd add just only an error message like the one I quoted above for the torque pid. 
If you don't see an error message for another type of control (e.g. position), then I consider it a bug, and I think it should be added....

lrapetti, 2021-01-14:
> generally speaking, you should not use machine units at all. What is your use case?

It was not a specific use-case, but I ended up in this problem while I was adding a configuration for a single joint, and leaving the others to `none`.
```xml
<group name="CONTROLS">
  <param name="torqueControl">none TRQ_PID_DEFAULT</param>
   ...
</group>
```
it seems that the default (`none`) configuration is in `machine_units`

> Maybe I'd add just only an error message like the one I quoted above for the torque pid.

Yes, I think that error message should be thrown for all those type of configurations.

pattacini, 2021-01-25:
We'll be dealing w/ this starting in PI06-sprint01 (next week basically).

closed 2021-02-10 by pattacini