robotology/icub-main #715: Missing pressure output from the skinDynLib

2103simon, 2021-01-25:
I am working on a skin plugin for gazebo and had problems with transmitting force values at the pressure position. After investigating it a bit closer I realized the following part in the skinContact.cpp:
https://github.com/robotology/icub-main/blob/464f592c22de531202f028ad74253ae9cb2d71f0/src/libraries/skinDynLib/src/skinContact.cpp#L76-L81

After changing it to:
bool skinContact::setPressure(double _pressure){
    if(_pressure<=0.0)
        return false;
    pressure = _pressure;
    return true;
}

it works just fine. Is there any specific reason to compare to the with 0.0 initialized variable, instead of the handed-over one?

pattacini, 2021-01-25:
It stinks like a bug, actually 🐛
Unless the aim was to put the pressure < 0 to signal an error persistently.
If @simeonedussoni and @Nicogene have nothing against it, I'll fix it up.

Thanks @2103simon

pattacini, 2021-01-25:
@2103simon out of sheer curiosity, do you know why the internal pressure was < 0?
Just by initialization?

xEnVrE, 2021-01-25:
> @2103simon out of sheer curiosity, do you know why the internal pressure was < 0?
> Just by initialization?

Hi @pattacini, it was 0 because I think we were using a constructor that initializes it to 0.

https://github.com/robotology/icub-main/blob/8f4afa21352d7ec1fa76f1a6890f9d348a772802/src/libraries/skinDynLib/src/skinContact.cpp#L26

pattacini, 2021-01-25:
Thanks @xEnVrE !

Regarding this, I believe that `0.0` is an allowed value, so I'm gonna apply the following patch:
```c++
bool skinContact::setPressure(double _pressure){
    if(_pressure<0.0)
        return false;
    pressure = _pressure;
    return true;
}
```

closed 2021-01-25 by pattacini

simeonedussoni, 2021-01-26:
> 
> 
> Thanks @xEnVrE !
> 
> Regarding this, I believe that `0.0` is an allowed value, so I'm gonna apply the following patch (note the use of `<` in lieu of `<=`):
> 
> ```c++
> bool skinContact::setPressure(double _pressure){
>     if(_pressure<0.0)
>         return false;
>     pressure = _pressure;
>     return true;
> }
> ```

Actually if the pressure reads `0.0` the device can be saturated, don't know whether this info is used elsewhere in the code, anyway we can stick to the `< ` comparison.

pattacini, 2021-01-26:
@simeonedussoni yes, physically speaking you're right, but this method is used more in a programmatic way so to speak.
Actually, it gets used only externally by WBD and right with `0.0` input value to somehow force a reset.

https://github.com/robotology/icub-main/blob/85ee4c96728b16623b6d9ba72fc6202b67b6d088/src/modules/wholeBodyDynamics/observerThread.cpp#L1560

2103simon, 2021-01-26:
@pattacini, is the fix also applied to robotology-superbuild? I pulled it but saw no change yet. Maybe the issue is on my side.
Thanks for any hint :)

commit 85ee4c96 (2021-01-25): fix #715

cc @2103simon @xEnVrE