robotology/icub-main #717: Fix shared library support on Windows

traversaro, 2021-02-10:
Back in 2015, https://github.com/robotology/icub-main/pull/243 added support for compiling icub-main as a shared library on Windows, but this stopped working in https://github.com/robotology/icub-main/pull/259 as some glocal constant variables. At the moment, compilation as a shared library on Windows is failing with errors:
~~~
2021-02-09T23:54:29.6955824Z   Auto build dll exports
2021-02-09T23:54:29.6957082Z      Creating library C:/Miniconda/envs/test/conda-bld/icub-main_1612914590596/work/build/lib/Release/iKin.lib and object C:/Miniconda/envs/test/conda-bld/icub-main_1612914590596/work/build/lib/Release/iKin.exp
2021-02-09T23:54:29.6961128Z iKinFwd.obj : error LNK2019: unresolved external symbol "double const iCub::ctrl::CTRL_RAD2DEG" (?CTRL_RAD2DEG@ctrl@iCub@@3NB) referenced in function "public: virtual bool __cdecl iCub::iKin::iCubFinger::getChainJoints(class yarp::sig::VectorOf<double> const &,class yarp::sig::VectorOf<double> const &,class yarp::sig::VectorOf<double> &,class yarp::sig::Matrix const &)" (?getChainJoints@iCubFinger@iKin@iCub@@UEAA_NAEBV?$VectorOf@N@sig@yarp@@0AEAV456@AEBVMatrix@56@@Z) [%SRC_DIR%\build\src\libraries\iKin\iKin.vcxproj]
2021-02-09T23:54:29.6963919Z iKinInv.obj : error LNK2001: unresolved external symbol "double const iCub::ctrl::CTRL_RAD2DEG" (?CTRL_RAD2DEG@ctrl@iCub@@3NB) [%SRC_DIR%\build\src\libraries\iKin\iKin.vcxproj]
2021-02-09T23:54:29.6965384Z iKinSlv.obj : error LNK2001: unresolved external symbol "double const iCub::ctrl::CTRL_RAD2DEG" (?CTRL_RAD2DEG@ctrl@iCub@@3NB) [%SRC_DIR%\build\src\libraries\iKin\iKin.vcxproj]
2021-02-09T23:54:29.6968257Z iKinFwd.obj : error LNK2019: unresolved external symbol "double const iCub::ctrl::CTRL_DEG2RAD" (?CTRL_DEG2RAD@ctrl@iCub@@3NB) referenced in function "public: virtual bool __cdecl iCub::iKin::iCubArm::alignJointsBounds(class std::deque<class yarp::dev::IControlLimits *,class std::allocator<class yarp::dev::IControlLimits *> > const &)" (?alignJointsBounds@iCubArm@iKin@iCub@@UEAA_NAEBV?$deque@PEAVIControlLimits@dev@yarp@@V?$allocator@PEAVIControlLimits@dev@yarp@@@std@@@std@@@Z) [%SRC_DIR%\build\src\libraries\iKin\iKin.vcxproj]
2021-02-09T23:54:29.6971020Z iKinIpOpt.obj : error LNK2001: unresolved external symbol "double const iCub::ctrl::CTRL_DEG2RAD" (?CTRL_DEG2RAD@ctrl@iCub@@3NB) [%SRC_DIR%\build\src\libraries\iKin\iKin.vcxproj]
2021-02-09T23:54:29.6972579Z iKinSlv.obj : error LNK2001: unresolved external symbol "double const iCub::ctrl::CTRL_DEG2RAD" (?CTRL_DEG2RAD@ctrl@iCub@@3NB) [%SRC_DIR%\build\src\libraries\iKin\iKin.vcxproj]
2021-02-09T23:54:29.6973890Z %SRC_DIR%\build\lib\Release\iKin.dll : fatal error LNK1120: 2 unresolved externals [%SRC_DIR%\build\src\libraries\iKin\iKin.vcxproj]
2021-02-09T23:54:29.6975176Z   Building Custom Rule C:/Miniconda/envs/test/conda-bld/icub-main_1612914590596/work/src/libraries/icubmod/gazeController/CMakeLists.txt
2021-02-09T23:54:29.6976129Z   ClientGazeController.cpp
~~~

To correctly expose global constant variables on Windows as a shared library, it is necessary to use the `generate_export_header` function, see https://github.com/ms-iot/ROSOnWindows/pull/37/files for related docs. 

Note that this is not a big or urgent issue, as apparently shared library compilation on Windows stopped working after one month and no one noticed for 5 years, but it is better to track it somewhere if anyone is experience this error.

pattacini, 2021-02-11:
I'll have a look at this.

@traversaro is there any testing suite in the superbuild we could use against the upcoming fix branch?

traversaro, 2021-03-26:
> 
> 
> I'll have a look at this.
> 
> @traversaro is there any testing suite in the superbuild we could use against the upcoming fix branch?

Sorry, I had missed this, Actually the problem emerges also on a standalone compilation of icub-main, just enable the `ICUB_SHARED_LIBRARY` option and compile it on Windows, you should see the error I reported in the issue.

pattacini, 2021-03-26:
I'll try to address this over the weekend.

closed 2021-03-28 by pattacini