robotology/icub-main #721: ssgprtune script in the learningMachine module is outdated

agijsberts, 2021-03-02:
Hi all,

The `learningMachine` module comes with a `ssgprtune` script written in Python to pretrain optimized incremental sparse spectrum GPR models for deployment via the module. Unfortunately, the code is in Python 2 and makes use of OpenOpt, which is no longer being maintained.

Since I occasionally still receive requests for the incremental SSGPR code, not long ago I have split off these scripts and cleaned them up: Python 3, removed OpenOpt, `setup.py` and some tests and documentation. The result is available on https://github.com/agijsberts/ssgpr.

May I propose to replace the outdated version in the `icub` repository with a pointer to the dedicated `ssgpr` repository (e.g., via a simple textfile or perhaps a git submodule)?

pattacini, 2021-03-02:
Hi Arjan, good to hear from you again!
Sure, go ahead with your proposal and open up a pull request.
No git submodule, please! A markdown file filled with some documentation and proper pointers should be spot on!

traversaro, 2021-03-02:
fyi @singhbal-baljinder @raffaello-camoriano

pattacini, 2021-03-03:
Done in #722 ✔

closed 2021-03-03 by pattacini