robotology/icub-main #728: IMU filter may produce continuous warnings

S-Dafarra, 2021-03-18:
While testing the robot, I noticed the following warning message being printed continuously
```
Provided non-increasing time vector
```
that I tracked to https://github.com/robotology/icub-main/blob/7945541930924a539f147450bace9a8a0571d41f/src/libraries/ctrlLib/src/adaptWinPolyEstimator.cpp#L130.

By looking at which modules use that class, I found it used in the ``ImuFilter``: https://github.com/robotology/icub-main/blob/7945541930924a539f147450bace9a8a0571d41f/src/libraries/icubmod/imuFilter/ImuFilter.cpp#L85

I don't know if some check can be implemented on ``stampBias.getTime()`` to avoid this warning to be printed.

@Nicogene @pattacini

pattacini, 2021-03-18:
This warning message might be critical so I think it's important that got printed.
Let me investigate its nature before deciding how to proceed.

S-Dafarra, 2021-03-18:
> This warning message might be critical so I think it's important that got printed.
> Let me investigate its nature before deciding how to proceed.

Yes, sure. From the log it is not straightforward to understand what is going on and it is most likely ignored. It is possible there is some hardware issue, but I could not understand why that is printed.

Nicogene, 2021-03-18:
A side note: probably this warning was printed also before, but since in the past it was an executable nobody noticed it.
It has been "recently" converted to a device and it is launched by the `yarprobotinterface` then it is now visible in its log

pattacini, 2021-03-18:
What you say @Nicogene can be true, but my hunch is that it intimately depends on the type of the IMU.
With Xsens I did not spot such a warning when I coded the module `imuFilter`.

We know that with the Bosch unit within the RFE we've been suffering from problems related to timing.

pattacini, 2021-03-18:
With non-increasing time values, the computation of the derivative of the magnetic info cannot be performed (in correspondence of the time instant when the misbehavior happens) and this goes to the detriment of the integration process that estimates the gyro offsets.

It would be beneficial to dump directly the data provided by the IMU YARP port to inspect the profile time stamps.

@Nicogene do you know what is the name of such a port?

S-Dafarra, 2021-03-18:
I don't know if it helpful, but on iCub3 (the robot where I had this issue), these are the corresponding lines https://github.com/robotology/robots-configuration/blob/3c71e095a7c1c6b5463518dcbe96746430e588b6/iCubGenova09/icub_wbd.xml#L86-L87

Nicogene, 2021-03-18:
> @Nicogene do you know what is the name of such a port?

The name of the port is `/icub/head/inertials/measures:o` for the raw IMU,  and `/imuFilter/measures:o` should be the filtered one

pattacini, 2021-03-18:
@S-Dafarra would you be willing to launch the following the next time you'll be by iCub 3?

```sh
$ yarpdatadumper --rxTime --txTime --connect /icub/head/inertials/measures:o
```

pattacini, 2021-03-18:
@Nicogene I wonder if the data shown in https://github.com/robotology/icub-main/issues/595#issue-480708248 are relevant here.

The second plot displays cases where dt = 0, which actually would bring up the warning message.

Is the same IMU device mounted on iCubGenova04?

S-Dafarra, 2021-03-19:
> Thanks @Nicogene !
> 
> @S-Dafarra would you be willing to launch the following the next time you'll be by iCub 3?
> 
> ```shell
> $ yarpdatadumper --rxTime --txTime --connect /icub/head/inertials/measures:o
> ```

Sorry I missed this, I will try on Monday!

> Is the same IMU device mounted on iCubGenova04?

I think that on iCubGenova04 we still have the old one connected with ``i2c``: https://github.com/robotology/robots-configuration/blob/3c71e095a7c1c6b5463518dcbe96746430e588b6/iCubGenova04/hardware/inertials/head-inertial.xml#L5-L7 while we have the ``rfe`` board on iCub3: https://github.com/robotology/robots-configuration/blob/3c71e095a7c1c6b5463518dcbe96746430e588b6/iCubGenova09/hardware/inertials/head-inertial.xml#L45-L48

S-Dafarra, 2021-03-22:
> > Thanks @Nicogene !
> > @S-Dafarra would you be willing to launch the following the next time you'll be by iCub 3?
> > ```shell
> > $ yarpdatadumper --rxTime --txTime --connect /icub/head/inertials/measures:o
> > ```
> 
> Sorry I missed this, I will try on Monday!

Here the data. I just run it for about 30s 
[dump.zip](https://github.com/robotology/icub-main/files/6181441/dump.zip)

pattacini, 2021-03-22:
Thanks heaps @S-Dafarra !
I'll have a look at the data then.

pattacini, 2021-03-22:
[image]

Apparently, the timestamps are ok – both the tx and the rx.

@S-Dafarra did you get the error message popping up during the acquisition?

S-Dafarra, 2021-03-22:
> [image]
> 
> Apparently, the timestamps are ok – both the tx and the rx – meaning that we don't have any time delta <=0.
> 
> @S-Dafarra did you get the error message popping up during the acquisition?

I did not check. I also removed the imu filter from the robot ``yarprobotinterface`` to avoid filling the logger. Do you think it might be related to a temporary problem in reading the data?

S-Dafarra, 2021-03-22:
Here some other data. This time I did see the message popping out, even if it was not continuous

[dump.zip](https://github.com/robotology/icub-main/files/6182315/dump.zip)

pattacini, 2021-03-22:
[image]

Thus, it doesn't seem that we have non-increasing timestamps. The plot above just highlights that there's been one packet lost.

@S-Dafarra just disable the imuFilter device while we'll be delving deeper.

pattacini, 2021-03-23:
I took the chance to craft a quick testing snippet in [`pattacini/icub-main-728`][1] with the aim to check whether the [library component][2] is doing its job correctly.

The same exact library component is used within the module [`velocityObserver`][3], which is the one I employed for the test.

The test itself generates a stream to which we can specify to attach an increasing timestamp. Also, we can yield a glitch each second, meaning that the timestamp is for just one step non-increasing.

Here's how to launch the test suite (you can run it in Gitpod):
```sh
$ test-728 --add-stamp [on|off] --add-glitch [on|off]
$ velocityObserver
$ yarp connect /test /velObs/pos:i
```

Results are that we spot the warning message only when we generate the glitch!
Therefore, the library component seems to be running as intended when fed with a correct flow of timestamps.

Given that we didn't observe from the logs anything weird, it might be that the bug lurks in the handling of the timestamps when they get received from the device driver, prior to being passed on to the library component. 

[1]: https://github.com/pattacini/icub-main-728
[2]: https://robotology.github.io/robotology-documentation/doc/html/group__adaptWinPolyEstimator.html
[3]: https://robotology.github.io/robotology-documentation/doc/html/group__velocityObserver.html

pattacini, 2021-03-23:
@Nicogene can we discuss this a bit together when you'll be available?

Nicogene, 2021-03-24:
Yes sure!

pattacini, 2021-03-24:
After a quick debriefing w/ @Nicogene we agreed that the only test we can do is to check if the timestamp `ts` retrieved in 

https://github.com/robotology/icub-main/blob/7945541930924a539f147450bace9a8a0571d41f/src/libraries/icubmod/imuFilter/ImuFilter.cpp#L74

is ok or not.

To do this, we're required to make a local hack to the `imuFilter` code.

@S-Dafarra, we can plan for this in the next sprint unless you're willing to insert the following snippet in the code and give it yet another whirl:

```c++
 stampBias.update(ts);

static double t_{0.);
if (stampBias.getTime() <= t_) {
    yDebug() << "Got it!";
}
t_ = stampBias.getTime();
 double t0=Time::now();
```

davidetome, 2021-04-23:
I see the message also on the iCub3

[image]

S-Dafarra, 2021-04-23:
> I see the message also on the iCub3

That's where I first noticed it.

davidetome, 2021-04-23:
Thank you @S-Dafarra , it was only to say that I can run the test w/ the modified imuFilter code on it w/o using the greeny :+1:

davidetome, 2021-04-23:
@pattacini I ran the modified code for `imuFilter.cpp`and got 

[image]

pattacini, 2021-04-23:
So, this demonstrates that we do receive from the YARP I/F non-increasing timestamps: `imuFilter` is not the guilty guy.

That's kind of puzzling as this problem didn't come up from the logs we analyzed:
- did we log from the wrong port? cc @Nicogene
- was it that by chance the log didn't contain the problem itself? cc @S-Dafarra (I'd say that's unlikely)
- is there any possibility that the data on the net is ok and the YARP I/F is messing it up ahead of delivering it to the consumer? cc @Nicogene

Nicogene, 2021-04-23:
> did we log from the wrong port? cc @Nicogene

It seems the correct ones to me

> is there any possibility that the data on the net is ok and the YARP I/F is messing it up ahead of delivering it to the consumer? cc @Nicogene

If I remember correctly we already discussed about the timestamps set by `multipleanalogsensorsserver/client`, maybe @traversaro remember about it, I cannot find it.

The chain of the ts should be:

icub-head:multipleanalogsensorserver -> icubsrv:mutlipleanalogsensorsclient([ref](https://github.com/robotology/icub-main/blob/7945541930924a539f147450bace9a8a0571d41f/src/libraries/icubmod/imuFilter/PassThroughInertial.cpp)) -> icubsrv:multipleanalogsensorsserver

The imuFilter should take the ts from MAS server of icub-head, what I suspect is that MASserver if it does not have new data give you the latest in cache, this is why we have data with the same timestamps

The imuFilter has a period of 0 seconds see
https://github.com/robotology/icub-main/blob/7945541930924a539f147450bace9a8a0571d41f/src/libraries/icubmod/imuFilter/ImuFilter.cpp#L35

That's why you get continuously warnings

pattacini, 2021-04-23:
Thanks @Nicogene 👍🏻 

>The imuFilter has a period of 0 seconds

Nope, as you have correctly noticed we use `setPeriod`

With this respect, the following line is conceptually wrong

https://github.com/robotology/icub-main/blob/a8e24e8c024c0a2cef23218be72b8585ece0e1d4/src/libraries/icubmod/imuFilter/ImuFilter.cpp#L139

as `m_period` needs to be [given in seconds][1] (as per the default value `0.01`) while there's a weird manipulation because the device driver expects the period in ms.

Probably, the best would be to change the XML files from `ms` to `s` and update the values from `10` to `0.02`. 
Finally, the imuFilter line above needs to be updated accordingly.

Would you be willing to do the fix @Nicogene ?

⚠ [The original imuFilter codebase was in sync with the incoming data][2].

[1]: http://yarp.it/git-master/classyarp_1_1os_1_1PeriodicThread.html#a070daf7983102f96609a6a956ae7771f
[2]: https://github.com/robotology/icub-main/blob/463355866dbf64fa9018409c1c345c8e77bc0a96/src/tools/imuFilter/main.cpp#L88-L99

pattacini, 2021-04-24:
>Probably, the best would be to change the XML files from ms to s and update the values from 10 to 0.02 or even more (i.e. a bit more of the streaming rate of the IMU).
Finally, the imuFilter line above needs to be updated accordingly.
>
>Would you be willing to do the fix @Nicogene ?

I've changed my mind! I believe it's better to stick to ms as of now and in sight of a major overhaul when all the device driver will switch to the use of seconds.

I'll do the job then @Nicogene!
I'm on it.

pattacini, 2021-04-24:
PR's opened up:
- #745 
- https://github.com/robotology/robots-configuration/pull/255

pattacini, 2021-04-24:
@Nicogene we underestimated the lack of being in sync w/ the incoming data.

Increasing the period to prevent warnings is ok as the device is not crucial; however, we're basically subsampling the signals this way, making imuFilter not useful at all.

I think the imuFilter device should really restore this in-sync functionality.
Would you be willing to take on this task?

Nicogene, 2021-04-26:
> Increasing the period to prevent warnings is ok as the device is not crucial; however, we're basically subsampling the signals this way, making imuFilter not useful at all.

So we merge #745 and https://github.com/robotology/robots-configuration/pull/255 or we await the proper fix?
Because I suspect that after those changes the imuFilter would work worse than the raw data

> I think the imuFilter device should really restore the in-sync functionality.

About this we should attach to a callback of the masclient that I suspect does not exist, the alternative is to roll-back to the read of the port, but we have to parse the mas message.
We could use the blocking read as synchronizer, and then we retrieve the data with through the mas client.
@traversaro do you know if exist a cleaner way to do that?

pattacini, 2021-04-26:
>So we merge #745 and robotology/robots-configuration#255 or we await the proper fix?

I would proceed anyway, as those warnings might be as dangerous as going at 20 ms.

pattacini, 2021-04-26:
>@traversaro do you know if exist a cleaner way to do that?

He has already a plan! 😀
See https://github.com/robotology/icub-main/issues/746#issuecomment-826065649.

pattacini, 2021-04-26:
Let's move the discussion to #746.

PR #745 (2021-04-26): Fix imuFilter sampling of incoming data
This PR aims to fix #728.

To this end, the option `period` is handled in `ms` instead of `s`, plus the default value is brought to 20 ms instead of 10 ms in order to prevent the generation of continuous warning: we take a sampling margin wrt the rate of the incoming data, which is 10 ms.
closed 2021-04-26 by pattacini