robotology/icub-main #731: icub-main faceExpression - expose the current emotion for each face part

dariopasquali, 2021-03-26:
Hi,
it would be really helpful to know at runtime the current emotion for each facial part (mouth, eyelids, left eyebrow, right eyebrow).

### Possible solutions

The current implementation presents a GET command in the RPC interface which is not implemented.
is it possible to implement it and return the state of the four facial parts (i.ie, a Bottle with hap hap hap hap)?

Even better, you could implement a new port (i.e., /icub/face/emotions/state) that continuously streams the abovementioned Bottle of states

### Why is it useful?

Knowing the current emotion, it would be possible to better handle the speech by:
- using the current emotion as the default value for closed mouth
- restoring the previous emotion after a speech

Thank you!

pattacini, 2021-03-26:
@dariopasquali 

It's certainly doable but I am afraid that this change won't take place any time soon since we're busy with other development.

We are open to PR anyway 😉

Nicogene, 2021-03-26:
Hi @dariopasquali, I can give you pointers on what you need to change if you need it

closed 2023-07-30 by pattacini