robotology/icub-main #738: Remove embObjVirtualAnalogSensor device?

traversaro, 2021-04-01:
The `embObjVirtualAnalogSensor` device was supposed to expose the [`IVirtualAnalogSensor`](http://www.yarp.it/git-master/classyarp_1_1dev_1_1IVirtualAnalogSensor.html) to permit software (mainly the whole-body-dynamics) running on icub-head (or the robot cluster) to publish torque estimates that are then sent to the firmware to close the joint torque control loop. However, as far as I can remember and I can see in the robots-configuration repo ( https://github.com/robotology/robots-configuration/search?q=embObjVirtualAnalogSensor ), these devices are not used at all, as the same functionality is provided in the `embObjMotionControl` device. that is actually used for this on the robot (see https://github.com/robotology/icub-main/blob/cf42375998f1450ecfeaa027a64d83bd84c37094/src/libraries/icubmod/embObjMotionControl/embObjMotionControl.cpp#L3598).

To avoid confusion to people that need to maintain this part of the iCub software that is related to torque control, could it make sense to just remove the `embObjVirtualAnalogSensor` device, given that is completely unused?

traversaro, 2021-04-01:
cc @pattacini @Nicogene @marcoaccame @randaz81

pattacini, 2021-04-01:
I'm in favor of this!

Nicogene, 2021-04-01:
Ok for me!

pattacini, 2021-04-03:
Next week I'll proceed with this removal if there's no objection.

closed 2021-04-06 by pattacini