robotology/icub-main #740: actionsRenderingEngine: calib table stopped working

Nicogene, 2021-04-07:
Today I tested the `calib table` command with `actionRenderingEngine` of the latest tag (`v1.19.2`) and it never detect a contact, I put some prints and I found out that `contactDetectionOn` was always false and it is used while checking the table height:
https://github.com/robotology/icub-main/blob/89971ccd0eda518f2b3c972d1b805c2f44c411fc/src/libraries/actionPrimitives/src/actionPrimitives.cpp#L2212-L2232

Then I searched who should set that flag to true and I found out that this check:
https://github.com/robotology/icub-main/blob/89971ccd0eda518f2b3c972d1b805c2f44c411fc/src/modules/actionsRenderingEngine/src/MotorThread.cpp#L2538-L2539

is always false because:

```
action: A2_PN4iCub6action16ActionPrimitivesE
Primitive: N4iCub6action22ActionPrimitivesLayer2E

```

pattacini, 2021-04-07:
Uhm, this has always worked and I'm pretty sure that no one changed anything in this respect. I can have a look at it any how.

We need to make sure that the ActionPrimitives are configured with layer 2 (for contact detection) and this is done through configuration parameters.

pattacini, 2021-04-07:
Hi @Nicogene 

In the meantime, could you please report what are the printouts of `ARE`, especially for what regards the startup and in particular related to these lines:

https://github.com/robotology/icub-main/blob/89971ccd0eda518f2b3c972d1b805c2f44c411fc/src/modules/actionsRenderingEngine/src/MotorThread.cpp#L1244-L1253

Nicogene, 2021-04-07:
It print out the "layer 2" line

pattacini, 2021-04-07:
So, even though `action` is of type `ActionPrimitivesLayer2`, you observed that:
- `typeid(action).name()` == `A2_PN4iCub6action16ActionPrimitivesE`
- `typeid(ActionPrimitivesLayer2).name()` == `N4iCub6action22ActionPrimitivesLayer2E`

is that correct?

pattacini, 2021-04-07:
Well, with a second thought, `action` is of type `*ActionPrimitivesLayer2`, actually. So, how has it worked thus far? 😄
Ok, let me try something neater then.

pattacini, 2021-04-07:
This [resource][1] says something on the use of `.name()`:

>Returns an implementation defined null-terminated character string containing the name of the type. No guarantees are given; in particular, the returned string can be identical for several types and change between invocations of the same program.

Therefore, it may have worked in the past only by chance!

I'll see how to patch it then.

[1]: https://en.cppreference.com/w/cpp/types/type_info/name

Nicogene, 2021-04-07:
Probably a way to check the type could be using std::is_same and decltype(action) (ref-> https://en.cppreference.com/w/cpp/types/is_same)

closed 2021-04-08 by pattacini

pattacini, 2021-04-08:
I managed to trace back the [change][1] that introduced the use of `typeid().name()` in the commit b55e056a4bd40e7d318a82aab5241f3b3e876b16, which dates back to 17 months ago.

Actually, given the pandemic, we didn't test `ARE` that much last year: thus, this goes against the belief I expressed above.

Nicogene, 2021-04-08:
We tested it both with `icub-main` and `idyntree` `wholeBodyDynamics` and now it works smoothly in both cases 👍🏻

commit 7ebc1d6c (2021-04-08): fix #740
PR #741 (2021-04-08): Fix `ARE` check on action type
This PR fixes #740.

I took the chance to analyze the problem by using the following snippet:
```c++
#include <cstdlib>
#include <typeinfo>
#include <string>
#include <iostream>

#include <iCub/action/actionPrimitives.h>

void test(const int i, const bool cond) {
    std::cout << "test " << i << ": "
              << (cond ? "passed ✅" : "failed ❌")
              << std::endl;
}

int main() {
    iCub::action::ActionPrimitives* action = new iCub::action::ActionPrimitivesLayer2();

    test(1, typeid(action).name() == typeid(iCub::action::ActionPrimitivesLayer2).n [... cut]