robotology/icub-main #750: Calibration type 5 for EMS4 control boards connected to 2FOC DC - Firmware integration in icub-firmware repos

plinioMoreno, 2021-05-27:
### Is your feature request related to a problem?

The `ems4` control board attached to a `2foc` currently support only calibration type 9. The initial report of the missing feature is related to [this discussion](https://github.com/robotology/community/discussions/500) and [this issue](https://github.com/vislab-tecnico-lisboa/eth-boards-setups/issues/15)

### The solution you would like to have available

We want to have the calibration type 5 ([hardware stop](http://wiki.icub.org/wiki/Calibration_types_available_on_iCub_and_R1_robots)) implemented in the firmware of `ems4` board, and integrated in the icub-firmware* repositories.

### Additional context

In the near future we want to change all the control boards of our robot [Vizzy](https://github.com/vislab-tecnico-lisboa/vizzy) to the ETH versions. Since our robot currently runs `yarprobotinterface` using the config files [here](https://github.com/robotology/robots-configuration/tree/1f15d9dd2224f1e64bf3043a452e2811cd81caf4/vizzy), we need the calibration type 5 working and integrated in the official firmware files in icub-firmware* repositories.

pattacini, 2021-05-28:
Thanks @plinioMoreno for opening up this feature request.
As anticipated, we will be dealing with that in PI08 (starting end of July).

pattacini, 2021-11-21:
Hi @plinioMoreno 

Sorry for being late but we kind of struggled with this activity while attacking a big problem of the AMO absolute sensors on iCub 3. However, we have finally completed the task and now the EMS/2FOC are capable of carrying out the incremental calibration.

Relevant PR's:
- https://github.com/robotology/icub-firmware/pull/217
- https://github.com/robotology/icub-firmware-build/pull/40
- https://github.com/robotology/robots-configuration/pull/307

We eventually re-used the <kbd>Type 10</kbd> calibration that is already employed on R1 for the incremental calibration of the pronosupination.

Calibration Type 10 is not yet documented at https://icub-tech-iit.github.io/documentation/icub_r1_icub3_calibration_types/icub_r1_calibration_types/ but we will do it soon. In the meantime, I hope you could inherit the relevant quantities from the body of https://github.com/robotology/robots-configuration/pull/307 and with the help of @ale-git.

plinioMoreno, 2021-11-22:
Hi @pattacini,

Thanks for the update. If I understood correctly, we should update to the current devel repositories and upload the ems 3.45 firwmare to the board. We hope to run our tests here soon and will ask @ale-git for help.

cc @wevmacedo

pattacini, 2021-11-22:
> If I understood correctly, we should update to the current devel repositories and upload the ems 3.45 firwmare to the board. 

That's correct.

plinioMoreno, 2021-11-25:
Hi @pattacini and @ale-git,

We managed to upgrade the repositories and upload the firmwares of [EMS4](https://github.com/robotology/icub-firmware-build/blob/460680f657de754f3d4a82f235eb441bae22c420/ETH/EMS/bin/application/ems.hex) and [2foc](https://github.com/robotology/icub-firmware-build/blob/460680f657de754f3d4a82f235eb441bae22c420/CAN/2foc/2foc.hex).

We used the calibration parameters as follows:

```xml
      <group name="HOME">
        <param name="positionHome">                       0        </param>
        <param name="velocityHome">                       10.00    </param>
        </group>
               
        
        <group name="CALIBRATION">
        <param name="calibrationType">                    10        </param>
        <param name="calibration1">	2000		</param> 
        <param name="calibration2">	0	    </param> 
        <param name="calibration3">	0		</param>
     
        
        <param name="calibration4">                       0.0        </param>
        <param name="calibration5">                       0.0        </param>
        <param name="calibrationZero">                 10.0        </param>
        <param name="calibrationDelta">                   0.1        </param>
        <param name="startupPosition">                   0.00       </param>
        <param name="startupVelocity">                    10.0        </param>
        <param name="startupMaxPwm">                   2000          </param>
        <param name="startupPosThreshold">                2          </param>
        </group>

	<param name="CALIB_ORDER">(0) </param>
```

The output from `yarprobotinterface` is as follows:

```
[DEBUG] Knee_one_Calibrator : Calling calibrateJoint on joint  0 : type  10  with params:  2000 0 0 0 0
[DEBUG] Knee_one_Calibrator : set 1 j 0 : Calibrating... enc values AFTER calib:  0
[ERROR]  from BOARD 10.1.3.11 (Pq`U¶☻), src LOCAL, adr 0, time 31s 278m 461u: (code 0x0200000c, par16 0x0000 par64 0x0000000000000000) -> MC: 2FOC quadrature encoder index broken. The reference special thick was not detected during a complete revolution of the motor, please check motor encoder cables. In par64 0xFF is the mask of raw encoder value. par16 = ID of joint. + .
[ERROR] unable to complete calibration: joint 0 in 'hw_fault status' inside doneRaw() function
[ERROR] unable to complete calibration: joint 0 in 'hw_fault status' inside doneRaw() function
[ERROR] unable to complete calibration: joint 0 in 'hw_fault status' inside doneRaw() function
[ERROR] unable to complete calibration: joint 0 in 'hw_fault status' inside doneRaw() function
[ERROR] unable to complete calibration: joint 0 in 'hw_fault status' inside doneRaw() function
[ERROR]  from BOARD 10.1.3.11 (Pq`U¶☻), src LOCAL, adr 0, time 36s 278m 461u: (code 0x0200000c, par16 0x0000 par64 0x0000000000000000) -> MC: 2FOC quadrature encoder index broken. The reference special thick was not detected during a complete revolution of the motor, please check motor encoder cables. In par64 0xFF is the mask of raw encoder value. par16 = ID of joint. + .
[ERROR] unable to complete calibration: joint 0 in 'hw_fault status' inside doneRaw() function
[ERROR] unable to complete calibration: joint 0 in 'hw_fault status' inside doneRaw() function
[ERROR] unable to complete calibration: joint 0 in 'hw_fault status' inside doneRaw() function
[ERROR] unable to complete calibration: joint 0 in 'hw_fault status' inside doneRaw() function
[ERROR]  from BOARD 10.1.3.11 (Pq`U¶☻), src LOCAL, adr 0, time 41s 279m 461u: (code 0x0200000c, par16 0x0000 par64 0x0000000000000000) -> MC: 2FOC quadrature encoder index broken. The reference special thick was not detected during a complete revolution of the motor, please check motor encoder cables. In par64 0xFF is the mask of raw encoder value. par16 = ID of joint. + .
[ERROR] unable to complete calibration: joint 0 in 'hw_fault status' inside doneRaw() function
[ERROR] unable to complete calibration: joint 0 in 'hw_fault status' inside doneRaw() function
[ERROR] unable to complete calibration: joint 0 in 'hw_fault status' inside doneRaw() function
[ERROR] unable to complete calibration: joint 0 in 'hw_fault status' inside doneRaw() function
[ERROR] Knee_one_Calibrator : Timeout while calibrating  0
[ERROR] Knee_one_Calibrator : set 1 : Calibration went wrong! Disabling axes and keeping safe pid limit
[ERROR] embObjMotionControl::checkRemoteControlModeStatus(0, 7103593) is unable to check the control mode of BOARD knee-eb10-j1 IP 10.1.3.11 because it is now in HW_FAULT
```
The motor that we are using started to move in one direction, then it moved in the other direction. This happened several times up until it did not move anymore. Let us know what can we change in the calibration file, or if we need to use another 2foc firmware file.

cc @wevmacedo

ale-git, 2021-11-29:
Hi @wevmacedo, I think that first of all we have to check if the 2FOC configuration matches the real characteristics of your motor.
The error is related to quadrature encoder index detection. How many ticks has your encoder disk? Is the index provided? Can you address me to your configuration files?

pattacini, 2021-12-01:
Hi @plinioMoreno 

@ale-git provided the documentation of the parameters for Type 10.
Check it out at https://icub-tech-iit.github.io/documentation/icub_r1_icub3_calibration_types/icub_r1_calibration_types/#calibration-10.

wevmacedo, 2021-12-03:
Hi @ale-git and @pattacini,

Thanks for sending the documentation link of calibration type 10.

We are using the encoder [Faulhaber](https://www.faulhaber.com/fileadmin/Import/Media/EN_IE2-1024_DFF.pdf), with a resolution of 512. The motor we are using for our tests is a [Faulhaber 3242g024cr](https://www.faulhaber.com/fileadmin/Import/Media/EN_2342_CR_DFF.pdf). The motor is free, so there is no gearbox attached to it. About the `Gearbox_E2J`, we tested both 512 and 1, and for both values we got the same error as before:

```
[ERROR]  from BOARD 10.1.3.11 (░Æ⌂╔ƒ☺), src LOCAL, adr 0, time 124s 745m 92u: (code 0x0200000c, par16 0x0000 par64 0x0000000000000000) -> MC: 2FOC quadrature encoder index broken. The reference special thick was not detected during a complete revolution of the motor, please check motor encoder cables. In par64 0xFF is the mask of raw encoder value. par16 = ID of joint. + .
```

The configuration files are in [this commit](https://github.com/vislab-tecnico-lisboa/eth-boards-setups/tree/7aed1e73c2787acd826f945af14b56d76df884e1/single-ETH-ems4-1-motor-test-vislab)

ale-git, 2021-12-14:
Hi @wevmacedo, I see that the motor is DC, while only the 2FOC firmware for BLDC motors can send the index error message above, so I guess that the wrong firmware has been uploaded to the 2FOC board by mistake. The right firmware is

https://github.com/robotology/icub-firmware-build/blob/master/CAN/2foc/2foc.r1.dc.hex

About the quadrature encoder, if your wheel has 512 ticks the resolution in the configuration file must be 2048 because the board is configured to count any quadrature signal change.

wevmacedo, 2022-01-26:
Hi @ale-git,

We changed the firmware to the [r1 version of the previous message](https://github.com/robotology/icub-firmware-build/blob/master/CAN/2foc/2foc.r1.dc.hex), and also changed the `Gearbox_E2J` parameter in the file [knee-eb10-j1-mec-V25.xml](https://github.com/vislab-tecnico-lisboa/eth-boards-setups/blob/9d51e3e993ce35d304f063406845c9501ee4d0cb/single-ETH-ems4-1-motor-test-vislab/hardware/mechanicals/knee-eb10-j1-mec-V25.xml#L17). In addition, the calibration file is [knee-one-calib.xml](https://github.com/vislab-tecnico-lisboa/eth-boards-setups/blob/9d51e3e993ce35d304f063406845c9501ee4d0cb/single-ETH-ems4-1-motor-test-vislab/calibrators/knee-one-calib.xml).

When running the calibration procedure, the motor does not move and the process reaches the timeout. The screen info as follows:

```
[DEBUG] Reading file c:\robotology\eth-boards-setups\single-ETH-ems4-1-motor-test-vislab\.\test_knee_joint_V25.xml
[DEBUG] yarprobotinterface: using xml parser for DTD v3.x
[DEBUG] Reading file c:\robotology\eth-boards-setups\single-ETH-ems4-1-motor-test-vislab\.\test_knee_joint_V25.xml
[DEBUG] Preprocessor complete in:  0.0077202 s
[INFO] |yarp.os.Port| Port /icub/yarprobotinterface active at tcp://10.1.3.104:10004/
[INFO] startup phase starting...
[DEBUG] eth::parser::print(pc104Data) for PC104:
[DEBUG] PC104/PC104IpAddress:PC104IpPort =  10.1.3.104:12345
[DEBUG] PC104/PC104TXrate =  1
[DEBUG] PC104/PC104RXrate =  5
[DEBUG] EthSender is a PeriodicThread with txrate = 1 ms
[DEBUG] EthReceiver is a PeriodicThread with rxrate = 5 ms
[WARNING] in EthReceiver::config() the config socket has queue size =  1048576 ; you request ETHRECEIVER_BUFFER_SIZE=
[DEBUG] eth::parser::print(boardData) for BOARD knee-eb10-j1
[DEBUG] ETH_BOARD/ETH_BOARD_PROPERTIES:
[DEBUG] ETH_BOARD/ETH_BOARD_PROPERTIES/IpAddress =  10.1.3.11
[DEBUG] ETH_BOARD/ETH_BOARD_PROPERTIES/IpPort =  12345
[DEBUG] ETH_BOARD/ETH_BOARD_PROPERTIES/Type =  ems4
[DEBUG] ETH_BOARD/ETH_BOARD_PROPERTIES/maxSizeRXpacket =  768
[DEBUG] ETH_BOARD/ETH_BOARD_PROPERTIES/maxSizeROP =  384
[DEBUG] ETH_BOARD/ETH_BOARD_SETTINGS:
[DEBUG] ETH_BOARD/ETH_BOARD_SETTINGS/Name =  knee-eb10-j1
[DEBUG] ETH_BOARD/ETH_BOARD_SETTINGS/RUNNINGMODE/(period, maxTimeOfRXactivity, maxTimeOfDOactivity, maxTimeOfTXactivity, TXrateOfRegularROPs) =  1000 400 300 300 5
[DEBUG] ETH_BOARD/ETH_BOARD_ACTIONS/MONITOR_ITS_PRESENCE
[DEBUG] ETH_BOARD/ETH_BOARD_ACTIONS/MONITOR_ITS_PRESENCE/(enabled, timeout, periodOfMissingReport) =  true 0.02 60
[DEBUG] eth::EthMonitorPresence::config(): monitoring of presence is ON for BOARD 10.1.3.11 (knee-eb10-j1) with timeout = 0.02 sec and period of missing report = 60 sec
[DEBUG] TheEthManager::requestResource2(): has just succesfully created a new EthResource for board of type ems4 with IP =  10.1.3.11
[INFO] ServiceParser::parse_encoder_port(): ===========================>  eomc_enc_none
[INFO] ServiceParser::parse_encoder_port(): ===========================>  eomc_enc_roie
[WARNING] embObjMC BOARD  knee-eb10-j1 Missing OTHER_CONTROL_PARAMETERS.DeadZone parameter. I'll use default value. (see documentation for more datails)
[DEBUG] embObjMC BOARD  knee-eb10-j1 :fromConfig() detected that IMPEDANCE parameters section is found
[INFO] embObjMC BOARD  knee-eb10-j1 IMPEDANCE section: parameters successfully loaded
[DEBUG] EthResource::verifyBoardPresence() found BOARD knee-eb10-j1 with IP 10.1.3.11 after 0.0033091 seconds
[DEBUG] EthResource::verifyBoardTransceiver() has validated the transceiver of BOARD knee-eb10-j1 with IP 10.1.3.11
[DEBUG] EthResource::cleanBoardBehaviour() has cleaned the application in BOARD knee-eb10-j1 with IP 10.1.3.11 : config mode + cleared all its regulars
[DEBUG] EthResource::setTimingOfRunningCycle() for BOARD knee-eb10-j1 with IP 10.1.3.11 has succesfully set: cycletime = 1000 usec, RX DO TX = ( 400 300 300 ) usec and TX rate = 5  every cycle
[INFO] EthResource::askBoardVersion() found BOARD knee-eb10-j1 @ IP 10.1.3.11 of type ems4 with FW = ver 3.47 built on 2021 Nov 23 16:34
[DEBUG]  from BOARD 10.1.3.11 ( é(­═☺), src LOCAL, adr 0, time 333s 38m 184u: (code 0x05000015, par16 0x1000 par64 0x00000000000f000f) -> CFG: EOtheEncoderReader can be correctly configured. p16&0xf000: number of joint; primary encs: failure mask in p16&0x000f and errorcodes in p64&0x0000ffff; secondary encs: failure mask in p16&0x00f0 and errorcodes in p64&0xffff0000 + .
[INFO]  from BOARD 10.1.3.11 ( é(­═☺) @ 333s 38m 352u: CAN discovery has started for 1 eobrd_foc boards on (can1map, can2map) = (0x0002, 0x0000) with target can protocol ver 1.6 and application ver 5.3.220.
[INFO]  from BOARD 10.1.3.11 ( é(­═☺) @ 333s 38m 744u: CAN discovery has detected a eobrd_foc board in CAN1 addr 1 with can protocol ver 1.6 and application ver 5.3.220 Search time was 0 ms
[DEBUG] embObjMotionControl:serviceVerifyActivate OK!
[INFO]  from BOARD 10.1.3.11 ( é(­═☺) @ 333s 38m 905u: CAN discovery is OK for 1 eobrd_foc boards with target can protocol ver 1.6 and application ver 5.3.220. Search time was 0 ms
[DEBUG]  from BOARD 10.1.3.11 ( é(­═☺), src LOCAL, adr 0, time 333s 39m 92u: (code 0x0500000d, par16 0x0000 par64 0x0000000000000000) -> CFG: EOtheMotionController can correctly configure 2foc-based motion. more info will follow + .
[DEBUG]  from BOARD 10.1.3.11 ( é(­═☺), src LOCAL, adr 0, time 333s 79m 190u: (code 0x04000001, par16 0x000f par64 0x0000010000000040) -> DEBUG: tag01 + CURRENT PID
[... cut ...]
[ERROR] A 0.250000 sec timeout occured in embObjMotionControl::checkRemoteControlModeStatus(), BOARD knee-eb10-j1 IP 10.1.3.11, joint 0, current mode: cal, requested: idl
[ERROR] failure of embObjMotionControl::checkRemoteControlModeStatus(j=0, targetmode=idl) for BOARD knee-eb10-j1 IP 10.1.3.11 after 17 attempts and 0.256480 seconds
[ERROR] In embObjMotionControl::setControlModeRaw(j= 0 , mode= idl ) for  BOARD knee-eb10-j1 (IP 10.1.3.11)   has failed checkRemoteControlModeStatus()
[DEBUG] knee-one-calibrator finished calibration of device knee-one-mc_wrapper
```

Let us know what should we change

ale-git, 2022-03-25:
Hi @wevmacedo, I succeeded at last in making calibration type 10 working with a joint without primary encoder that uses the motor encoder connected to the 2FOC board. I think that it should work with your DC motor as well.
[ems004.lisboa.zip](https://github.com/robotology/icub-main/files/8352637/ems004.lisboa.zip)

@pattacini

wevmacedo, 2022-05-13:
Hi @ale-git, we uploaded the new firmware to the EMS board, and after running `yarprobotinterface` the output is this file:
[yarprobotinterfaceOutput.txt](https://github.com/robotology/icub-main/files/8689470/yarprobotinterfaceOutput.txt). It seems that is trying to calibrate, but our DC motor does not move. The final state of the joint in `yarpmotorgui` is:

[image]

The paramerters of the calibrator are as follows:

```xml
        <group name="HOME">
        <param name="positionHome">                       0        </param>
        <param name="velocityHome">                       10.00    </param>
        </group>
               
        
        <group name="CALIBRATION">
        <param name="calibrationType">                    10        </param>
        <param name="calibration1">	4000		</param> 
        <param name="calibration2">	0	    </param> 
        <param name="calibration3">	0		</param>
     
        
        <param name="calibration4">                       0.0        </param>
        <param name="calibration5">                       0.0        </param>
        <param name="calibrationZero">                 10.0        </param>
        <param name="calibrationDelta">                   0.1        </param>
        <param name="startupPosition">                   0.00       </param>
        <param name="startupVelocity">                    0.0        </param>
        <param name="startupMaxPwm">                   5000          </param>
        <param name="startupPosThreshold">                0.0          </param>
        </group>

	<param name="CALIB_ORDER">(0) </param>
```

Let us know if we should run more tests.

pattacini, 2022-07-28:
@ale-git and @wevmacedo are working these days side by side on this.
I reckon we can close this task.

closed 2022-07-28 by pattacini