robotology/icub-main #754: CanBusMotionControl warnings

traversaro, 2021-06-12:
### Is your feature request related to a problem?

While working on another issue, I noticed the following warnings:
~~~
[ 14%] Building CXX object src/libraries/icubmod/canBusMotionControl/CMakeFiles/canmotioncontrol.dir/CanBusMotionControl.cpp.o
/workspace/gitpod-ubuntu-20.04/robotology-superbuild/src/ICUB/src/libraries/icubmod/canBusMotionControl/CanBusMotionControl.cpp: In member function 'unsigned char yarp::dev::CanBusMotionControl::from_modevocab_to_modeint(int)':
/workspace/gitpod-ubuntu-20.04/robotology-superbuild/src/ICUB/src/libraries/icubmod/canBusMotionControl/CanBusMotionControl.cpp:3934:17: warning: unsigned conversion from 'yarp::conf::vocab32_t' {aka 'int'} to 'unsigned char' changes value from '2003529333' to '117' -Woverflow]
 3934 |         return  VOCAB_CM_UNKNOWN;
      |                 ^~~~~~~~~~~~~~~~
/workspace/gitpod-ubuntu-20.04/robotology-superbuild/src/ICUB/src/libraries/icubmod/canBusMotionControl/CanBusMotionControl.cpp:3943:16: warning: unsigned conversion from 'yarp::conf::vocab32_t' {aka 'int'} to 'unsigned char' changes value from '2003529333' to '117' -Woverflow]
 3943 |         return VOCAB_CM_UNKNOWN;
      |                ^~~~~~~~~~~~~~~~
/workspace/gitpod-ubuntu-20.04/robotology-superbuild/src/ICUB/src/libraries/icubmod/canBusMotionControl/CanBusMotionControl.cpp: In member function 'virtual void yarp::dev::CanBusMotionControl::run()':
/workspace/gitpod-ubuntu-20.04/robotology-superbuild/src/ICUB/src/libraries/icubmod/canBusMotionControl/CanBusMotionControl.cpp:3614:49: warning: '%s' directive writing up to 254 bytes into a region of size between 1 and 255 [-Wformat-overflow=]
 3614 |                             sprintf(message, "%s%s", message, tmp);
      |                                                 ^~            ~~~
In file included from /usr/include/stdio.h:867,
                 from /usr/include/c++/9/cstdio:42,
                 from /workspace/gitpod-ubuntu-20.04/robotology-superbuild/src/ICUB/src/libraries/icubmod/canBusMotionControl/CanBusMotionControl.cpp:17:
/usr/include/x86_64-linux-gnu/bits/stdio2.h:36:34: note: '__builtin___sprintf_chk' output between 1 and 509 bytes into a destination of size 255
   36 |   return __builtin___sprintf_chk (__s, __USE_FORTIFY_LEVEL - 1,
      |          ~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
   37 |       __bos (__s), __fmt, __va_arg_pack ());
      |       ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/workspace/gitpod-ubuntu-20.04/robotology-superbuild/src/ICUB/src/libraries/icubmod/canBusMotionControl/CanBusMotionControl.cpp:3628:41: warning: '%s' directive writing up to 254 bytes into a region of size between 1 and 255 [-Wformat-overflow=]
 3628 |                     sprintf(message, "%s%s", message, tmp);
      |                                         ^~            ~~~
In file included from /usr/include/stdio.h:867,
                 from /usr/include/c++/9/cstdio:42,
                 from /workspace/gitpod-ubuntu-20.04/robotology-superbuild/src/ICUB/src/libraries/icubmod/canBusMotionControl/CanBusMotionControl.cpp:17:
/usr/include/x86_64-linux-gnu/bits/stdio2.h:36:34: note: '__builtin___sprintf_chk' output between 1 and 509 bytes into a destination of size 255
   36 |   return __builtin___sprintf_chk (__s, __USE_FORTIFY_LEVEL - 1,
      |          ~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
   37 |       __bos (__s), __fmt, __va_arg_pack ());
      |       ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
~~~

### The solution you would like to have available

I would like to check if those warnings are actually dangerous.

pattacini, 2021-06-12:
Actually, [CI is failing](https://github.com/robotology/icub-main/runs/2807883992?check_suite_focus=true) ❌

Perhaps, as a result of the huge changes in https://github.com/robotology/yarp/pull/2596?
Need to investigate.

cc @drdanz

traversaro, 2021-06-12:
> Actually, [CI is failing](https://github.com/robotology/icub-main/runs/2807883992?check_suite_focus=true) ❌
> 
> Perhaps, as a result of the huge changes in [robotology/yarp#2596](https://github.com/robotology/yarp/pull/2596)?
> Need to investigate.
> 
> cc @drdanz

The nightly failure was detected in https://github.com/robotology/robotology-superbuild/issues/794, I am in the process of opening PRs with the fixes.

pattacini, 2021-08-03:
Fixed in #761.

closed 2021-08-03 by pattacini