robotology/icub-main #759: Get the desired current sent to the current controller

isorrentino, 2021-07-16:
### Is your feature request related to a problem?

@GiulioRomualdi and I, were investigating the possibility of getting the desired current generated by the ems when the torque-control via current is enabled. Looking in yarp, there is a IPIDControl interface already used in https://github.com/dic-iit/bipedal-locomotion-framework, to get the reference joint position and error. It is possible to set the different PIDTYPE. Looking at its implementation (https://github.com/robotology/icub-main/blob/26282a9bf84dd0a162105a111c285617312bda69/src/libraries/icubmod/embObjMotionControl/embObjMotionControl.cpp#L4395) running on icub-head, I saw that VOCAB_PIDTYPE_CURRENT return always zero and so, we cannot read the reference motor currents and errors.

cc @traversaro 

### The solution you would like to have available

It would be useful to be able to access the reference motor currents when the current control is active on the 2FOC boards (i.e. when the control mode is set to `current` or to `torque via current`).

pattacini, 2021-07-16:
All clear @isorrentino.
We could be looking at this in the first sprint of our upcoming quarter PI08.

isorrentino, 2021-07-16:
Cc @nunoguedelha

mfussi66, 2021-08-02:
@ale-git and I checked the code and we understood that if the 2FOC is current-controlled, the set-point sent by the EMS will already be a current quantity (in mA). As long as you configure the 2FOC to be current-controlled, you do not need to change the code, and you can read the current output with the parameter `VOCAB_PIDTYPE_TORQUE`. Therefore, you can use the function 
[getPidOutputRaw(const PidControlTypeEnum& pidtype, int j, double *out) ](https://github.com/robotology/icub-main/blob/26282a9bf84dd0a162105a111c285617312bda69/src/libraries/icubmod/embObjMotionControl/embObjMotionControl.cpp#L4395) and specify `pidtype` as `VOCAB_PIDTYPE_TORQUE`.

This is possible because the appropriate controller is instantiated depending on the 2FOC configuration, and the EMS output unit of measurement can be deduced from it. Additionally, the entry [VOCAB_PIDTYPE_CURRENT](https://github.com/robotology/icub-main/blob/26282a9bf84dd0a162105a111c285617312bda69/src/libraries/icubmod/embObjMotionControl/embObjMotionControl.cpp#L4421) is misleading, since there is not an explicit "current" EMS control mode.

Finally, you can control the robot joints with current setpoints generated from the higher level software. In this case the EMS control mode will be "openloop" and the output to the 2FOC will be a current quantity.

Please report us back if it works as you desired, or anything suspicious!

pattacini, 2021-09-02:
Closing, feel free to open it up again if required.

closed 2021-09-02 by pattacini