robotology/icub-main #763: Compilation failure when compiling icub-main devel branch against yarp latest master branch (2021/09/02)

traversaro, 2021-09-02:
### Bug description

Compiling latest icub-main devel against YARP master branch fails with error:

~~~
2021-09-02T02:41:33.3567443Z [334/458] Building CXX object src/simulators/iCubSimulation/CMakeFiles/iCub_SIM.dir/odesdl/MS3D.cpp.o
2021-09-02T02:41:33.3568333Z FAILED: src/simulators/iCubSimulation/CMakeFiles/iCub_SIM.dir/odesdl/MS3D.cpp.o 
2021-09-02T02:41:33.3575273Z /usr/bin/c++ -DICUB_SIM_ENABLE_FAKE -DICUB_SIM_ENABLE_ODESDL -DOMIT_LOGPOLAR -DUSE_GLUT -D_USE_MATH_DEFINES -DdDOUBLE -I/home/runner/work/robotology-superbuild/robotology-superbuild/src/ICUB/src/simulators/iCubSimulation -I/home/runner/work/robotology-superbuild/robotology-superbuild/src/ICUB/src/simulators/iCubSimulation/wrapper -I/home/runner/work/robotology-superbuild/robotology-superbuild/src/ICUB/src/simulators/iCubSimulation/fake -I/usr/include/SDL -I/home/runner/work/robotology-superbuild/robotology-superbuild/src/ICUB/src/simulators/iCubSimulation/odesdl -I/home/runner/work/robotology-superbuild/robotology-superbuild/src/ICUB/src/libraries/iKin/include -I/home/runner/work/robotology-superbuild/robotology-superbuild/src/ICUB/src/libraries/ctrlLib/include -I/home/runner/work/robotology-superbuild/robotology-superbuild/src/ICUB/src/libraries/skinDynLib/include -isystem /home/runner/work/robotology-superbuild/robotology-superbuild/build/install/include -O3 -DNDEBUG -fPIE -std=c++17 -MD -MT src/simulators/iCubSimulation/CMakeFiles/iCub_SIM.dir/odesdl/MS3D.cpp.o -MF src/simulators/iCubSimulation/CMakeFiles/iCub_SIM.dir/odesdl/MS3D.cpp.o.d -o src/simulators/iCubSimulation/CMakeFiles/iCub_SIM.dir/odesdl/MS3D.cpp.o -c /home/runner/work/robotology-superbuild/robotology-superbuild/src/ICUB/src/simulators/iCubSimulation/odesdl/MS3D.cpp
2021-09-02T02:41:33.3581518Z In file included from /home/runner/work/robotology-superbuild/robotology-superbuild/src/ICUB/src/simulators/iCubSimulation/odesdl/MS3D.h:40,
2021-09-02T02:41:33.3583084Z                  from /home/runner/work/robotology-superbuild/robotology-superbuild/src/ICUB/src/simulators/iCubSimulation/odesdl/MS3D.cpp:28:
2021-09-02T02:41:33.3584113Z /usr/include/SDL/SDL_opengl.h:116: warning: "GL_GLEXT_VERSION" redefined
2021-09-02T02:41:33.3584686Z   116 | #define GL_GLEXT_VERSION 29
2021-09-02T02:41:33.3585030Z       | 
2021-09-02T02:41:33.3585432Z In file included from /usr/include/GL/gl.h:2050,
2021-09-02T02:41:33.3586034Z                  from /usr/include/SDL/SDL_opengl.h:46,
2021-09-02T02:41:33.3587168Z                  from /home/runner/work/robotology-superbuild/robotology-superbuild/src/ICUB/src/simulators/iCubSimulation/odesdl/MS3D.h:40,
2021-09-02T02:41:33.3588699Z                  from /home/runner/work/robotology-superbuild/robotology-superbuild/src/ICUB/src/simulators/iCubSimulation/odesdl/MS3D.cpp:28:
2021-09-02T02:41:33.3589741Z /usr/include/GL/glext.h:54: note: this is the location of the previous definition
2021-09-02T02:41:33.3590335Z    54 | #define GL_GLEXT_VERSION 20190805
2021-09-02T02:41:33.3590703Z       | 
2021-09-02T02:41:33.3591747Z In file included from /home/runner/work/robotology-superbuild/robotology-superbuild/src/ICUB/src/simulators/iCubSimulation/odesdl/MS3D.cpp:28:
2021-09-02T02:41:33.3593445Z /home/runner/work/robotology-superbuild/robotology-superbuild/src/ICUB/src/simulators/iCubSimulation/odesdl/MS3D.h:84:5: error: reference to ‘byte’ is ambiguous
2021-09-02T02:41:33.3594377Z    84 |     byte m_flags;
2021-09-02T02:41:33.3594692Z       |     ^~~~
2021-09-02T02:41:33.3595691Z In file included from /home/runner/work/robotology-superbuild/robotology-superbuild/build/install/include/yarp/conf/numeric.h:19,
2021-09-02T02:41:33.3597208Z                  from /home/runner/work/robotology-superbuild/robotology-superbuild/build/install/include/yarp/os/ConnectionReader.h:10,
2021-09-02T02:41:33.3598828Z                  from /home/runner/work/robotology-superbuild/robotology-superbuild/build/install/include/yarp/os/PortReaderCreator.h:10,
2021-09-02T02:41:33.3600306Z                  from /home/runner/work/robotology-superbuild/robotology-superbuild/build/install/include/yarp/os/Port.h:13,
2021-09-02T02:41:33.3601731Z                  from /home/runner/work/robotology-superbuild/robotology-superbuild/build/install/include/yarp/os/BufferedPort.h:11,
2021-09-02T02:41:33.3603276Z                  from /home/runner/work/robotology-superbuild/robotology-superbuild/src/ICUB/src/simulators/iCubSimulation/odesdl/VideoTexture.h:34,
2021-09-02T02:41:33.3604882Z                  from /home/runner/work/robotology-superbuild/robotology-superbuild/src/ICUB/src/simulators/iCubSimulation/odesdl/rendering.h:33,
2021-09-02T02:41:33.3606435Z                  from /home/runner/work/robotology-superbuild/robotology-superbuild/src/ICUB/src/simulators/iCubSimulation/odesdl/MS3D.h:52,
2021-09-02T02:41:33.3607951Z                  from /home/runner/work/robotology-superbuild/robotology-superbuild/src/ICUB/src/simulators/iCubSimulation/odesdl/MS3D.cpp:28:
2021-09-02T02:41:33.3609312Z /usr/include/c++/9/cstddef:69:14: note: candidates are: ‘enum class std::byte’
2021-09-02T02:41:33.3609977Z    69 |   enum class byte : unsigned char {};
2021-09-02T02:41:33.3610355Z       |              ^~~~
2021-09-02T02:41:33.3611444Z In file included from /home/runner/work/robotology-superbuild/robotology-superbuild/src/ICUB/src/simulators/iCubSimulation/odesdl/MS3D.cpp:28:
2021-09-02T02:41:33.3613140Z /home/runner/work/robotology-superbuild/robotology-superbuild/src/ICUB/src/simulators/iCubSimulation/odesdl/MS3D.h:73:23: note:                 ‘typedef unsigned char byte’
2021-09-02T02:41:33.3614131Z    73 | typedef unsigned char byte;
2021-09-02T02:41:33.3614523Z       |                       ^~~~
~~~

### Steps to reproduce

The easiest way is to just compile the robotology-superbuild with `-DROBOTOLOGY_PROJECT_TAGS:STRING=Unstable` , see https://github.com/robotology/robotology-superbuild/issues/872 .

### Expected behavior

I would expect the compilation to be successful.

Nicogene, 2021-09-02:
Maybe it is related to https://github.com/robotology/yarp/pull/2702

pattacini, 2021-09-02:
I'm on it.

PR #764 (2021-09-02): Fix name clash with `std::byte`
Fixes #763.
closed 2021-09-02 by pattacini