robotology/icub-main #765: cartesian Interface

cedricgoubard, 2021-09-14:
### Device name 🤖

iCubLondon01

### Request/Failure description

Hello, 

I am trying to use the cartesian interface to move `iCub`'s arms to a position received from a ROS topic. However, I don't think the `ICartesianControl` is created properly, and my code fails with a `segmentation fault` when I start publishing coordinates to the ROS topic.

### Detailed context

<details> <summary>Here is the code</summary>

```c++
#include <cstdio>
#include <cmath>
#include <unistd.h>
#include <string>

#include <yarp/os/Network.h>
#include <yarp/os/RFModule.h>
#include <yarp/os/PeriodicThread.h>
#include <yarp/os/LogStream.h>
#include <yarp/os/Time.h>
#include <yarp/sig/Vector.h>
#include <yarp/math/Math.h>

#include <yarp/dev/Drivers.h>
#include <yarp/dev/CartesianControl.h>
#include <yarp/dev/PolyDriver.h>

#include <ros/ros.h> 
#include <geometry_msgs/PointStamped.h>

using namespace std;
using namespace yarp::os;
using namespace yarp::dev;
using namespace yarp::sig;
using namespace yarp::math;
  

ICartesianControl* setup_cart_control(string side){
    Property option;
    option.put("device","cartesiancontrollerclient");
    
    string s = string("/icub/cartesianController/") + side + string("_arm");
    yInfo() << "Connecting to " << s;
    option.put("remote", s);
    option.put("local","/client/"+ side + string("_arm"));

    PolyDriver clientCartCtrl(option);

    ICartesianControl *icart=NULL;

    if (clientCartCtrl.isValid()) {
        clientCartCtrl.view(icart);
    }
    return icart;
}

class ROSCommands2YARP{
    private:
        ros::Subscriber subLeft;
        // ros::Subscriber subRight;

        ICartesianControl *icartLeft;
        // ICartesianControl *icartRight;

    public:
        void cmd_callback(const geometry_msgs::PointStamped::ConstPtr& point, ICartesianControl *icart){
            Vector x, o;
            icart->getPose(x,o);

            x[0] = point->point.x;
            x[1] = point->point.y;
            x[2] = point->point.z;

            string str(x.begin(), x.end());
            yInfo() << "Moving x to " << str;

            icart->goToPoseSync(x, o);    // send request and wait for reply
            icart->waitMotionDone(0.04);  // wait until the motion is done and ping at each 0.04 seconds
        }

        // void cmd_callback_right(const geometry_msgs::PointStamped::ConstPtr& point){
        //     cmd_callback(point, icartRight);
        // }

        void cmd_callback_left(const geometry_msgs::PointStamped::ConstPtr& point){
            cmd_callback(point, icartLeft);
        }

        ROSCommands2YARP(ros::NodeHandle* n){
            Network yarp;
            if (!yarp.checkNetwork())
            {
                fprintf(stdout,"Error: yarp server does not seem available\n");
                exit;
            }

            icartLeft = setup_cart_control("left");
            // icartRight = setup_cart_control("right");

            subLeft = n->subscribe<geometry_msgs::PointStamped>("/ros2yarp/left_arm", 1000, &ROSCommands2YARP::cmd_callback_left, this);
            // subRight = n->subscribe<geometry_msgs::PointStamped>("/ros2yarp/right_arm", 1000, &ROSCommands2YARP::cmd_callback_right, this);
  
        }
};

int main(int argc, char **argv)
{
    ros::init(argc, argv, "yarp2ros");
    ros::NodeHandle n;
    ROSCommands2YARP Cmd(&n);

    ros::Rate loop_rate(30);
    while (ros::ok()) { 
        ros::spinOnce();
    }
    return 0;

}

```
</details>

Here is the output when I run it:
```bash
  ~/ros-ws/src/yarp2ros on  feat/hand_ctrl •1 ✗ ❯ rosrun yarp2ros yarp2ros_hand_ctrl 
[INFO] Connecting to  /icub/cartesianController/left_arm
[INFO] |yarp.os.Port| Port /client/left_arm/command:o active at tcp://10.0.2.107:10236/
[INFO] |yarp.os.Port| Port /client/left_arm/state:i active at tcp://10.0.2.107:10237/
[INFO] |yarp.os.Port| Port /client/left_arm/events:i active at tcp://10.0.2.107:10238/
[INFO] |yarp.os.Port| Port /client/left_arm/rpc:o active at tcp://10.0.2.107:10239/
[INFO] |yarp.os.impl.PortCoreOutputUnit| Sending output from /client/left_arm/rpc:o to /icub/cartesianController/left_arm/rpc:i using tcp
[INFO] |yarp.os.impl.PortCoreOutputUnit| Sending output from /client/left_arm/command:o to /icub/cartesianController/left_arm/command:i using udp
[INFO] |yarp.os.impl.PortCoreInputUnit| Receiving input from /icub/cartesianController/left_arm/state:o to /client/left_arm/state:i using udp
[INFO] |yarp.os.impl.PortCoreInputUnit| Receiving input from /icub/cartesianController/left_arm/events:o to /client/left_arm/events:i using udp
[INFO] |yarp.dev.PolyDriver| Created wrapper <cartesiancontrollerclient>. See C++ class ClientCartesianController for documentation.
[INFO] |yarp.os.impl.PortCoreOutputUnit| output for route /client/left_arm/command:o->udp->/icub/cartesianController/left_arm/command:i asking other side to close by out-of-band means
[INFO] |yarp.os.impl.PortCoreOutputUnit| Removing output from /client/left_arm/command:o to /icub/cartesianController/left_arm/command:i
[INFO] |yarp.os.impl.PortCoreInputUnit| Removing input from /icub/cartesianController/left_arm/state:o to /client/left_arm/state:i
[INFO] |yarp.os.impl.PortCoreInputUnit| Removing input from /icub/cartesianController/left_arm/events:o to /client/left_arm/events:i
[INFO] |yarp.os.impl.PortCoreOutputUnit| Removing output from /client/left_arm/rpc:o to /icub/cartesianController/left_arm/rpc:i
Segmentation fault (core dumped)
```

The final `segmentation fault` only appears when I start publishing to the topic.
From what I understand from the logs, the `icartLeft` object is created, connects to the robot, but is then immediately destroyed (which results in the second part of the logs, with all the ports closing). However, I am not really sure why this is happening...

Thank you for any help you could provide on this!

traversaro, 2021-09-14:
The problem is that `clientCartCtrl` goes out of scope when the `setup_cart_control` function exits. The `icart` pointer that you obtained via `clientCartCtrl.view(icart)` is only valid as long as the `clientCartCtrl` object is not destroyed.

traversaro, 2021-09-14:
Probably you can just make the `clientCartCtrl` object a member of `ROSCommands2YARP, and `setup_cart_control` a method.

cedricgoubard, 2021-09-14:
Wow, that was a quick answer! 
You were right, that fixed the issue. I also had to move the `Network` object as an attribute, and now it's working great!

<details><summary>Here is the new code</summary>

```c++
#include <cstdio>
#include <cmath>
#include <unistd.h>
#include <string>

#include <yarp/os/Network.h>
#include <yarp/os/RFModule.h>
#include <yarp/os/PeriodicThread.h>
#include <yarp/os/LogStream.h>
#include <yarp/os/Time.h>
#include <yarp/sig/Vector.h>
#include <yarp/math/Math.h>

#include <yarp/dev/Drivers.h>
#include <yarp/dev/CartesianControl.h>
#include <yarp/dev/PolyDriver.h>

#include <ros/ros.h> 
#include <geometry_msgs/PointStamped.h>

using namespace std;
using namespace yarp::os;
using namespace yarp::dev;
using namespace yarp::sig;
using namespace yarp::math;
  

class ROSCommands2YARP{
    private:
        ros::Subscriber subLeft;
        // ros::Subscriber subRight;

        Network yarp;

        ICartesianControl *icartLeft;
        // ICartesianControl *icartRight;

        PolyDriver clientCartCtrlLeft;
        PolyDriver clientCartCtrlRight;

        ICartesianControl* setup_cart_control(string side){
            Property option;
            option.put("device","cartesiancontrollerclient");
            
            string s = string("/icub/cartesianController/") + side + string("_arm");
            yInfo() << "Connecting to " << s;
            option.put("remote", s);
            option.put("local","/client/"+ side + string("_arm"));

            ICartesianControl *icart=NULL;

            if(side == "left"){
                clientCartCtrlLeft.open(option);
                    if (clientCartCtrlLeft.isValid()) {
                    clientCartCtrlLeft.view(icart);
                }
            }

            if(side == "right"){
                clientCartCtrlRight.open(option);
                if (clientCartCtrlRight.isValid()) {
                    clientCartCtrlRight.view(icart);
                }
            }
            
            yInfo() << "Successfully connected to " << s;
            return icart;
        }

    public:
        void cmd_callback(const geometry_msgs::PointStamped::ConstPtr& point, ICartesianControl *icart){
            Vector x, o;
            icart->getPose(x,o);

            x[0] = point->point.x;
            x[1] = point->point.y;
            x[2] = point->point.z;

            yInfo() << "Moving to " << x.toString();

            icart->goToPoseSync(x, o);    // send request and wait for reply
            icart->waitMotionDone(0.04);  // wait until the motion is done and ping at each 0.04 seconds
            yInfo() << "Movement finished";
        }

        // void cmd_callback_right(const geometry_msgs::PointStamped::ConstPtr& point){
        //     cmd_callback(point, icartRight);
        // }

        void cmd_callback_left(const geometry_msgs::PointStamped::ConstPtr& point){
            cmd_callback(point, icartLeft);
        }

        ROSCommands2YARP(ros::NodeHandle* n){
            if (!yarp.checkNetwork())
            {
                fprintf(stdout,"Error: yarp server does not seem available\n");
                exit;
            }

            icartLeft = setup_cart_control("left");
            // icartRight = setup_cart_control("right");

            subLeft = n->subscribe<geometry_msgs::PointStamped>("/ros2yarp/left_arm", 1000, &ROSCommands2YARP::cmd_callback_left, this);
            // subRight = n->subscribe<geometry_msgs::PointStamped>("/ros2yarp/right_arm", 1000, &ROSCommands2YARP::cmd_callback_right, this);
  
        }
};

int main(int argc, char **argv)
{
    ros::init(argc, argv, "yarp2ros");
    ros::NodeHandle n;
    ROSCommands2YARP Cmd(&n);

    ros::Rate loop_rate(30);
    while (ros::ok()) { 
        ros::spinOnce();
    }
    return 0;

}

```

</details>

closed 2021-09-14 by cedricgoubard

traversaro, 2021-09-14:
> Wow, that was a quick answer!

Don't get too used to it, sometimes we are not so quick. : )

cedricgoubard, 2021-09-14:
@traversaro 
> Don't get too used to it, sometimes we are not so quick. : )

I am already really grateful for all the answers you have provided so far, even when it takes a bit longer :smile: 
(cc @pattacini @davidetome )