robotology/icub-main #777: macOS iCub_SIM not executing

smwasif, 2021-12-02:
Hi dear,
I am new bee to macOS and i installed from robotology-superbuild using source installation by following step by step guide provided at this page. Installation was successful without any error in both conda and without conda installation. In both cases `**yarpserver** `started but whenever i executed `**iCub_SIM**` command I got this result:
**command not found: iCub_SIM**
Please help to address this issue or appropriate command to start `**iCub_SIM**`  in macOS.

smwasif, 2021-12-03:
Hi @traversaro 
Thank you very much for updating me regarding iCub_SIM legacy simulator. Actually, in past I used the legacy simulator in Ubuntu environment. Now i need help  for Gazebo-based iCub simulator usage in macOS.

traversaro, 2021-12-03:
> Now i need help for Gazebo-based iCub simulator usage in macOS.

Which kind of help do you need in particular? Once you installed the robotology-superbuild with `ROBOTOLOGY_USES_GAZEBO`, if you start `yarpserver` and then `gazebo`, you can see the iCub-related model of Gazebo on the left panel.

smwasif, 2021-12-03:
Hi @traversaro 
How to install ROS in macOS? more precisely whether should I install ros2 or version 1 ? There are number of pages available instructing its installation e.g., [https://docs.ros.org/en/foxy/Installation/macOS-Development-Setup.html](url). 
If you could please guide me specifically for ROS installation steps in macOS Monterey followed by gazebo based iCub simulator.

traversaro, 2021-12-03:
Hi @smwasif,
  ROS is not required to run the iCub on Gazebo simulation. I am not sure how are you installing your dependencies, but you can install gazebo on conda with:
~~~
conda install -c conda-forge gazebo
~~~
or with homebrew with:
~~~
brew install gazebo
~~~

traversaro, 2022-05-19:
Hi @smwasif, do  you still have problems or we can close this issue? Thanks!

pattacini, 2022-05-20:
Closing.
Feel free to open it up again if needed.

closed 2022-05-20 by pattacini