robotology/icub-main #784: Some libraries and executables could be placed in the wrong path during the build process

sgiraz, 2022-03-03:
### Bug description

**SW Configuration:**
- Windows 11
- Visual Studio Code 1.64.2 (with CMake and C++ extensions installed)
- Vcpkg package management program version 2022-02-24-7efe7f2b04912a7d674f86a406f9f2711c0221c6 (with robotology dependencies manually installed)
- Yarp@master version 3.6.102+20220302.2+git01b3de525 
- icub-firmware-shared@devel 1.24.0
- icub-main@devel version 1.24.0
  - ENABLE_icubmod_embObjMotionControl --> ON
  - ICUB_USE_icub_firmware_shared --> ON
  - ENABLE_icubmod_parametricCalibratorEth --> ON
- User Environment Variables:
  - YARP_DATA_DIRS=C:\dev\yarp\build\share\yarp;C:\dev\icub-main\build\share\iCub
  - YARP_DIR=C:\dev\yarp\build
  - YCM_DIR=C:\dev\ycm\build
  - Path=\C:...;C:\dev\yarp\build\bin\Debug;C:\dev\icub-main\build\bin\Debug;C:\dev\icub-firmware-shared\build\eth\Debug

---

**Description:**
Trying to run the `yarprobotinterface` the following error stops the execution:

```console
[ERROR] |yarp.os.YarpPluginSettings| Error while opening C:/dev/icub-main/build/lib/iCub/Debug/embObjMotionControld.dll:
  The specified module could not be found.

[WARNING] |yarp.os.YarpPluginSettings| Cannot load plugin from shared library (embObjMotionControl)
[WARNING] |yarp.os.YarpPluginSettings| (The specified module could not be found.
```

Thanks to @traversaro, using the [Dependencies][1] tool on `embObjMotionControld.dll`, I spotted that among all its dependencies the `ethResourcesd.dll` cannot be found (see the image below)

[image]

After the building process the `ethResourcesd.dll` library should be in `C:\dev\icub-main\build\bin\Debug` folder, but it was in `C:\dev\icub-main\build\lib\Debug` folder.

---

**How to solve**
In my case, the fix has been the introduction of the following variable inside the main [CMakeLists.txt][2]:

```diff
set_property(GLOBAL PROPERTY ICUB_INCLUDE_DIRS)
set_property(GLOBAL PROPERTY ICUB_TARGETS)
set_property(GLOBAL PROPERTY ICUB_DEPENDENCIES_FLAGS) # this is populated iCubFindDependencies.cmake

+set(CMAKE_RUNTIME_OUTPUT_DIRECTORY "${CMAKE_BINARY_DIR}/${CMAKE_INSTALL_BINDIR}")
+set(CMAKE_LIBRARY_OUTPUT_DIRECTORY "${CMAKE_BINARY_DIR}/${CMAKE_INSTALL_LIBDIR}")
+set(CMAKE_ARCHIVE_OUTPUT_DIRECTORY "${CMAKE_BINARY_DIR}/${CMAKE_INSTALL_LIBDIR}")
```

🔍 After a quick discussion with @pattacini, the problem seems to be inside the [CMakeLists.txt][3] of the `ethResources` project, which uses an old macro that should be updated with the new one ([here][4] is an example from iKin project).

[1]: https://github.com/lucasg/Dependencies/releases
[2]: https://github.com/robotology/icub-main/blob/devel/CMakeLists.txt
[3]: https://github.com/robotology/icub-main/blob/master/src/libraries/icubmod/embObjLib/CMakeLists.txt#L98
[4]: https://github.com/robotology/icub-main/blob/master/src/libraries/iKin/CMakeLists.txt#L59-L68

### Steps to reproduce

1. Install the tools and build the projects as written in the description above
2. Open a 2 Terminal windows/tab
3. Run
  - Terminal 1: `yarpserver`
  - Terminal 2: `yarprobotinterface`
4. Looking at the log of the `yarprobotinterface`, it should be as the following:
<details>
<summary>Click to expand</summary>

[... 74 lines cut ...]
[DEBUG] EthResource::verifyBoardPresence() found BOARD head-eb21-j2_5 with IP 10.0.1.1 after 0.0028385 seconds
[DEBUG] EthResource::verifyBoardTransceiver() has validated the transceiver of BOARD head-eb21-j2_5 with IP 10.0.1.1
[DEBUG] EthResource::cleanBoardBehaviour() has cleaned the application in BOARD head-eb21-j2_5 with IP 10.0.1.1 : config mode + cleared all its regulars
[DEBUG] EthResource::setTimingOfRunningCycle() for BOARD head-eb21-j2_5 with IP 10.0.1.1 has succesfully set: cycletime = 1000 usec, RX DO TX = ( 400 300 300 ) usec and TX rate = 5  every cycle
[INFO] EthResource::askBoardVersion() found BOARD head-eb21-j2_5 @ IP 10.0.1.1 of type mc4plus with FW = ver 3.48 built on 2022 Mar 02 21:11
[DEBUG]  from BOARD 10.0.1.1 (░Ä≥ Æ☺), src LOCAL, adr 0, time 27958s 406m 290u: (code 0x05000015, par16 0x1000 par64 0x0000000000000000) -> CFG: EOtheEncoderReader can be correctly configured. p16&0xf000: number of joint; primary encs: failure mask in p16&0x000f and errorcodes in p64&0x0000ffff; secondary encs: failure mask in p16&0x00f0 and errorcodes in p64&0xffff0000 + .
[DEBUG] embObjMotionControl:serviceVerifyActivate OK!
[DEBUG]  from BOARD 10.0.1.1 (░Ä≥ Æ☺), src LOCAL, adr 0, time 27958s 406m 567u: (code 0x05000013, par16 0x0000 par64 0x0000000000000000) -> CFG: EOtheMotionController can correctly configure mc4plus-based motion. more info will follow + .
[INFO] |yarp.dev.PolyDriver|head-eb21-j2_5-mc| Created device <embObjMotionControl>. See C++ class yarp::dev::embObjMotionControl for documentation.

...

[DEBUG] HeadV3_Calibrator : set 1 : Calibration ended, going to zero!
[DEBUG] HeadV3_Calibrator : Sending positionMove to joint 0  (desired pos:  0 desired speed:  10  )
[DEBUG] HeadV3_Calibrator: checkGoneToZeroThreshold: joint: 0 curr: 0.000 des: 0.000 -> delta: 0.000 threshold: 90.000 output: 0.000 mode: pos
[DEBUG] HeadV3_Calibrator: checkGoneToZeroThreshold: joint: 0 completed with delta: 0.000 over: 90.000
[DEBUG] HeadV3_Calibrator : set 1 : Reached zero position!
[DEBUG] head-calibrator finished calibration of device head-mc_wrapper
[INFO] All actions for action level 10 of startup phase finished.
[INFO] startup phase finished.
[INFO] run phase starting...
[DEBUG] yarprobotinterface running happily
```

</details>

pattacini, 2022-03-03:
Very nice bug report @sgiraz, thanks!

The old macro `icub_export_library` is also used in a few other places: https://github.com/robotology/icub-main/search?q=icub_export_library.

We should therefore address the fix extensively.

pattacini, 2022-03-03:
cc @marcoaccame

sgiraz, 2022-03-18:
Thanks to the suggestions provided by @traversaro and confirmations by @Nicogene the solution for the issue reported in here relies in the following 3 instructions applied on each libraries mentioned above by @pattacini:

```diff
+set_target_properties(${PROJECT_NAME} PROPERTIES RUNTIME_OUTPUT_DIRECTORY "${CMAKE_BINARY_DIR}/${CMAKE_INSTALL_BINDIR}")
+set_target_properties(${PROJECT_NAME} PROPERTIES LIBRARY_OUTPUT_DIRECTORY "${CMAKE_BINARY_DIR}/${CMAKE_INSTALL_LIBDIR}")
+set_target_properties(${PROJECT_NAME} PROPERTIES ARCHIVE_OUTPUT_DIRECTORY "${CMAKE_BINARY_DIR}/${CMAKE_INSTALL_LIBDIR}")
 ```

Talking with @Nicogene, we think that unless these libraries are necessary for other repositories, we could leave the old macros.
Otherwise, we can change them as suggested and then we have to test the installation on the real robot.

pattacini, 2022-03-18:
The solution proposed in https://github.com/robotology/icub-main/issues/784#issuecomment-1072618751 seems to be exactly the same as that reported in the OP with the difference that it acts at the level of the single libraries.

My point was:
- Why do all the other libraries (e.g., `ctrlLib`, `iKin`...) build flawlessly as intended without resorting to this change?
- Is there any particular difference among these libraries and the ones that do not build correctly?

Can we answer these questions ?

traversaro, 2022-03-18:
> Why do all the other libraries (e.g., `ctrlLib`, `iKin`...) build flawlessly as intended (meaning they put the products in the expected paths) without resorting to this change?

From what I understand, the question is rather why all other libraries (e.g., `ctrlLib`, `iKin`...) do not put their products as intended, as I do not understand why they should put their product there, even if indeed they do. :D However, let me check.

pattacini, 2022-03-18:
Yep, investigating the reason for this difference was my original aim. Applying the patch lightheartedly will quickly fix this particular problem but will potentially hide some others.

Nicogene, 2022-03-21:
> The solution proposed in [#784 (comment)](https://github.com/robotology/icub-main/issues/784#issuecomment-1072618751) seems to be exactly the same as that reported in the OP with the difference that it acts at the level of the single libraries.
> 
> My point was:
> 
>     * Why do all the other libraries (e.g., `ctrlLib`, `iKin`...) build flawlessly as intended (meaning they put the products in the expected paths) without resorting to this change?
> 
>     * Is there any particular difference among these libraries and the ones that do not behave correctly?
> 
> 
> Can we answer these questions?

I had the same doubts, the libraries of the embobj part have a quite old cmake structure (~ 9/12 years) and I hope that here https://github.com/robotology/icub-main/pull/652 I did not change something unintentionally. I probably forgot to test the case of using the build directory, that is probably the reason why @sgiraz spotted this problem only now.

>  while for some reason in this case all those libraries place its binaries in ${CMAKE_BINARY_DIR}/lib.

I thought that was that the default one 🤔 If not, I don't know who is setting ` ${CMAKE_BINARY_DIR}/lib`

traversaro, 2022-03-21:
> > while for some reason in this case all those libraries place its binaries in ${CMAKE_BINARY_DIR}/lib.
> 
> I thought that was that the default one 🤔

Actually no! You can try to compile https://github.com/robotology/how-to-export-cpp-library/ after commenting the three lines https://github.com/robotology/how-to-export-cpp-library/blob/afb21efb655e7b1cc11636c3d42aad9e02fb626f/CMakeLists.txt#L41 and checking where the libraries are placed.

PR #790 (2022-05-13): Fix placing of shared libraries in Windows
This PR fixes #784.

@sgiraz narrowed down the reason why part of the libraries was ok and part not. See below.

I've also reinstated the test stage of the CI on Windows.

---

After a quick comparison between commit [87cd0a1][1] and commit [b6e0f7b][2] I found the difference that caused the "fix".

In particular in [`src/libraries/icubmod/embObjLib/CMakeLists.txt`][3] we have the following difference:
```diff
- add_library(${PROJECT_NAME} SHARED ${embobj_source} ${embobj_header})
+ add_library(${PROJECT_NAME} ${embobj_source} ${embobj_header})
```

A [... cut]
commit 04fc6f1b (2022-05-13): fix #784
closed 2022-05-13 by pattacini