robotology/icub-main #796: Compilation failure on Windows. `` cannot open input file 'm.lib'``

S-Dafarra, 2022-05-20:
### Bug description

When installing the ``robotology-superbuild`` from scratch on Windows, I got the following error
```
(robsub) C:\robotology-superbuild\build\src\ICUB>cmake --build . --config Release
Microsoft (R) Build Engine version 16.6.0+5ff7b0c9e for .NET Framework
Copyright (C) Microsoft Corporation. All rights reserved.

  Automatic MOC for target QtICubSkinGuiPlugin
  QtICubSkinGuiPlugin.vcxproj -> C:\robotology-superbuild\build\src\ICUB\lib\qt5\qml\Release\robotology\icub\skingui\Qt
  ICubSkinGuiPlugin.dll
  ctrlLib.vcxproj -> C:\robotology-superbuild\build\src\ICUB\lib\Release\ctrlLib.lib
  iKin.vcxproj -> C:\robotology-superbuild\build\src\ICUB\lib\Release\iKin.lib
  cartesiancontrollerclient.vcxproj -> C:\robotology-superbuild\build\src\ICUB\lib\iCub\Release\cartesiancontrollerclie
  nt.dll
  cartesiancontrollerserver.vcxproj -> C:\robotology-superbuild\build\src\ICUB\lib\iCub\Release\cartesiancontrollerserv
  er.dll
  gazecontrollerclient.vcxproj -> C:\robotology-superbuild\build\src\ICUB\lib\iCub\Release\gazecontrollerclient.dll
  imuFilter.vcxproj -> C:\robotology-superbuild\build\src\ICUB\lib\iCub\Release\imuFilter.dll
  learningMachine.vcxproj -> C:\robotology-superbuild\build\src\ICUB\lib\Release\learningMachine.lib
  perceptiveModels.vcxproj -> C:\robotology-superbuild\build\src\ICUB\lib\Release\perceptiveModels.lib
  actionPrimitives.vcxproj -> C:\robotology-superbuild\build\src\ICUB\lib\Release\actionPrimitives.lib
  actionsRenderingEngine.vcxproj -> C:\robotology-superbuild\build\src\ICUB\bin\Release\actionsRenderingEngine.exe
  camCalib.vcxproj -> C:\robotology-superbuild\build\src\ICUB\bin\Release\camCalib.exe
  camCalibWithPose.vcxproj -> C:\robotology-superbuild\build\src\ICUB\bin\Release\camCalibWithPose.exe
  cartesianInterfaceControl.vcxproj -> C:\robotology-superbuild\build\src\ICUB\bin\Release\cartesianInterfaceControl.ex
  e
  iCubDev.vcxproj -> C:\robotology-superbuild\build\src\ICUB\lib\Release\iCubDev.lib
  controlBoardDumper.vcxproj -> C:\robotology-superbuild\build\src\ICUB\bin\Release\controlBoardDumper.exe
  ctpService.vcxproj -> C:\robotology-superbuild\build\src\ICUB\bin\Release\ctpService.exe
  optimization.vcxproj -> C:\robotology-superbuild\build\src\ICUB\lib\Release\optimization.lib
  depth2kin.vcxproj -> C:\robotology-superbuild\build\src\ICUB\bin\Release\depth2kin.exe
  dualCamCalib.vcxproj -> C:\robotology-superbuild\build\src\ICUB\bin\Release\dualCamCalib.exe
  emotionInterface.vcxproj -> C:\robotology-superbuild\build\src\ICUB\bin\Release\emotionInterface.exe
  fingersTuner.vcxproj -> C:\robotology-superbuild\build\src\ICUB\bin\Release\fingersTuner.exe
  Automatic MOC for target frameGrabberGui2
  frameGrabberGui2.vcxproj -> C:\robotology-superbuild\build\src\ICUB\bin\Release\frameGrabberGui2.exe
  skinDynLib.vcxproj -> C:\robotology-superbuild\build\src\ICUB\lib\Release\skinDynLib.lib
  iDyn.vcxproj -> C:\robotology-superbuild\build\src\ICUB\lib\Release\iDyn.lib
  gravityCompensator.vcxproj -> C:\robotology-superbuild\build\src\ICUB\bin\Release\gravityCompensator.exe
  Automatic MOC for target iCubGui
LINK : warning LNK4044: unrecognized option '/LC:/mambaforge/envs/robsub/Library/lib'; ignored [C:\robotology-superbuil
d\build\src\ICUB\src\tools\iCubGui\src\iCubGui.vcxproj]
LINK : warning LNK4044: unrecognized option '/lglut'; ignored [C:\robotology-superbuild\build\src\ICUB\src\tools\iCubGu
i\src\iCubGui.vcxproj]
LINK : warning LNK4044: unrecognized option '/lopengl32'; ignored [C:\robotology-superbuild\build\src\ICUB\src\tools\iC
ubGui\src\iCubGui.vcxproj]
LINK : warning LNK4044: unrecognized option '/lwinmm'; ignored [C:\robotology-superbuild\build\src\ICUB\src\tools\iCubG
ui\src\iCubGui.vcxproj]
LINK : warning LNK4044: unrecognized option '/lgdi32'; ignored [C:\robotology-superbuild\build\src\ICUB\src\tools\iCubG
ui\src\iCubGui.vcxproj]
LINK : warning LNK4044: unrecognized option '/lm'; ignored [C:\robotology-superbuild\build\src\ICUB\src\tools\iCubGui\s
rc\iCubGui.vcxproj]
LINK : fatal error LNK1181: cannot open input file 'm.lib' [C:\robotology-superbuild\build\src\ICUB\src\tools\iCubGui\s
rc\iCubGui.vcxproj]
```

cc @traversaro 

### Steps to reproduce

- Download the ``robotology-superbuild`` on Windows
- Switch to the branch https://github.com/robotology/robotology-superbuild/tree/releases/2022.02
- Compile (using ``conda`` dependencies)

traversaro, 2022-05-20:
It seems something that should have been fixed by https://github.com/conda-forge/freeglut-feedstock/pull/31 .

traversaro, 2022-05-20:
See also https://github.com/robotology/robotology-superbuild/issues/1032 .

S-Dafarra, 2022-05-20:
```
(robsub) C:\>mamba list

                  __    __    __    __
                 /  \  /  \  /  \  /  \
                /    \/    \/    \/    \
███████████████/  /██/  /██/  /██/  /████████████████████████
              /  / \   / \   / \   / \  \____
             /  /   \_/   \_/   \_/   \    o \__,
            / _/                       \_____/  `
            |/
        ███╗   ███╗ █████╗ ███╗   ███╗██████╗  █████╗
        ████╗ ████║██╔══██╗████╗ ████║██╔══██╗██╔══██╗
        ██╔████╔██║███████║██╔████╔██║██████╔╝███████║
        ██║╚██╔╝██║██╔══██║██║╚██╔╝██║██╔══██╗██╔══██║
        ██║ ╚═╝ ██║██║  ██║██║ ╚═╝ ██║██████╔╝██║  ██║
        ╚═╝     ╚═╝╚═╝  ╚═╝╚═╝     ╚═╝╚═════╝ ╚═╝  ╚═╝

        mamba (0.15.3) supported by @QuantStack

        GitHub:  https://github.com/mamba-org/mamba
        Twitter: https://twitter.com/QuantStack

█████████████████████████████████████████████████████████████

# packages in environment at C:\mambaforge\envs\robsub:
#
# Name                    Version                   Build  Channel
ace                       7.0.3                h0e60522_0    conda-forge
asio                      1.14.1               h0e60522_1    conda-forge
assimp                    5.1.0                hc2aa0de_1    conda-forge
boost                     1.74.0           py39hefe7e4c_3    conda-forge
boost-cpp                 1.74.0               h5b4e17d_4    conda-forge
bzip2                     1.0.8                h8ffe710_4    conda-forge
c-compiler                1.3.0                h8ffe710_0    conda-forge
ca-certificates           2022.5.18.1          h5b45459_0    conda-forge
cairo                     1.16.0            hb19e0ff_1008    conda-forge
cfitsio                   3.470                h0af3d06_7    conda-forge
clang_variant             1.0                     default    conda-forge
clangdev                  5.0.0             h5aa907a_1012    conda-forge
cmake                     3.23.0               h39d44d4_1    conda-forge
compilers                 1.3.0                h57928b3_0    conda-forge
console_bridge            1.0.1                h5362a0b_0    conda-forge
cppzmq                    4.8.1                h4324990_0    conda-forge
curl                      7.79.0               h789b8ee_0    conda-forge
cxx-compiler              1.3.0                h2d74725_0    conda-forge
dirent                    1.21                          0    conda-forge
dlfcn-win32               1.3.0                h0e60522_0    conda-forge
eigen                     3.4.0                h2d74725_0    conda-forge
expat                     2.4.1                h39d44d4_0    conda-forge
ffmpeg                    4.3.1                ha925a31_0    conda-forge
flang                     5.0.0           he025d50_20180525    conda-forge
flang_win-64              5.0.0           h13ae965_20180526    conda-forge
font-ttf-dejavu-sans-mono 2.37                 hab24e00_0    conda-forge
font-ttf-inconsolata      3.000                h77eed37_0    conda-forge
font-ttf-source-code-pro  2.038                h77eed37_0    conda-forge
font-ttf-ubuntu           0.83                 hab24e00_0    conda-forge
fontconfig                2.13.1            h1989441_1005    conda-forge
fonts-conda-ecosystem     1                             0    conda-forge
fonts-conda-forge         1                             0    conda-forge
fortran-compiler          1.3.0                h9655429_0    conda-forge
freeglut                  3.2.1                h0e60522_2    conda-forge
freeimage                 3.18.0               hfcb8cd7_7    conda-forge
freetype                  2.10.4               h546665d_1    conda-forge
freexl                    1.0.6                ha8e266a_0    conda-forge
fribidi                   1.0.10               h8d14728_0    conda-forge
gazebo                    11.8.1               h2e5bf29_0    conda-forge
geos                      3.9.1                h39d44d4_2    conda-forge
geotiff                   1.6.0                ha8a8a2d_6    conda-forge
getopt-win32              0.1                  h8ffe710_0    conda-forge
[... 133 lines cut ...]
vs2015_runtime            14.29.30037          h902a5da_5    conda-forge
vs2017_win-64             19.16.27038          h2e3bad8_2    conda-forge
vswhere                   2.8.4                h57928b3_0    conda-forge
wheel                     0.37.0             pyhd8ed1ab_1    conda-forge
xerces-c                  3.2.3                h0e60522_2    conda-forge
xorg-kbproto              1.0.7             hcd874cb_1002    conda-forge
xorg-libice               1.0.10               hcd874cb_0    conda-forge
xorg-libsm                1.2.3             hcd874cb_1000    conda-forge
xorg-libx11               1.7.2                hcd874cb_0    conda-forge
xorg-libxau               1.0.9                hcd874cb_0    conda-forge
xorg-libxdmcp             1.1.3                hcd874cb_0    conda-forge
xorg-libxext              1.3.4                hcd874cb_1    conda-forge
xorg-libxpm               3.5.13               hcd874cb_0    conda-forge
xorg-libxt                1.2.1                hcd874cb_2    conda-forge
xorg-xextproto            7.3.0             hcd874cb_1002    conda-forge
xorg-xproto               7.0.31            hcd874cb_1007    conda-forge
xz                        5.2.5                h62dcd97_1    conda-forge
yaml                      0.2.5                he774522_0    conda-forge
zeromq                    4.3.4                h0e60522_1    conda-forge
zlib                      1.2.11            h8ffe710_1013    conda-forge
zstd                      1.5.0                h6255e5f_0    conda-forge
zziplib                   0.13.69              h1d00b33_1    conda-forge
```

I guess I might have old versions then

S-Dafarra, 2022-05-20:
I wonder if it is worth suggesting to run 
```
mamba update --all
```
in https://github.com/robotology/robotology-superbuild/blob/releases/2022.02/doc/conda-forge.md

S-Dafarra, 2022-05-20:
Running ``mamba update --all`` and deleting the ``ICUB`` build fixed the issue.

pattacini, 2022-06-07:
Hi @S-Dafarra @traversaro 

Any progress on this?

S-Dafarra, 2022-06-07:
That was a problem of not update dependencies. I think we can close this.

traversaro, 2022-06-07:
Yep. There was still a doubt on how can we document "you should update your environment", but this is unrelated to this specific issue.

closed 2022-06-07 by traversaro