robotology/icub-main #797: Bug: undefined reference to yarp::dev::PolyDriver::PolyDriver

jhielson, 2022-05-21:
### Bug description

Hi everyone,

Please, help me with this issue. Every time I try to compile my code, I get the following error:

```
~/yarp_ws/motor_control_pinch$ make
Consolidate compiler generated dependencies of target motor
[ 50%] Building CXX object CMakeFiles/motor.dir/main.cpp.o
[100%] Linking CXX executable motor
/usr/bin/ld: CMakeFiles/motor.dir/main.cpp.o: warning: relocation against `_ZTIN4yarp3dev12DeviceDriverE' in read-only section `.text._ZN4yarp3dev12DeviceDriver4viewINS0_14ITorqueControlEEEbRPT_[_ZN4yarp3dev12DeviceDriver4viewINS0_14ITorqueControlEEEbRPT_]'
/usr/bin/ld: CMakeFiles/motor.dir/main.cpp.o: in function `main':
main.cpp:(.text+0xb72): undefined reference to `yarp::dev::PolyDriver::PolyDriver(yarp::os::Searchable&)'
/usr/bin/ld: main.cpp:(.text+0xb81): undefined reference to `yarp::dev::PolyDriver::isValid() const'
/usr/bin/ld: main.cpp:(.text+0xbba): undefined reference to `yarp::dev::PolyDriver::PolyDriver(yarp::os::Searchable&)'
/usr/bin/ld: main.cpp:(.text+0xbc9): undefined reference to `yarp::dev::PolyDriver::isValid() const'
/usr/bin/ld: main.cpp:(.text+0x1cfc): undefined reference to `yarp::dev::PolyDriver::~PolyDriver()'
/usr/bin/ld: main.cpp:(.text+0x1d0b): undefined reference to `yarp::dev::PolyDriver::~PolyDriver()'
/usr/bin/ld: main.cpp:(.text+0x22bd): undefined reference to `yarp::dev::PolyDriver::~PolyDriver()'
/usr/bin/ld: main.cpp:(.text+0x22d5): undefined reference to `yarp::dev::PolyDriver::~PolyDriver()'
/usr/bin/ld: CMakeFiles/motor.dir/main.cpp.o: in function `yarp::sig::VectorBase::VectorBase()':
main.cpp:(.text._ZN4yarp3sig10VectorBaseC2Ev[_ZN4yarp3sig10VectorBaseC5Ev]+0x1f): undefined reference to `vtable for yarp::sig::VectorBase'
/usr/bin/ld: main.cpp:(.text._ZN4yarp3sig10VectorBaseC2Ev[_ZN4yarp3sig10VectorBaseC5Ev]+0x2d): undefined reference to `vtable for yarp::sig::VectorBase'
/usr/bin/ld: CMakeFiles/motor.dir/main.cpp.o: in function `yarp::sig::VectorBase::~VectorBase()':
main.cpp:(.text._ZN4yarp3sig10VectorBaseD2Ev[_ZN4yarp3sig10VectorBaseD5Ev]+0x13): undefined reference to `vtable for yarp::sig::VectorBase'
/usr/bin/ld: main.cpp:(.text._ZN4yarp3sig10VectorBaseD2Ev[_ZN4yarp3sig10VectorBaseD5Ev]+0x21): undefined reference to `vtable for yarp::sig::VectorBase'
/usr/bin/ld: CMakeFiles/motor.dir/main.cpp.o: in function `bool yarp::dev::DeviceDriver::view<yarp::dev::IPositionControl>(yarp::dev::IPositionControl*&)':
main.cpp:(.text._ZN4yarp3dev12DeviceDriver4viewINS0_16IPositionControlEEEbRPT_[_ZN4yarp3dev12DeviceDriver4viewINS0_16IPositionControlEEEbRPT_]+0x4c): undefined reference to `typeinfo for yarp::dev::DeviceDriver'
/usr/bin/ld: CMakeFiles/motor.dir/main.cpp.o: in function `bool yarp::dev::DeviceDriver::view<yarp::dev::IEncoders>(yarp::dev::IEncoders*&)':
main.cpp:(.text._ZN4yarp3dev12DeviceDriver4viewINS0_9IEncodersEEEbRPT_[_ZN4yarp3dev12DeviceDriver4viewINS0_9IEncodersEEEbRPT_]+0x4c): undefined reference to `typeinfo for yarp::dev::DeviceDriver'
/usr/bin/ld: CMakeFiles/motor.dir/main.cpp.o: in function `bool yarp::dev::DeviceDriver::view<yarp::dev::IControlMode>(yarp::dev::IControlMode*&)':
main.cpp:(.text._ZN4yarp3dev12DeviceDriver4viewINS0_12IControlModeEEEbRPT_[_ZN4yarp3dev12DeviceDriver4viewINS0_12IControlModeEEEbRPT_]+0x4c): undefined reference to `typeinfo for yarp::dev::DeviceDriver'
/usr/bin/ld: CMakeFiles/motor.dir/main.cpp.o: in function `bool yarp::dev::DeviceDriver::view<yarp::dev::ITorqueControl>(yarp::dev::ITorqueControl*&)':
main.cpp:(.text._ZN4yarp3dev12DeviceDriver4viewINS0_14ITorqueControlEEEbRPT_[_ZN4yarp3dev12DeviceDriver4viewINS0_14ITorqueControlEEEbRPT_]+0x4c): undefined reference to `typeinfo for yarp::dev::DeviceDriver'
/usr/bin/ld: CMakeFiles/motor.dir/main.cpp.o:(.data.rel.ro._ZTVN4yarp3sig8VectorOfIdEE[_ZTVN4yarp3sig8VectorOfIdEE]+0x20): undefined reference to `yarp::sig::VectorBase::read(yarp::os::ConnectionReader&)'
/usr/bin/ld: CMakeFiles/motor.dir/main.cpp.o:(.data.rel.ro._ZTVN4yarp3sig8VectorOfIdEE[_ZTVN4yarp3sig8VectorOfIdEE]+0x30): undefined reference to `yarp::sig::VectorBase::write(yarp::os::ConnectionWriter&) const'
/usr/bin/ld: CMakeFiles/motor.dir/main.cpp.o:(.data.rel.ro._ZTVN4yarp3sig8VectorOfIdEE[_ZTVN4yarp3sig8VectorOfIdEE]+0x70): undefined reference to `yarp::sig::VectorBase::getFormatStr[abi:cxx11](int) const'
/usr/bin/ld: CMakeFiles/motor.dir/main.cpp.o:(.data.rel.ro._ZTVN4yarp3sig8VectorOfIdEE[_ZTVN4yarp3sig8VectorOfIdEE]+0x98): undefined reference to `non-virtual thunk to yarp::sig::VectorBase::write(yarp::os::ConnectionWriter&) const'
/usr/bin/ld: CMakeFiles/motor.dir/main.cpp.o:(.data.rel.ro._ZTIN4yarp3sig8VectorOfIdEE[_ZTIN4yarp3sig8VectorOfIdEE]+0x10): undefined reference to `typeinfo for yarp::sig::VectorBase'
/usr/bin/ld: warning: creating DT_TEXTREL in a PIE
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/motor.dir/build.make:99: motor] Error 1
make[1]: *** [CMakeFiles/Makefile2:83: CMakeFiles/motor.dir/all] Error 2
make: *** [Makefile:91: all] Error 2

``` 

That's my code: 

```
#include <iostream>
#include <fstream>
#include <string>
#include <cstdio>
#include <cmath>

#include <yarp/os/Node.h>
#include <yarp/os/LogComponent.h>
#include <yarp/os/LogStream.h>
#include <yarp/os/Network.h>
#include <yarp/os/Publisher.h>
#include <yarp/os/Subscriber.h>
#include <yarp/os/Time.h>
#include <yarp/dev/ControlBoardInterfaces.h>
#include <yarp/sig/Vector.h>
#include <yarp/rosmsg/std_msgs/String.h>
#include <yarp/dev/PolyDriver.h>

using yarp::os::Network;
[... 299 lines cut ...]
            command_right[joint] = command_right[joint] + adjust;
            // Clip if it is out of range 
            if(command_right[joint] < min) command_right[joint] = min;
            else if(command_right[joint] > max) command_right[joint] = max;
            // Stop previous command
            pos_right->stop(1);
            // Initiate new command
            std::cout << "\nStep: " << adjust << "\nNew position: " << command_right[joint] << "\nMin: " << min << " Max: " << max;  
            pos_right->positionMove(command_right.data());
              
        }
        	    
    }
    

    robotDevice_left.close();
    robotDevice_right.close();
     
    return 0;
}
```

### Steps to reproduce

I only try to compile my code and get the error above. Not sure how to solve it.

pattacini, 2022-05-23:
Hi @jhielson,

Your code isn't concerned with any possible bug related to `icub-main`, i.e. this repo.
My warm suggestion is thus that you open up a [Q&A discussion](https://github.com/robotology/community/discussions/new?category=q-a).

Alternatively, you could consider filing an issue on [`yarp`](https://github.com/robotology/yarp). In my view, a community discussion is a better fit though.

Remember that `Gists` come in handy for sharing your code, as per our [general rules for posting](https://github.com/robotology/community/blob/master/.github/SUPPORT.md#:~:text=If%20you%20would%20like%20us%20to%20review%20your%20code%2C%20please%20post%20your%20snippets%20to%20Gist%2C%20including%20instructions%20to%20run%20them.). In particular, you should post also the `CMakeLists.txt` file you used to configure the building phase. I suspect that you are not linking YARP libs.

closed 2022-05-23 by pattacini

jhielson, 2022-05-23:
Thank you! I have created a new issue on ```robotology/community``` as suggested (https://github.com/robotology/community/discussions/577).