robotology/icub-main #812: embObjFTsensor: SensorName and SensorFrameName should be different parameters

randaz81, 2022-07-01:
### Bug description

In `embObjFTsensor`  device driver: https://github.com/robotology/icub-main/blob/6a5922c7d6ef15e864486217b7ae4638805bd0d2/src/libraries/icubmod/embObjFTsensor/embObjFTsensor.cpp#L448-L458

The two parameters `name` and `frameName` should not be assigned to the same value.
Indeed: 
- sensor name is the UNIQUE name of the sensor.
- frame name is the name of the robot link to which the sensor is attached. Many sensors can be attached to the same frame link. The parameter is typically used in ROS message headers and should match a valid frame link name defined in the robot model (.sdf / .urdf). If wrongly assigned, ROS will not work properly.

### Steps to reproduce

- - -

### Expected behavior

The parser should handle two different parameters:
- `SensorName` should be a mandatory parameter (`id` is currently used in our robot .xml files. It's acceptable but, if possible, i'd recommend changing it in the future to `SensorId` or `SensorName`  because parameters like `id` or `name` can be used in different contexts by other attached devices (wrappers, remappers) and it's preferable to have unique names).
- `FrameName` should be preferably mandatory, but optional is acceptable with a warning message and a default value of "unknown_frame_name".

### Additional context

The `embObjFTsensor`  device is wrapped by:
- yarp:  https://github.com/robotology/yarp/tree/master/src/devices/multipleanalogsensorsserver 
- ros: https://github.com/robotology/yarp/tree/master/src/devices/multipleAnalogSensorsRosPublishers

pattacini, 2022-07-01:
We should also get `embObjMultipleFTsensors` covered.

cc @marcoaccame @triccyx

pattacini, 2022-07-02:
We will have time to review and correct this bug during PI12.

pattacini, 2022-07-02:
/remind July 25 plan on this

randaz81, 2022-09-09:
1- yes, framename is not currently present in xml files, and will be introduced by your modification
2 -yes, your xml is correct, the name of the frame can be chosen arbitrarily by you but I recommend matching the name defined in the gazebo sdf model (either of the sensor or the attached link)

traversaro, 2022-09-10:
I agree in general that it could be useful in some cases to separate SensorName and FrameName, but in the case of the icub  can't we use the same string for both? This is quite of important from the point of view of multipleanalogsensorremapper: the remapper uses the **sensorname** to permit users to remap devices (a bit like for controlboard we use joint names), but if you use different sensor names in simulated and real robots, we would need to mantain differnt configuration files for real and simulated robots, that I am afraid they will be a bit confusing to mantain.

pattacini, 2022-09-10:
>but if you use different sensor names in simulated and real robots, we would need to mantain different configuration files for real and simulated robots, that I am afraid they will be a bit confusing to maintain.

I would aim to avoid this 👍🏻

randaz81, 2022-09-12:
I agree too, simulated and real robot must be consistent. But this does not imply that framename and sensorname should be identical. It depends on your convenience and the current status of your urdf. The only constraint is that the frame name must be an existing frame of the transform tree, while the sensor name can be chosen arbitrarily (you can even keep your current one). However, I suspect that a revision of the urdf is necessary.
NB: This issue was opened because wrong names are currently preventing ROS to work properly on icub.

pattacini, 2022-09-12:
>NB: This issue was opened because wrong names are currently preventing ROS to work properly on icub.

Gotcha 👍🏻 

At any rate, what we're doing now is just applying a fix to the SW in order to be able to manage the two different parameters.

How we're going to update the URDF and the like can come at a later convenience after the PR is merged.

closed 2022-09-14 by pattacini