robotology/icub-main #840: malloc_consolidate(): invalid chunk size Aborted (core dumped) error when running yarpdev --list

traversaro, 2022-11-03:
### Bug description

Running `yarpdev --list` results in:

~~~
malloc_consolidate(): invalid chunk size
Aborted (core dumped) error when running
~~~

### Steps to reproduce

Rung `yarpdev --list` 

### Expected behavior

Not to see any error.

### Example repository

N.a.

traversaro, 2022-11-04:
The issue is related to the use of dynamic memory allocation (due to the use of `std::string`) in a static object instance (in particular `eth::EthBoards::defaultnames[eth::EthBoards::maxEthBoards]` and `eth::EthBoards::errorname[1]`, defined in https://github.com/robotology/icub-main/blob/v1.28.1/src/libraries/icubmod/embObjLib/ethBoards.cpp#L55) that is contained in a static library (`ethResources`, defined in https://github.com/robotology/icub-main/blob/v1.28.1/src/libraries/icubmod/embObjLib/CMakeLists.txt#L90).

traversaro, 2022-11-04:
Apparently this bug was already fixed in https://github.com/robotology/icub-main/pull/380, but then there was a regression in https://github.com/robotology/icub-main/pull/786 (see discussion https://github.com/robotology/icub-main/pull/786/files#r837687805 ). Probably in the next fix we should add a test to avoid regressions.

Probably we should find a way to fix this issue that does not introduce regressions w.r.t. to https://github.com/robotology/icub-main/pull/786, @triccyx @marcoaccame @Nicogene .

pattacini, 2023-07-31:
Hi @martinaxgloria

Just to start on this, it'd be great to check if `yarpdev --list` launched on the head of iCubGenova11 will still cause the problem.

martinaxgloria, 2023-07-31:
Hi @pattacini , I booked iCubGenova11 for this afternoon in order to do this check

martinaxgloria, 2023-07-31:
Hi @pattacini, I tried to launch `yarpdev --list` on iCubGenova11 and, after listing all the devices available, I obtained:

```
double free or corruption (fasttop)
Aborted (core dumped)

```

It is similar to the message reported by @traversaro in the body of this issue, but not the same, probably they are related.

pattacini, 2023-07-31:
Thanks for the check @martinaxgloria!
We will allocate some time for the investigation then.

pattacini, 2023-07-31:
/remind August 21

Nicogene, 2023-08-24:
To start off, we could try to run `valgrind yarpdev --list` after having compiled in debug, for detecing which are the faulty plugins.

martinaxgloria, 2023-08-25:
Yesterday, @sgiraz and I started investigating this issue. With valgring, this was the last part of the output:
```
==25629== HEAP SUMMARY:
==25629==     in use at exit: 370,894 bytes in 1,747 blocks
==25629==   total heap usage: 230,132 allocs, 228,649 frees, 17,234,258 bytes allocated
==25629==
==25629== LEAK SUMMARY:
==25629==    definitely lost: 6,456 bytes in 268 blocks
==25629==    indirectly lost: 1,328 bytes in 14 blocks
==25629==      possibly lost: 0 bytes in 0 blocks
==25629==    still reachable: 361,094 bytes in 1,444 blocks
==25629==         suppressed: 0 bytes in 0 blocks
==25629== Rerun with --leak-check=full to see details of leaked memory
==25629==
==25629== For lists of detected and suppressed errors, rerun with: -s
==25629== ERROR SUMMARY: 276 errors from 20 contexts (suppressed: 0 from 0)
```
Looking at the errors, they were all of type `Invalid free() / delete / delete[] / realloc()` and seem all to refer to the same line:
https://github.com/robotology/icub-main/blob/39a7e037e3d5dbdf94cdbd4046bd8c9a8cfb666d/src/libraries/icubmod/embObjLib/ethBoards.cpp#L55

which is the same reported by @traversaro [here](https://github.com/robotology/icub-main/issues/840#issuecomment-1303959240), even if the printed error when running yarpdev --list is not `malloc_consolidate(): invalid chunk size` anymore, but `double free or corruption (fasttop)`. 

At the end. we found a possible solution: `defaultnames[EthBoards::maxEthBoards]` and `errorname[1]` are defined as private in [ethBoards.h](https://github.com/robotology/icub-main/blob/master/src/libraries/icubmod/embObjLib/ethBoards.h) and then initialized in [ethBoards.cpp](https://github.com/robotology/icub-main/blob/master/src/libraries/icubmod/embObjLib/ethBoards.cpp), where the problem occurred. We tried to do the declaration and initialization both into the .h file but we had this error during compilation:
```shell
error: in-class initialization of static data member ‘const string eth::EthBoards::defaultnames [32]’ of non-literal type
   99 |         static const string defaultnames[EthBoards::maxEthBoards] = {"noname-in-xml-board.1", "noname-in-xml-board.2", "noname-in-xml-board.3", "noname-in-xml-board.4",
      |                             ^~~~~~~~~~~~
/home/martinagloria/robotology-superbuild/src/ICUB/src/libraries/icubmod/embObjLib/ethBoards.h:106:111: error: non-constant in-class initialization invalid for non-inline static member ‘eth::EthBoards::defaultnames’
  106 |         "noname-in-xml-board.29", "noname-in-xml-board.30", "noname-in-xml-board.31", "noname-in-xml-board.32"};
      |                                                                                                               ^
```

We found out that this can be done only with int types, not with strings. A possible workaround is to use `inline` before `static const` like:
````
inline static const string defaultnames[EthBoards::maxEthBoards] = {"noname-in-xml-board.1", "noname-in-xml-board.2", "noname-in-xml-board.3", "noname-in-xml-board.4",
        "noname-in-xml-board.5", "noname-in-xml-board.6", "noname-in-xml-board.7", "noname-in-xml-board.8",
        "noname-in-xml-board.9", "noname-in-xml-board.10", "noname-in-xml-board.11", "noname-in-xml-board.12",
        "noname-in-xml-board.13", "noname-in-xml-board.14", "noname-in-xml-board.15", "noname-in-xml-board.16",
        "noname-in-xml-board.17", "noname-in-xml-board.18", "noname-in-xml-board.19", "noname-in-xml-board.20",
        "noname-in-xml-board.21", "noname-in-xml-board.22", "noname-in-xml-board.23", "noname-in-xml-board.24",
        "noname-in-xml-board.25", "noname-in-xml-board.26", "noname-in-xml-board.27", "noname-in-xml-board.28",
        "noname-in-xml-board.29", "noname-in-xml-board.30", "noname-in-xml-board.31", "noname-in-xml-board.32"};
        
inline static const string errorname[1] = {"wrong-unknown-board"};

This solution works without any error at compilation time and running `yarpdev --list` we don't have the error reported above anymore.
Thanks @sgiraz for the precious help!

cc @traversaro @Nicogene @pattacini

pattacini, 2023-08-28:
Cool 👍🏻 
You may consider opening up a PR for this, @martinaxgloria

martinaxgloria, 2023-08-28:
Hi @pattacini, I'm going to do it right now

pattacini, 2023-08-28:
Since it's fix, please target the `master` branch.

martinaxgloria, 2023-08-28:
Here the PR:

- https://github.com/robotology/icub-main/pull/900

cc @traversaro @Nicogene @pattacini @marcoaccame

PR #900 (2023-09-04): Fix 'double free or corruption' error on yarpdev --list
This PR should fix #840. 

Together with @sgiraz, we found out that the `double free or corruption` error that was retrieved when running `yarpdev --list` may occur in `eth::EthBoards` class. In particular, it seems that the destructor of this class is responsible for destroying some members of this class, but then another function is trying to free memory that was already freed. 

@marcoaccame what do you think about it? Could you please have a look?
closed 2023-09-04 by pattacini