robotology/icub-main #842: iCub_SIM not compiling against YARP 3.8

traversaro, 2022-11-24:
### Bug description

Compiling the latest icub-main with the latest YARP master, iCub_SIM fails with:
~~~
2022-11-24T19:54:03.9891162Z [334/477] Building CXX object src/simulators/iCubSimulation/CMakeFiles/iCub_SIM.dir/wrapper/SimulationRun.cpp.o
2022-11-24T19:54:03.9891356Z FAILED: src/simulators/iCubSimulation/CMakeFiles/iCub_SIM.dir/wrapper/SimulationRun.cpp.o 
2022-11-24T19:54:03.9895157Z /usr/share/miniconda3/envs/test/bin/x86_64-conda-linux-gnu-c++ -DICUB_SIM_ENABLE_FAKE -DICUB_SIM_ENABLE_ODESDL -DOMIT_LOGPOLAR -DUSE_GLUT -D_USE_MATH_DEFINES -DdDOUBLE -I/home/runner/work/robotology-superbuild/robotology-superbuild/src/ICUB/src/simulators/iCubSimulation -I/home/runner/work/robotology-superbuild/robotology-superbuild/src/ICUB/src/simulators/iCubSimulation/wrapper -I/home/runner/work/robotology-superbuild/robotology-superbuild/src/ICUB/src/simulators/iCubSimulation/fake -I/usr/share/miniconda3/envs/test/include/SDL -I/home/runner/work/robotology-superbuild/robotology-superbuild/src/ICUB/src/simulators/iCubSimulation/odesdl -I/home/runner/work/robotology-superbuild/robotology-superbuild/src/ICUB/src/libraries/iKin/include -I/home/runner/work/robotology-superbuild/robotology-superbuild/src/ICUB/src/libraries/ctrlLib/include -I/home/runner/work/robotology-superbuild/robotology-superbuild/src/ICUB/src/libraries/skinDynLib/include -isystem /home/runner/work/robotology-superbuild/robotology-superbuild/build/install/include -fvisibility-inlines-hidden -std=c++17 -fmessage-length=0 -march=nocona -mtune=haswell -ftree-vectorize -fPIC -fstack-protector-strong -fno-plt -O2 -ffunction-sections -pipe -isystem /usr/share/miniconda3/envs/test/include -O3 -DNDEBUG -fPIE -std=c++17 -MD -MT src/simulators/iCubSimulation/CMakeFiles/iCub_SIM.dir/wrapper/SimulationRun.cpp.o -MF src/simulators/iCubSimulation/CMakeFiles/iCub_SIM.dir/wrapper/SimulationRun.cpp.o.d -o src/simulators/iCubSimulation/CMakeFiles/iCub_SIM.dir/wrapper/SimulationRun.cpp.o -c /home/runner/work/robotology-superbuild/robotology-superbuild/src/ICUB/src/simulators/iCubSimulation/wrapper/SimulationRun.cpp
2022-11-24T19:54:03.9895760Z /home/runner/work/robotology-superbuild/robotology-superbuild/src/ICUB/src/simulators/iCubSimulation/wrapper/SimulationRun.cpp:23:10: fatal error: yarp/dev/DriverLinkCreator.h: No such file or directory
2022-11-24T19:54:03.9895889Z    23 | #include <yarp/dev/DriverLinkCreator.h>
2022-11-24T19:54:03.9895987Z       |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
2022-11-24T19:54:03.9896085Z compilation terminated.
~~~

These seems related to https://github.com/robotology/yarp/pull/2903 .

### Steps to reproduce

.

randaz81, 2022-11-24:
yep, that piece of code needs some cleanup, too. I can take care of it.

pattacini, 2022-11-25:
I had a couple of discussions w/ @traversaro and @randaz81 and the idea is to remove the glorious yet old iCub_SIM (already deprecated) from icub-main and store it as a legacy code for the record.

Thus, Distro 2022.11.0 will be the last one providing a functioning iCub_SIM. That said, I'm going to remove the old simulator soon after the distro delivery (cc @robotology/everyone).

Considering that 2022.11.0 will still resort to yarp-3.7.2 and that fixing this bug in iCub_SIM to still get it working requires some effort (checked w/ @randaz81), the proposal is that we simply ignore the bug.

The nasty downside will be that the superbuild CI's covering the unstable branches will most likely fail until 2022.11.0 is out. Is this acceptable for you @traversaro?

traversaro, 2022-11-25:
> The nasty downside will be that the superbuild CI's covering the unstable branches will most likely fail until 2022.11.0 is out. Is this acceptable for you @traversaro @Nicogene?

I will disable iCub_Sim from unstable for the time being.

traversaro, 2022-11-25:
> > The nasty downside will be that the superbuild CI's covering the unstable branches will most likely fail until 2022.11.0 is out. Is this acceptable for you @traversaro @Nicogene?
> 
> I will disable iCub_Sim from unstable for the time being.

Done in https://github.com/robotology/robotology-superbuild/pull/1309 .

pattacini, 2022-11-25:
Thank you @traversaro!

closed 2022-11-25 by pattacini

alecive, 2022-11-28:
End of an era!