robotology/icub-main #851: Expose information provided by embObjMAIS YARP device using a non-deprecated interface

traversaro, 2023-01-24:
### Is your feature request related to a problem?

I noticed that the `embObjMAIS` device is currently exposing its information via the `IAnalogSensor` YARP interface, that anyhow has been deprecated (see https://www.yarp.it/latest/group__dev__impl__deprecated.html). 

Probably we could consider to expose the info via either:
* [`yarp::dev::IEncoderArrays`](https://www.yarp.it/latest/classyarp_1_1dev_1_1IEncoderArrays.html): interface that is part of the multiple analog sensor interfaces, however has the downside that you cannot expose a name for each encoder
* [`yarp::dev::IAxisInfo`](https://www.yarp.it/latest/classyarp_1_1dev_1_1IAxisInfo.html) and [`yarp::dev::IEncoders`](https://www.yarp.it/latest/classyarp_1_1dev_1_1IEncoders.html) these interfaces are the same used to expose "regular"  joint encoders, but I am not sure that the `nws`  and `nwc`  work fine if one implements only these interfaces and not all the rest of control-related interfaces.

### The solution you would like to have available

See feature description.

### Alternatives you have considered

See feature description.

### Additional context

I realized this while thinking about https://github.com/orgs/robotology/discussions/609 .

pattacini, 2023-01-25:
Regarding the `analog` services, we would need first to understand how to upgrade them to the use of the latest NWS/NWC layers.

cc @randaz81

pattacini, 2023-08-02:
We will need to address this task as per https://github.com/robotology/yarp/pull/2999#issuecomment-1662253604.

cc @Nicogene @traversaro @randaz81

pattacini, 2023-08-02:
/remind August 28

Nicogene, 2023-09-04:
Work started:

- https://github.com/robotology/icub-main/pull/905

I will test it on `iCubGenova11` tomorrow afternoon

pattacini, 2023-09-05:
Great!

PR #905 (2023-09-05): embObjMais: expose data via MAS interface
It fixes #851.

Not sure if this implementation makes sense @marcoaccame @traversaro.
If it is ok I will test it on `iCubGenova11`
closed 2023-09-05 by pattacini