robotology/icub-main #852: Expose sensorName parameter for embObjIMU

Nicogene, 2023-01-27:
### Is your feature request related to a problem?

Right now this is the list of parameters associated with an `embObjIMU` device:

https://github.com/robotology/icub-main/blob/1dccde4859da2d0068666375541211d627a969aa/src/libraries/icubmod/embObjLib/serviceParser.h#L68-L75

In particular `id` is used also as the return of `get<type>FrameName` and `get<type>SensorName`. It has to be unique since each id is referred to a different logical sensor (e.g. acc, gyro, eul, magn) but usually the measurements of all the logical sensors are referred to a unique frame.

### The solution you would like to have available

In this sense, it came out necessary to specify a parameter called `sensorName` that groups all the logical sensors under one name, and use it as the return of  `get<type>FrameName` and `get<type>SensorName`.

### Alternatives you have considered

An alternative is to implement a separate `frameName` parameter as did #822

### Additional context

See https://github.com/icub-tech-iit/ergocub-software/issues/36

Nicogene, 2023-01-30:
#853 contains the changes for adding `sensorName` as parameter of `embObjIMU` device, https://github.com/robotology/robots-configuration/pull/422 contains the changes for the XML files (for now done only for the `rfe`).

The new parameter is optional, then can be added where is needed.

Test on `iCubGenova11` scheduled for tomorrow 31/01/2023 (cc @xEnVrE)

pattacini, 2023-01-31:
PR's merged ✅

PR #853 (2023-01-31): embObjIMU: add sensorName param
`sensorName` is **optional**, if defined it is used as return of `get<type>SensorName` and for now also `get<type>FrameName`, (we should in the future also add `frameName`  as done #822 ) otherwise `id` is used.

It fixes #852 

cc @traversaro @marcoaccame @pattacini
closed 2023-01-31 by pattacini