robotology/icub-main #864: Deal with modules still relying on `FrameGrabberInterfaces.h`

pattacini, 2023-04-17:
### Bug description

The I/F `yarp/dev/FrameGrabberInterfaces.h` has been declared deprecated since `YARP-3.5.0`.
However, we do still have a few modules depending on it:
- `frameGrabberGui2`
- `dragonfly2`

### Steps to reproduce

See https://github.com/search?q=repo%3Arobotology%2Ficub-main%20FrameGrabberInterfaces&type=code.

pattacini, 2023-04-17:
cc @Nicogene

pattacini, 2023-05-19:
See also https://github.com/orgs/robotology/discussions/621#discussioncomment-5940831.

Nicogene, 2023-05-29:
The only device that still relies on deprecated interfaces is `dragonfly2`, but the methods of those interfaces are not used since a while.

see #882 

> See also [github.com/orgs/robotology/discussions/621#discussioncomment-5940831](https://github.com/orgs/robotology/discussions/621#discussioncomment-5940831).

I am afraid this is related to something else.

PR #882 (2023-05-29): dragonfly2: remove deprecated interfaces
It fixes #864

In YARP 3.5.0 the `FrameGrabberInterfaces` header has been deprecated, but the interfaces included not, except for very old interfaces such as `IFrameGrabber` and `iFrameGrabberRgb`, that were implemented but not used by anyone. The `frameGrabber_nws_yarp` uses the "new" interfaces.
closed 2023-05-29 by pattacini