robotology/icub-main #899: Sole skin layout in configuration files and kinematic model do not match

BoPot, 2023-08-15:
### Bug description

The triangle layout in both right and left sole skin .ini files ([right_foot.ini](https://github.com/robotology/icub-main/blob/master/app/skinGui/conf/skinGui/right_foot.ini) and [left_foot.ini](https://github.com/robotology/icub-main/blob/master/app/skinGui/conf/skinGui/left_foot.ini)) is different from their layout in [model.urdf](https://github.com/robotology/icub-models/blob/master/iCub/robots/iCubGazeboV2_7/model.urdf)
More precisely, the top patch is differently rotated.

@matejhof @rustlluk @tra

### Steps to reproduce

My code for plotting taxels based on triangle positions from .ini file: [read_ini.py](https://gitlab.fel.cvut.cz/body-schema/icub/icub-skin-calibration/-/blob/main/skin_models/read_ini.py)
My code for plotting taxels based on triangle positions from .urdf file: [skin_from_urdf.py](https://gitlab.fel.cvut.cz/body-schema/icub/icub-skin-calibration/-/blob/main/skin_models/skin_from_urdf.py)

### Additional context

Results of the visualization for left foot (see the difference in top patch):

Positions from left_foot.ini
[image]

Positions from model.urdf
[image]

Also compared the visualizations to the picture inserted in ini generator [left_foot_ini_generator.xlsx](https://github.com/robotology/icub-main/blob/master/app/skinGui/iniGenerators/left_foot_ini_generator.xlsx) (see the "top" sheet). The layout there corresponds to the one in model.urdf
[image]

pattacini, 2023-08-17:
cc @maggia80 @fiorisi @simeonedussoni

maggia80, 2023-08-21:
@BoPot you are right there is a mismatch from the assembly notes: 
[AN2013_ICUB_FOOT_0001.pdf](https://github.com/robotology/icub-main/files/12393888/AN2013_ICUB_FOOT_0001.pdf)
 and the ini generator file.

The position you reconstruct is the correct one wrt the real robot.

pattacini, 2023-08-22:
Hi @BoPot 

We've planned to work on this fix at the beginning of September.
I think this is not blocking you anyhow.

/remind September 04.

rustlluk, 2023-08-31:
Hi all,
We have been testing position files extracted from the URDF model, and this issue may be more complicated.

Triangle IDs in our test are like this: URDF = assembly notes and .ini = real robot. 
See the attached video. When touching the left foot, skinGui shows the correct movement, but visualization that uses position files extracted from URDF has the triangles in wrong order. When the triangles in position files are reordered based on triangle IDs from the .ini files, it works as expected.
 
 So, it may be worth testing it on some of your robots before making changes to the .ini files.

https://github.com/robotology/icub-main/assets/54062527/93eec558-7cdb-4e38-8e2f-053208b52047

pattacini, 2023-09-14:
Related PR:
- https://github.com/robotology/icub-models-generator/pull/248

pattacini, 2023-10-30:
PR merged with this follow-up:
- #912

closed 2023-10-30 by pattacini