robotology/icub-main #908: Update icub-main modules to the use of MAS for MAIS boards

Nicogene, 2023-09-13:
### The solution you would like to have available

Following-up of #851, right now the MAIS are published via `multipleanalogsensorsserver`, then we need to adapt the code in order to not use anymore the `analog:o` port but the `multipleanalogsensorsclient`.

pattacini, 2023-09-18:
I'll deal with the module `depth2kin`, @davidetome.
If there's something else to address, we will coordinate.

pattacini, 2023-09-20:
PR opened:
- https://github.com/robotology/icub-main/issues/908

pattacini, 2023-09-20:
There shouldn't be any other component requiring update.

closed 2023-09-20 by pattacini