robotology/icub-main #913: embObj: diagnostic prints strange symbols under Windows

Nicogene, 2023-11-13:
### Bug description

Today I ran `yarprobotinterface` in windows and I got a log like this:

```
[INFO] from BOARD 10.0.1.1 (á↔╕╬╧☺
[INFO] from BOARD 10.0.1.1 (á↔╕╬╧☺
[INFO] from BOARD 10.0.1.1 (á↔╕╬╧☺
[DEBUG] from BOARD 10.0.1.1 (á↔╕╬╧☺
```

This regression has been introduced probably after #886 

cc @valegagge @sg

### Steps to reproduce

Run `yarprobotinterface` under windows

sgiraz, 2023-11-13:
Hi @Nicogene, 

I would remark that the issue is strictly related to the `Windows` systems (despite this, not less important Imo).
On `Ubuntu`, the diagnostic is shown correctly.

pattacini, 2023-11-13:
Maybe @valegagge and/or @MSECode can take on this.

valegagge, 2023-12-04:
Hi @Nicogene,
could you explain better which type of service you run? Did you run with gazebo or with the `dryrun` option?
Could you attach here the entire log? thanks

MSECode, 2024-03-06:
I saw that onmy Windows I get the following:

[image]

while it should be like this, as in Ubuntu:

[image]

I suppose the problem is with the UTF-8 or something related to that but I'm not sure how to solve it, since we are not writing weird chars.
Could be due to some libraries that are different for Windows??

cc: @valegagge @Nicogene

sgiraz, 2024-03-06:
Hi @MSECode,

I saw that [m_dnginfo.baseMessage][1] is an `std::string` type.
The `std::string::c_str()` method 

> I suppose the problem is with the UTF-8 or something related to that

It is possible, In the following Reddit discussion you can find multiple explanations and solutions: https://www.reddit.com/r/cpp_questions/comments/ov1xqw/paths_with_nonascii_chars_on_windows/

A similar behavior is happening here: https://stackoverflow.com/questions/10227806/c-c-str-adding-strange-characters-to-the-end-of-the-string, where the `c_str()` is used in combination w/ `strcpy`.

[1]: https://github.com/robotology/icub-main/blob/4595c48eb71a0ce64d61ea309bea329fb0c8bff7/src/libraries/icubmod/embObjLib/diagnosticLowLevelFormatter_hid.h#L49C1-L50C1

MSECode, 2024-03-11:
I'm trying to find the cause of the error but it is impossible to work on Windows.
If I'm compiling the whole robotology-superbuild on conda it is failing with a different error each time I try to do it.
Today I was able to start to work after compiling all the repos separately using the vcpkg but now, after I just change one line of code on icub-main printing a string with `.c_str()` instead of using the `yDebug()` with an std::string (just to check if we have issue with that), I'm not able to re-compile neither YARP nor icub-main with the following error:

[image]

This is ridiculous since with YARP I'm working on the main branch. Even if trying to checkout on the last tag is throwing me the same error. I cannot understand why on windows we have such error and the compilations cannot work multiple days in the same manner even if I'm not changing a single line of code.

Therefore, I cannot go on with this issue until I can be sure to have a stable environment, which I can work. I'm loosing lots of time just trying to compile code on windows and this does not make sense to me.

cc: @valegagge @pattacini

traversaro, 2024-03-11:
> If I'm compiling the whole robotology-superbuild on conda it is failing with a different error each time I try to do it.

Can you report at least one of this errors? We have a CI compiling nightly the robotology-superbuild with conda and several AMI users (mainly related to teloperation) that use it, so in theory the compilation should work fine, but obviously the devil is in the details.
Our typical workflow is to try to debug problems that users are having, and if necessary we improve the documentation with feedback from the users.

> Today I was able to start to work after compiling all the repos separately using the vcpkg but now, after I just change one line of code on icub-main printing a string with .c_str() instead of using [...]

In my experience vcpkg tend to be less user friendly and less repeatable then conda, but also there we still have the CI running nightly, so I do not expect any fundamental problem with the code. Did you use the pre-compiled vcpkg binaries available at https://github.com/robotology/robotology-superbuild-dependencies-vcpkg, or you compile the vcpkg dependencies on your own? That specific error seems to be a problem in OpenCV's cmake config file, but it is hard to know without more details on the vcpkg setup. If you have any question feel free to pass by my desk or ping me on Teams.

(Nitpick: if you could copy&paste errors in issue, then it would be easier for future users that may experience the same error to find related issue)

MSECode, 2024-03-20:
The problem related to the printing on Windows of the weird chars seems to be solved. I've tested on both Windows and Ubuntu thus to check we did not loose any compatibility with respect to before. I'm adding all the considerations and screenshots to the issue that wraps this tickets.

@valegagge

closed 2024-04-08 by pattacini