robotology/icub-main #922: ICUB devel does not compile against YARP master (December 2023)

traversaro, 2023-12-18:
### Bug description

This is not an urgent issue, but I wanted to report it as soon as possible to avoid that incompatibilities to pile up.

Due to the change of name of `yarp::os::IConfig` that was renamed to `yarp::os::IOpenClose` in https://github.com/robotology/yarp/pull/3059, we have the current compilation failures:
~~~
[1/178] Building CXX object src/libraries/learningMachine/CMakeFiles/learningMachine.dir/src/IFixedSizeTransformer.cpp.o
FAILED: src/libraries/learningMachine/CMakeFiles/learningMachine.dir/src/IFixedSizeTransformer.cpp.o
/usr/bin/c++  -I/home/traversaro/robotology-superbuild/src/ICUB/src/libraries/learningMachine/include -isystem /home/traversaro/robotology-superbuild/build-apt/install/include -O3 -DNDEBUG -fPIC -std=c++17 -MD -MT src/libraries/learningMachine/CMakeFiles/learningMachine.dir/src/IFixedSizeTransformer.cpp.o -MF src/libraries/learningMachine/CMakeFiles/learningMachine.dir/src/IFixedSizeTransformer.cpp.o.d -o src/libraries/learningMachine/CMakeFiles/learningMachine.dir/src/IFixedSizeTransformer.cpp.o -c /home/traversaro/robotology-superbuild/src/ICUB/src/libraries/learningMachine/src/IFixedSizeTransformer.cpp
In file included from /home/traversaro/robotology-superbuild/src/ICUB/src/libraries/learningMachine/include/iCub/learningMachine/IFixedSizeTransformer.h:22,
                 from /home/traversaro/robotology-superbuild/src/ICUB/src/libraries/learningMachine/src/IFixedSizeTransformer.cpp:22:
/home/traversaro/robotology-superbuild/src/ICUB/src/libraries/learningMachine/include/iCub/learningMachine/ITransformer.h:25:10: fatal error: yarp/os/IConfig.h: No such file or directory
   25 | #include <yarp/os/IConfig.h>
      |          ^~~~~~~~~~~~~~~~~~~
compilation terminated.
[2/178] Building CXX object src/libraries/learningMachine/CMakeFiles/learningMachine.dir/src/Normalizer.cpp.o
FAILED: src/libraries/learningMachine/CMakeFiles/learningMachine.dir/src/Normalizer.cpp.o
/usr/bin/c++  -I/home/traversaro/robotology-superbuild/src/ICUB/src/libraries/learningMachine/include -isystem /home/traversaro/robotology-superbuild/build-apt/install/include -O3 -DNDEBUG -fPIC -std=c++17 -MD -MT src/libraries/learningMachine/CMakeFiles/learningMachine.dir/src/Normalizer.cpp.o -MF src/libraries/learningMachine/CMakeFiles/learningMachine.dir/src/Normalizer.cpp.o.d -o src/libraries/learningMachine/CMakeFiles/learningMachine.dir/src/Normalizer.cpp.o -c /home/traversaro/robotology-superbuild/src/ICUB/src/libraries/learningMachine/src/Normalizer.cpp
In file included from /home/traversaro/robotology-superbuild/src/ICUB/src/libraries/learningMachine/include/iCub/learningMachine/Normalizer.h:22,
                 from /home/traversaro/robotology-superbuild/src/ICUB/src/libraries/learningMachine/src/Normalizer.cpp:24:
/home/traversaro/robotology-superbuild/src/ICUB/src/libraries/learningMachine/include/iCub/learningMachine/IScaler.h:24:10: fatal error: yarp/os/IConfig.h: No such file or directory
   24 | #include <yarp/os/IConfig.h>
      |          ^~~~~~~~~~~~~~~~~~~
compilation terminated.
[3/178] Building CXX object src/libraries/learningMachine/CMakeFiles/learningMachine.dir/src/FixedRangeScaler.cpp.o
FAILED: src/libraries/learningMachine/CMakeFiles/learningMachine.dir/src/FixedRangeScaler.cpp.o
/usr/bin/c++  -I/home/traversaro/robotology-superbuild/src/ICUB/src/libraries/learningMachine/include -isystem /home/traversaro/robotology-superbuild/build-apt/install/include -O3 -DNDEBUG -fPIC -std=c++17 -MD -MT src/libraries/learningMachine/CMakeFiles/learningMachine.dir/src/FixedRangeScaler.cpp.o -MF src/libraries/learningMachine/CMakeFiles/learningMachine.dir/src/FixedRangeScaler.cpp.o.d -o src/libraries/learningMachine/CMakeFiles/learningMachine.dir/src/FixedRangeScaler.cpp.o -c /home/traversaro/robotology-superbuild/src/ICUB/src/libraries/learningMachine/src/FixedRangeScaler.cpp
In file included from /home/traversaro/robotology-superbuild/src/ICUB/src/libraries/learningMachine/include/iCub/learningMachine/FixedRangeScaler.h:22,
                 from /home/traversaro/robotology-superbuild/src/ICUB/src/libraries/learningMachine/src/FixedRangeScaler.cpp:24:
/home/traversaro/robotology-superbuild/src/ICUB/src/libraries/learningMachine/include/iCub/learningMachine/IScaler.h:24:10: fatal error: yarp/os/IConfig.h: No such file or directory
   24 | #include <yarp/os/IConfig.h>
      |          ^~~~~~~~~~~~~~~~~~~
compilation terminated.
~~~

### Steps to reproduce

Compile icub-main devel against YARP master

### Expected behavior

That compilation is successful.

### Example repository

.

### Additional context

Probably the easiest solution is just to remove the inheritance from IConfig interface.

traversaro, 2023-12-18:
fyi @pattacini @randaz81 @Nicogene

traversaro, 2023-12-18:
> Probably the easiest solution is just to remove the inheritance from IConfig interface.

I implemented this in https://github.com/robotology/icub-main/pull/923, but now I have a few more failures:
~~~
FAILED: src/modules/skinManager/CMakeFiles/skinManager.dir/src/compensator.cpp.o
/usr/bin/c++ -D_USE_MATH_DEFINES -I/home/traversaro/robotology-superbuild/src/ICUB/src/modules/skinManager/include -I/home/traversaro/robotology-superbuild/src/ICUB/src/libraries/icubmod/skinLib -I/home/traversaro/robotology-superbuild/src/ICUB/src/libraries/skinDynLib/include -I/home/traversaro/robotology-superbuild/src/ICUB/src/libraries/ctrlLib/include -isystem /home/traversaro/robotology-superbuild/build-apt/install/include -O3 -DNDEBUG -fPIE -std=c++17 -MD -MT src/modules/skinManager/CMakeFiles/skinManager.dir/src/compensator.cpp.o -MF src/modules/skinManager/CMakeFiles/skinManager.dir/src/compensator.cpp.o.d -o src/modules/skinManager/CMakeFiles/skinManager.dir/src/compensator.cpp.o -c /home/traversaro/robotology-superbuild/src/ICUB/src/modules/skinManager/src/compensator.cpp
In file included from /home/traversaro/robotology-superbuild/src/ICUB/src/modules/skinManager/src/compensator.cpp:23:
/home/traversaro/robotology-superbuild/src/ICUB/src/modules/skinManager/include/iCub/skinManager/compensator.h: In member function ‘std::string iCub::skinManager::Compensator::getInputPortName()’:
/home/traversaro/robotology-superbuild/src/ICUB/src/modules/skinManager/include/iCub/skinManager/compensator.h:189:61: error: ‘class yarp::dev::PolyDriver’ has no member named ‘getValue’
  189 |     string getInputPortName(){  return tactileSensorDevice->getValue("remote").asString().c_str(); }
      |                                                             ^~~~~~~~
[12/22] Building CXX object src/modules/camCalib/CMakeFiles/camCalib.dir/src/CamCalibModule.cpp.o
FAILED: src/modules/camCalib/CMakeFiles/camCalib.dir/src/CamCalibModule.cpp.o
/usr/bin/c++  -I/home/traversaro/robotology-superbuild/src/ICUB/src/modules/camCalib/include -isystem /usr/include/opencv4 -isystem /home/traversaro/robotology-superbuild/build-apt/install/include -O3 -DNDEBUG -fPIE -std=c++17 -MD -MT src/modules/camCalib/CMakeFiles/camCalib.dir/src/CamCalibModule.cpp.o -MF src/modules/camCalib/CMakeFiles/camCalib.dir/src/CamCalibModule.cpp.o.d -o src/modules/camCalib/CMakeFiles/camCalib.dir/src/CamCalibModule.cpp.o -c /home/traversaro/robotology-superbuild/src/ICUB/src/modules/camCalib/src/CamCalibModule.cpp
/home/traversaro/robotology-superbuild/src/ICUB/src/modules/camCalib/src/CamCalibModule.cpp: In member function ‘virtual bool CamCalibModule::configure(yarp::os::ResourceFinder&)’:
/home/traversaro/robotology-superbuild/src/ICUB/src/modules/camCalib/src/CamCalibModule.cpp:113:15: error: ‘class yarp::os::Bottle’ has no member named ‘setMonitor’
  113 |     botConfig.setMonitor(rf.getMonitor());
      |               ^~~~~~~~~~
/home/traversaro/robotology-superbuild/src/ICUB/src/modules/camCalib/src/CamCalibModule.cpp:113:29: error: ‘class yarp::os::ResourceFinder’ has no member named ‘getMonitor’
  113 |     botConfig.setMonitor(rf.getMonitor());
      |                             ^~~~~~~~~~
~~~

Also the removal of `setMonitor` and `getValue` are related to this PR: https://github.com/robotology/yarp/pull/3059 .

pattacini, 2023-12-19:
https://github.com/robotology/yarp/pull/3059 seems to be part of a bigger overhaul.
Do we need to wait for more to come as pointed out in the body of that PR, in terms of new PR's addressing the refactoring, @randaz81?

randaz81, 2023-12-19:
@pattacini Indeed the Yarp refactoring is not complete yet: my plan was to work on it to complete it asap, before pushing fixes on the other repos. (something more can be broken in the next few days, I apologize for that). 
However, the changes mentioned so far in this issue are already final.
I already have a fix for them and I'm going to push it in a separate PR.

randaz81, 2023-12-19:
fixed by #925

pattacini, 2023-12-28:
Hi @traversaro 

Any other problem to report related to `yarp@master`?

pattacini, 2024-01-04:
> Hi @traversaro
> 
> Any other problem to report related to `yarp@master`?

Closing. Feel free to open it again if required.

closed 2024-01-04 by pattacini

traversaro, 2024-01-04:
Sure, thanks for handling this!