robotology/icub-main #929: Remove the usage of `yarp::dev::IAnalogSensors`

Nicogene, 2023-12-28:
I let things get out of hand and in the PR listed below I removed it also from other devices that had been already ported to the usage of the MultipleAnalogSensorsInterfaces

- https://github.com/robotology/icub-main/pull/928

Looking at the usage of `yarp::dev::IAnalogSensor` interface in `icub-main` some devices still use it and they have not been ported to the MAS interfaces:

- `canBusAnalogSensor`
- `canBusInertialMTB`
- `canBusMotionControl`
- `canBusSkin`
- `embObjMulltiEnc`
- `embObjPOS`
- `embObjPSC`
- `embObjSkin`
- `skinWrapper`
- `skinManager`

We know that the skin pipeline has to be refactored, maybe some of them can be removed completely as we did for the `embObjAnalogSensor`

For the remaining we have to chat with @marcoaccame

_Originally posted by @Nicogene in https://github.com/robotology/icub-main/issues/919#issuecomment-1871189740_

Nicogene, 2024-03-29:
| Device  |   |   |   |   |
|---|---|---|---|---|
| `canBusAnalogSensor`  |   |   |   |   |
| `canBusInertialMTB`  |   |   |   |   |
| `canBusMotionControl`  |   |   |   |   |
| `canBusMotionControl`  |   |   |   |   |
| `canBusMotionControl`  |   |   |   |   |

Nicogene, 2024-03-29:
| Device  | Usage in robots-configurations  | Interfaces implemented  | Possible action  | 
|---|---|---|---|
| `canBusAnalogSensor`  | ☑️  [view usage](https://github.com/search?q=repo%3Arobotology%2Frobots-configuration%20canBusAnalogSensor&type=code) (CAN iCub) | `yarp::dev::IAnalogSensor`  | Use [`canBusFtSensor`](https://github.com/robotology/icub-main/tree/master/src/libraries/icubmod/canBusFtSensor) for strain, create `canBusMais` similar to [`embObjMais`](https://github.com/robotology/icub-main/blob/master/src/libraries/icubmod/embObjMais/embObjMais.h), remove `canBusAnalogSensor` from `robots-configuration` and `icub-main`  | 
| `canBusInertialMTB`  |  ☑️  [view usage](https://github.com/search?q=repo%3Arobotology%2Frobots-configuration%20canBusInertialMTB&type=code)  (CAN iCub)  |  `yarp::dev::IAnalogSensor`  |  Port it to MAS interfaces | 
| `canBusMotionControl`  | ☑️ [view usage](https://github.com/search?q=repo%3Arobotology%2Frobots-configuration%20canmotioncontrol&type=code)  (CAN iCub)  | Has a list of  `yarp::dev::IAnalogSensor`  |  ❓❓❓ | 
| `canBusSkin`  |  ☑️ [view usage](https://github.com/search?q=repo%3Arobotology%2Frobots-configuration%20canBusSkin&type=code) (CAN iCub) |  `yarp::dev::IAnalogSensor`  |  Refactor skin sw stack  | 
| `embObjMulltiEnc`  | ❌ [view usage](https://github.com/search?q=repo%3Arobotology%2Frobots-configuration+embObjMulltiEnc&type=code)   | `yarp::dev::IAnalogSensor`  |  Let's remvove it from `icub-main` | 
| `embObjPOS`  | ☑️ [view usage](https://github.com/search?q=repo%3Arobotology%2Frobots-configuration%20embObjPOS&type=code) (ergoCub) | `yarp::dev::IAnalogSensor`  | ❓❓❓  | 
| `embObjPSC`  |  ☑️ [view usage](https://github.com/search?q=repo%3Arobotology%2Frobots-configuration%20embObjPSC&type=code)  (only CER) | `yarp::dev::IAnalogSensor`  |  ❓❓❓  | 
| `embObjSkin`  |  ☑️ [view usage](https://github.com/search?q=repo%3Arobotology%2Frobots-configuration+embObjSkin&type=code) (both ergoCub and iCub) | `yarp::dev::IAnalogSensor`  |  Refactor skin sw stack  | 

`skinWrapper` and `skinManager` are executables that until we do not refactor the skin sw stack will rely on `IAnalogSensor`

I tried to write down the current situation with some possible action points for what I know, @marcoaccame @traversaro @pattacini please feel free to contribute to this table

traversaro, 2024-03-29:
fyi @valegagge , as apparently she was the original author of `embObjMultiEnc`: https://github.com/robotology/icub-main/commit/7578d28f389883705b53a763bd8fef0a5b4f78a4 .

valegagge, 2024-03-29:
Hi guys,
I report here some information related to the `embObjMultiEnc`.
In r1SN001 there is an encoder called [`spichainof3`](https://github.com/robotology/robots-configuration/blob/3b3548b835b89fbf6b7b565f8d8c08d343a954b0/R1SN001/hardware/motorControl/cer_left_hand-mcp9-mc-service.xml#L25C52-L25C64) at joint.
As the name says, the joint position is calculated by a combination of 3 spi encoder using the matrix defined in [the mechanical file](https://github.com/robotology/robots-configuration/blob/3b3548b835b89fbf6b7b565f8d8c08d343a954b0/R1SN001/hardware/mechanicals/cer_left_hand-mcp9-mec.xml#L44). 
So the joint position in this robot is provided as usual, but we needed a way to publish the value of each single joint, especially in the first stages of development and testing on this architecture.
From this need, we developed the `embObjMultiEnc`.

I also looked around in the fw. the code that can publish such values still exists, but I didn't understand if they are broadcasted toward YRI.(I guess no)

Said that, I suppose that now there are very occasions where we need to check the raw value of the spichainof3 encoder. they are only on R1 robot.

In my opinion, we can remove the device, but we need to take into account that maybe we need to develop it. (it is a slim chance.)

Nicogene, 2024-04-16:
/remind 6th may plan for action points

Nicogene, 2024-06-27:
AP:
- Remove `embObjPSC` from both `icub-main` and `robots-configuration` -> @Nicogene 
- Remove `embObjMulltiEnc` from both `icub-main` and `robots-configuration`-> @Nicogene 
- `embObjSkin` -> @marcoaccame + @simeonedussoni 
- `embObjPOS` -> https://www.yarp.it/latest/classyarp_1_1dev_1_1IEncoderArrays.html or new five interface -> @valegagge @MSECode @marcoaccame 
- `canBusMotionControl` -> not clear how `iAnalogSensor` is used, needed a study on it -> @valegagge 
- `embObjMais` -> fix `springyFingers` readout @pattacini 
- `canBusMais` -> @mfussi66 
- `canBusInertialMTB` -> @Nicogene

Nicogene, 2024-06-27:
I converted this to an Epic, you can link your activities to this.

cc @pattacini @valegagge @marcoaccame @MSECode @simeonedussoni @mfussi66

Nicogene, 2025-06-06:
Fyi:
- https://github.com/robotology/yarp/pull/3240