robotology/icub-main #933: Skin on legs get assigned wrong bodyPartId in skinContactList on /skinManager/skin_events:o

rustlluk, 2024-01-12:
### Bug description

When reading data through `skinManager` on `/skinManager/skin_events:o`, skin on legs (LEFT/RIGHT_LEG_UPPER, LEFT/RIGHT_LEG_LOWER, LEFT/RIGHT_FOOT) get correct skinPart is assigned with respect to [skinPart enum in skinDynLib/common.h](https://github.com/robotology/icub-main/blob/92430ce9c1a249964f99382da3df7a91537fd42e/src/libraries/skinDynLib/include/iCub/skinDynLib/common.h#L56-L63C3), but bodyPartId is always 0 (BODY_PART_UNKNOWN). 

### Steps to reproduce

Run skinManager with any config that contains skin on legs and read data on /skinManager/skin_events:o

### Expected behavior

The correct `bodyPartId` for skin on left leg should be 5 (`LEFT_LEG`), and 6 (`RIGHT_LEG`) for the right leg. From [BodyPart enum](https://github.com/robotology/icub-main/blob/92430ce9c1a249964f99382da3df7a91537fd42e/src/libraries/skinDynLib/include/iCub/skinDynLib/common.h#L44-L48).

### Additional context

The error is probably coming from not having the association of skin on legs to body part in [SkinPart_2_BodyPart in skinDynLib/common.h](https://github.com/robotology/icub-main/blob/92430ce9c1a249964f99382da3df7a91537fd42e/src/libraries/skinDynLib/include/iCub/skinDynLib/common.h#L77-L87) and thus function [iCub::skinDynLib::getBodyPart(SkinPart s)](https://github.com/robotology/icub-main/blob/92430ce9c1a249964f99382da3df7a91537fd42e/src/libraries/skinDynLib/src/common.cpp#L26-L32) returns `BODY_PART_UNKNOWN`.  But I am not brave enough to compile and test the library on our robot :)

pattacini, 2024-01-12:
Thanks @rustlluk for reporting 👍🏻 
We will probably deal with this while checking #932.

traversaro, 2024-01-12:
@rustlluk do you actually are using bodyPartId somewhere? That part of the code is quite iCub-centric, and it would be great to make it more general. However, to do so we would need to understand how people are actually using data like bodyPartId .

rustlluk, 2024-01-12:
Well, actually not.  I would say that `bodyPartId` and `linkNumber` are redundant, as you can easily extract both from `skinPart`.  @traversaro 

I even looked to an old code I used with our Nao with skin and the only thing I did with `bodyPartId` was saving it to log.

Nicogene, 2024-01-23:
Unfortunately the iCub we have here have not the legs, but I can backtrack the code that assigns the bodypart and try to fix it, the other parts are ok instead.

PR #936 (2024-01-29): skinDynLib: fix missing body part id for the lowerbody
This PR add the missing enums to the map used for defining the bodyPart

It fixes #933 

~TO BE TESTED~
cc @traversaro @pattacini @rustlluk
closed 2024-01-29 by pattacini