robotology/icub-main #937: Fix diagnostic messages

valegagge, 2024-01-29:
### Bug description

In the log reported [here](https://github.com/robotology/icub-tech-support/issues/1726#issue-2105089872), I noticed some weird diagnostic messages that might be worth investigating further. 

1. ` 426.026906 <ERROR> from BOARD 10.0.1.8 (left_leg-eb8-j0_3) time=467s 485m 342u :  MC: generic motor error. (Error is a000000)` .==>  Here the joint ID is missing

3. ` 418.960928 <WARNING> from BOARD 10.0.1.2 (left_arm-eb2-j0_1) time=460s 417m 341u : SYS: a service has detected that some CAN boards are not broacasting anymore. Type of service category is eomn_serv_category_temperatures. Lost can boards on (can1map, can2map) = ([ 13  ], [  ] )`. ==>  Why temperature? it was disabled. And what is the address `13`?

5. `420.013520 <ERROR> from BOARD 10.0.1.2 (left_arm-eb2-j0_1) time=461s 466m 194u : SYS: a service has detected that some CAN boards have stopped transmission. Type of service category is eomn_serv_category_all. Lost CAN boards are on (can1map, can2map) = ([ 2  ], [  ]). Time since last contact: 0 [ms]`. ==>  Why 0 ms? 

### Steps to reproduce

See the ticket https://github.com/robotology/icub-tech-support/issues/1726

pattacini, 2024-01-30:
A bug related to diagnostic is also:
- #913

MSECode, 2024-02-05:
Regarding the errors:
- Point 3 is easy to fix --> if seen that in the EMS code we are not passing some of the info we are trying to request on the icub-main part. As shown by the screenshots we are not adding to the last significative bits of the par64 the timing information, as it is done for other errors:

[image]

but we are just passing the masks of the CAN boards that are `justLOST` :

[image]
 

so we just need to add that piece of information to the par64, since `mspassed` can be calculated for all the cases.

MSECode, 2024-02-05:
Point 2 instead is quite weird, since it is related to this message:
`eocanprotASperiodic_parser_PER_AS_MSG__THERMOMETER_MEASURE` which should not be related to any board, since I suppose there's not any board which currently uses this feature.

MSECode, 2024-02-05:
For point 1, we just need to parse at high level the par16, which contains the joint ID.

MSECode, 2024-02-08:
[image]

MSECode, 2024-02-12:
## **UPDATE**

### Point 3:

```
420.013520 <ERROR> from BOARD 10.0.1.2 (left_arm-eb2-j0_1) time=461s 466m 194u : 
SYS: a service has detected that some CAN boards have stopped transmission.
Type of service category is eomn_serv_category_all. Lost CAN boards are on (can1map, can2map) = ([ 2  ], [  ]). 
Time since last contact: 0 [ms]
```
Need to remove the parsing at high level of the `time since last contact` since it would not make sense to ask for the that time when the boards are just lost. It would be basically zero. So it is correct that that time interval is not passed for that precise error at fw level. 
Instead we realized that we are not parsing at software level the `time since last contact` for the warning `eoerror_value_SYS_canservices_monitor_retrievedcontact`, which is triggered when the boards are re-touched after being missed. So we need to update the software part of the diagnostic related to that warning adding both the boards CAN mapping and the time data.

### Branches were we are keeping the development are:

- https://github.com/MSECode/icub-firmware/tree/fix/diagnosticLostCANBoards
- https://github.com/MSECode/icub-main/tree/fix/diagnosticLostCANBoards
- https://github.com/MSECode/icub-main/tree/fix/diagnosticGenError
-

MSECode, 2024-02-13:
## **UPDATE 2**

### Point 1 

Solved in the way that now the ID is shown. Working in improving wisely the management of a generic error so that it can be fully readable by the user

### Point 2
Under investigation. Need to understand if services are initialized finely.

### Point 3 
Now should be fully solved in all its parts and fix are available here:

- https://github.com/MSECode/icub-firmware-shared/tree/fix/diagnosticLostCANBoards
- https://github.com/MSECode/icub-main/tree/fix/diagnosticLostCANBoards
- https://github.com/MSECode/icub-firmware/tree/fix/diagnosticLostCANBoards

Problems related to that error can be summarized as follows:

1. As already shown for some of the "error" (info, warning and error) conditions of the CANmonitor different info were passed towards the higher level but the icub-main code was treating them in the same way for all cases. First thing is that if we have not any problem in reading the boards or if we are continuously loosing the transmission with the, i.e. if we are still in the conditions `State::OK` or `State::stillLOST` there's no reason to send up the time from last contact, which will be always be close to the tick() time of the CANmonitor (usually close to 1ms). So we fix the software part in `icub-main` in order to parse the `mspassed` only for the conditions `State::justFOUND` and `State::justLOST`, when they are actually meaningful
2. Then, another problem was related to the fact that the service category was not displayed correctly in the message. This was due to a problem with the array `s_mn_servicecategory_strings`.
3. Finally, we had a couple of issues with the CANmonitor. First of all the `constexpr Config`, responsible for taking in input the custom configuration parameters for the CANmonitor and update its private variables with those values, was missing the update of the private variable `periodofreport` and therefore, even if we were modifying it value the CANmonitor object was always set with the default value of 10s
4. Then, the last problem was related to the not correct updating of the CAN mask for the `boards2report`, whose values were always zero in the case in which the boolean value `forcereport` was set to false, such as for the states `OK` and `stillLOST`. This was due to the following situation: 
    if `forcereport` is set to `false` then the CANmonitor is sending the report for the states only if `regularreportnow` is set to true (supposed that our report mode is Report::ALL). Then, the boolean value `regularreportnow` is set to true only in the case that `(timeoflastreport + _config.periodofreport) < timenow`. Moreover, we can see from the old code that at each `tick()` of the CANmonitor (that happens at each 1ms) the `MAP boards2report` was re-initialized. Furthermore,  we can see that we enter the state-machine part where the variable `board2report` is updated if and only if the boolean flag `checknow` is equal to true and this happens only the times in which `(timeoflastcheck + _config.periodofcheck) < timenow`. 
    All that said, the problem of having always at zero the content of the CAN masks happened becuase, without forcing the report at each `periodofcheck`, in the moment in which `regularreportnow` goes to true, the other boolean `checknow` is set to false, because of timing conditions due to the fact that the `tick()` is executed at each 1ms, and so we never update the variable `boards2report` and being it re-initialized at each `tick()` is explained why we are sending the var always to zero.

**NOTE:**
Timing conditions here explained. Supposed that  we have the following condition:
- the tick() runs at each 1ms
- `periodofcheck` is 250ms
- `periodofreport` is 5s

So, in the moment in which we are going to be ready for the report the current time will be 50001 ms and therefore it would be different from the relative 250s in which will be set the boolean `checknow` to true.

Here some text and images showing the difference before and after the modifications and the timing printed in the TRACE of the Debugger:

**BEFORE**

```
Touched the boards @ S31:m752:u993 with timeoflastcheck: S31:m752:u935
Checked now the boards @ S32:m3:u935 with timeoflastcheck: S31:m752:u935
Touched the boards @ S32:m3:u992 with timeoflastcheck: S32:m3:u935
Checked now the boards @ S32:m254:u935 with timeoflastcheck: S32:m3:u935
Touched the boards @ S32:m254:u993 with timeoflastcheck: S32:m254:u935
Checked now the boards @ S32:m505:u935 with timeoflastcheck: S32:m254:u935
Touched the boards @ S32:m505:u993 with timeoflastcheck: S32:m505:u935
Regular report on the boards @ S32:m680:u935 with timeoflastreport: S27:m679:u935
[INFO] (theBATservice tsk11 @S32:m680:u0)-> {0x3c, p16 0x0009, p64 0x0000000000000000, dev 0, adr 9}: SYS: a service has verified that the TX of its CAN boards is regular..
Reported the boards @ S32:m681:u120 with timeoflastreport: S32:m680:u935
Checked now the boards @ S32:m756:u935 with timeoflastcheck: S32:m505:u935
Touched the boards @ S32:m756:u993 with timeoflastcheck: S32:m756:u935
Checked now the boards @ S33:m7:u935 with timeoflastcheck: S32:m756:u935
Touched the boards @ S33:m7:u992 with timeoflastcheck: S33:m7:u935
Checked now the boards @ S33:m258:u935 with timeoflastcheck: S33:m7:u935
Touched the boards @ S33:m258:u993 with timeoflastcheck: S33:m258:u935
Checked now the boards @ S33:m509:u935 with timeoflastcheck: S33:m258:u935
Touched the boards @ S33:m509:u993 with timeoflastcheck: S33:m509:u935
[... cut ...]
Touched the boards @ S65:m92:u933 with timeoflastcheck: S65:m92:u875
Checked now the boards @ S65:m343:u875 with timeoflastcheck: S65:m92:u875
Touched the boards @ S65:m343:u934 with timeoflastcheck: S65:m343:u875
Checked now the boards @ S65:m594:u875 with timeoflastcheck: S65:m343:u875
Touched the boards @ S65:m594:u934 with timeoflastcheck: S65:m594:u875
Checked now the boards @ S65:m845:u875 with timeoflastcheck: S65:m594:u875
Touched the boards @ S65:m845:u934 with timeoflastcheck: S65:m845:u875
```
[image]

valegagge, 2024-02-14:
@MSECode well done!!! 🚀

valegagge, 2024-02-14:
We still need to perform the tests on the setup with two boards connected to the same CAN port (i.e. strain +bat).

MSECode, 2024-02-14:
## UPDATE 3:

### **Point 2**
regarding the error at point 2, i.e.:

```
418.960928 <WARNING> from BOARD 10.0.1.2 (left_arm-eb2-j0_1) time=460s 417m 341u : SYS: a service has detected that some CAN boards are not broacasting anymore. Type of service category is eomn_serv_category_temperatures. Lost can boards on (can1map, can2map) = ([ 13  ], [  ] ). ==> Why temperature? it was disabled. And what is the address 13?
```
Trying to reproduce it with a setup basically equivalent to the ems eb1 of ergoCub SN00x, where we have the strain2c board and the BAT on the the CAN2, and the 2foc on CAN1, I realized that the aforementioned problem was related to the problem already found on the typo of the array used for converting the service category form the enum to the equivalent string. 
Because of, as shown in [this line](https://github.com/MSECode/icub-firmware-shared/commit/7796c8e4bd88af066c103ecf73f2935420679d2c#diff-35ce1cbcfac00cc92906f1633802bb0f8678c2fcafe5092866e19f42bdde7231R99) fixed the other day, we were picking wrongly the index of the array for a specific enum value and as you can see the categories `eomn_serv_category_inertials3` (which is the one instantiated for the strain2c) and `eomn_serv_category_temperatures` (the one wrongly shown in the warning message) are one after the other as shown here:

[image]

So the service `eomn_serv_category_inertials3` was correctly instantiated but we were translating wrongly the category enum when received from the fw. 
Then the error of the CAN address 13 (the default for the strain) being on the CAN1 mask was due to the fact that on icub-main sw the masks were swapped.
Now as you can see from the image below now the warning message is correctly shown.
I got it by disconnecting  the CAN of the strain2c from the ems just for the purpose of the test. Otherwise it works fine.

[image]

valegagge, 2024-02-15:
Just another minor improvement: from the image above, the period of transmission of the info `CAN board is regular` has a short period. I guess you do for testing purpose, but we shouldn't forget to update it to 10 or also 20 seconds.

Now we are sure, that when we lose a CAN board the error message is triggered suddenly, are we?

MSECode, 2024-02-15:
Reset to 10 seconds. It looks reasonable to me.

MSECode, 2024-02-16:
Opened all PRS:
- icub-firmware-shared --> https://github.com/robotology/icub-firmware-shared/pull/92
- icub-main --> https://github.com/robotology/icub-main/pull/941
- icub-firmware --> https://github.com/robotology/icub-firmware/pull/471

valegagge, 2024-02-18:
Ciao @MSECode ,
after the changes required in iCub-main PR, we can go on with the tests.

For the final tests on the diagnostic, we can do the following actions on the setup you created (EMS+battery +Strain +2FOC).

1. start YRI ==> all is ok
2. remove Strain
3. reattach the strain 
4. disconnect strain and battery
5. reconnect strain a battery.
6. disconnect strain (or battery) and 2foc
7. reconnect strain and 2foc.

Please you attach here the YRI log. thank you!
 I'm not available on these two days but I'll check the issue or we'll catch up at lunch time quickly. 

What do you think?

MSECode, 2024-02-19:
Here the output of the Tests:

1. Start YRI --> All ok (the error is related to the fact that I request the FT service to be active without any sensor attached to the strain2c):
[image]

2. Removing the `strain2c` from the CAN we can see that we cannot find anymore the CAN boards related to both FT and inertials3 services, while the battery service is still ok:
[image]

3. Reconnecting the strain nothing changes --> the transmission continues to fail. The discovery cannot be refreshed until restarting of the YRI

4. If we disconnect the Battery first, we can see from the logs that we are loosing the strain too. However, when we reconnect the battery, both the services restart. The battery service can restart since the BAT is powered from the power supply and not from the CAN only. 
Instead, if we do the opposite, i.e. we disconnect before the strain2c and later the BAT the only service that restarts is the battery service. 
This is probably due to the fact that until the strain2c remains connected via CAN it continues to stream the frames, even if the BAT is removed. We have, anyway, errors since the BAT and strain2c bus are short-circuited and having the BAT not attached is like having no CAN terminations and this is not good for the EMS
[image]

5. If a disconnect 2foc only I get the following output, meaning that at first we have errors for the failing of the transmission over CAN but at the reconnection the errors go away
[image]

If we disconnect battery and 2foc the output is the following:
[image]

After some seconds
[image]

At reconnection
[image]

i suppose 2foc cannot be recovered until restarting

With 2foc and strain2c this one:
First disconnect the FT (warning messages)
Then the 2foc (Error and Warning)
[image]

At reconnection still errors
[image]

valegagge, 2024-02-19:
Hi @MSECode ,
thank you very much for the exhaustive tests.

I just wrote done some notes for each point.

2: In the reported log I cannot see the `just-lost` error. It might be in the previous line... Did you see it?

3: it is ok, the strain needs to receive the configuration and the start command because it is powered by CAN. 

4:here I can see the `just-stop` and the `restart`, but not the `still-lost`. Probably you attached the CAN before the `still-lost` appeared, Didn't you?

5: I noticed this discordance: 
[image]

Could you take a look at the diagnostic parser? If it is quickly we can fix it, otherwise we'll plan to fix it in the feature.

From your experiments and taking a look into the code, the 2foc or, more in general, the motion control hasn't the canMotitor service. So, we cannot take the motion control service into account for this tests.

- When you said `i suppose 2foc cannot be recovered until restarting` It is not clear for me. Sorry. 

From the penultimate image I see that we have another parser problem:
[image]

As before, If we have time, we can try to fix it otherwise we'll address it in the feature.

In general, it seems that the inertial3 service diagnostic messages hide the canMonitor info of ft service. Is it possible to remove the inertial3 service from the config file? So our log, should be more clear. We can check it together.

MSECode, 2024-02-20:
Point 2: We have the `justLOST` error, it was probably above wrt to the screen, here you can see it:

[image]

Point4: 
We have also the `stillLOST` message for both battery and ft (which have the canmonitor configured) in both cases in which we disconnect the only the battery or the strain2c. In the latter case the inertial and ft services cannot recover the transmission as said at point 3

[image]
[image]

[image]

MSECode, 2024-02-20:
### Generic Error Improvement

I've hardcoded on the ems the rising of some of the bits of the bitmask for the motor fault

[image]

MSECode, 2024-02-20:
Also test removing and connecting again 2foc and bat or strain2c is fine.
Here the full YRI log

[yarplog-2foc-ems-bat-strain-test.txt](https://github.com/robotology/icub-main/files/14345030/yarplog-2foc-ems-bat-strain-test.txt)

MSECode, 2024-02-21:
Here the logs of the test with the latest changes done today
[yarplog-2foc-ems-bat-strain-test-last.txt](https://github.com/robotology/icub-main/files/14363200/yarplog-2foc-ems-bat-strain-test-last.txt)

valegagge, 2024-02-23:
PRs merged.

Dod satisfied. closing

closed 2024-02-23 by valegagge