robotology/icub-main #939: FirmwareUpdater - Fail message on successful FW flashing.

davidetome, 2024-02-13:
### Bug description

Whem you do Upload Application, the message in the terminal at the end of the programming returns a failed operation even if it went well.

```bash
EthMaintainer::cmdProgram() is about to program the APP partition of 1 boards: 10.0.1.1 (candoit)
ERROR: EthMaintainer::program() could not program board none @ 0.0.0.0: 10.0.1.1: APP NOK
```

### Steps to reproduce

- Open Firmware Updater
- Do a discover
- Select an ETH board, an `mc4plus` (in my case) 
- Put the board un Maintenance
- Do Upload Application
- See the FAIL message in the terminal

### Expected behavior

The message should report a successful operation (APP OK)

davidetome, 2024-02-13:
cc @valegagge

davidetome, 2024-02-14:
I found the reason of the failure, in the `command_program` method because [two indexes differ by one](https://github.com/davidetome/icub-main/blob/686cfdac1b9bb8ac76e253e0d62808e41225a8a9/src/tools/ethLoader/ethLoaderLib/EthMaintainer.cpp#L566) 

I also noticed that one of them is increased by one just after [it is initialized at zero](https://github.com/davidetome/icub-main/blob/686cfdac1b9bb8ac76e253e0d62808e41225a8a9/src/tools/ethLoader/ethLoaderLib/EthMaintainer.cpp#L337)

Commenting out it, the program operation works properly : 

[image]

I have to double-check if it is ok to comment out this [line](https://github.com/davidetome/icub-main/blob/686cfdac1b9bb8ac76e253e0d62808e41225a8a9/src/tools/ethLoader/ethLoaderLib/EthMaintainer.cpp#L337) 👍🏼 

cc @valegagge

davidetome, 2024-02-21:
I tested the mod w/ 2 mc4plus and one ems and it works properly 👍🏼 

### 2 mc4plus programmed in parallel successfully
[image]

### 1 ems programmed successfully
[image]

Now I'm going to fix the message : 

```
OK: EthMaintainer::program() has succesfully programmed board none @ 0.0.0.0.
```
cc @valegagge

davidetome, 2024-02-22:
## Fixed message
With this [commit](https://github.com/robotology/icub-main/compare/devel...davidetome:icub-main:devel) in my fork I fixed also the message in the terminal : 

### 2 mc4plus programmed in parallel successfully
[image]

### 1 ems programmed successfully
[image]

@valegagge I wait for your green light 🚦 to rebase and open the PR 👍🏼

pattacini, 2024-02-27:
Hi @davidetome 

The issue will be closed upon merging the PR.

closed 2024-02-27 by pattacini