robotology/icub-main #943: Make the `Generic Error` more readable

valegagge, 2024-02-23:
### Is your feature request related to a problem?

Currently, when more errors occur on the 2FOC board, some are notified to the user with a bitmask.
We want to improve the information related to the errors by writing a list of the errors instead of a bitmask.

### The solution you would like to have available

We need to translate the bitmask into a list of significant strings.

MSECode, 2024-03-04:
This is related to what done at the branches related to these PRs:
- https://github.com/robotology/icub-main/pull/942
- https://github.com/robotology/icub-firmware/pull/472

MSECode, 2024-03-07:
Closing as duplicated

closed 2024-03-07 by MSECode