robotology/icub-main #947: Failed to compile Python bindings

S-Dafarra, 2024-02-29:
### Bug description

I am trying to compile ``ICUB`` with the Python bindings but I get the following error
```
(robsub) C:\Software\robotology-superbuild\build\src\ICUB>cmake --build . --config Release --target INSTALL
MSBuild version 17.8.5+b5265ef37 for .NET Framework

  1>Checking Build System
  Automatic MOC for target QtICubSkinGuiPlugin
  Fingertip4Left.cpp
  Fingertip4Right.cpp
  FingertipMID.cpp
  fakePalm.cpp
  qticubskinguiplugin.cpp
  qticubskinguiplugin_plugin.cpp
  SkinMeshThreadCan.cpp
  SkinMeshThreadPort.cpp
  mocs_compilation_Release.cpp
  Generating Code...
     Creating library C:/Software/robotology-superbuild/build/src/ICUB/lib/qt5/qml/Release/robotology/icub/skingui/QtICubSkinGuiPlugin.lib and object C:/Software/robotology-superbuild/build/src/ICUB/lib/qt5/qml/Releas
  e/robotology/icub/skingui/QtICubSkinGuiPlugin.exp
  QtICubSkinGuiPlugin.vcxproj -> C:\Software\robotology-superbuild\build\src\ICUB\lib\qt5\qml\Release\robotology\icub\skingui\QtICubSkinGuiPlugin.dll
  ctrlLib.vcxproj -> C:\Software\robotology-superbuild\build\src\ICUB\lib\Release\ctrlLib.lib
  iKin.vcxproj -> C:\Software\robotology-superbuild\build\src\ICUB\lib\Release\iKin.lib
  Taxel.cpp
C:\Software\robotology-superbuild\src\ICUB\src\libraries\skinDynLib\include\iCub\skinDynLib\common.h(272,27): warning C4267: 'argument': conversion from 'size_t' to '_Ty', possible loss of data [C:\Software\robotology
-superbuild\build\src\ICUB\src\libraries\skinDynLib\skinDynLib.vcxproj]
C:\Software\robotology-superbuild\src\ICUB\src\libraries\skinDynLib\include\iCub\skinDynLib\common.h(272,27): warning C4267:         with [C:\Software\robotology-superbuild\build\src\ICUB\src\libraries\skinDynLib\skin
DynLib.vcxproj]
C:\Software\robotology-superbuild\src\ICUB\src\libraries\skinDynLib\include\iCub\skinDynLib\common.h(272,27): warning C4267:         [ [C:\Software\robotology-superbuild\build\src\ICUB\src\libraries\skinDynLib\skinDyn
Lib.vcxproj]
C:\Software\robotology-superbuild\src\ICUB\src\libraries\skinDynLib\include\iCub\skinDynLib\common.h(272,27): warning C4267:             _Ty=unsigned int [C:\Software\robotology-superbuild\build\src\ICUB\src\libraries
\skinDynLib\skinDynLib.vcxproj]
C:\Software\robotology-superbuild\src\ICUB\src\libraries\skinDynLib\include\iCub\skinDynLib\common.h(272,27): warning C4267:         ] [C:\Software\robotology-superbuild\build\src\ICUB\src\libraries\skinDynLib\skinDyn
Lib.vcxproj]
  (compiling source file '../../../../../../src/ICUB/src/libraries/skinDynLib/src/Taxel.cpp')

  common.cpp
C:\Software\robotology-superbuild\src\ICUB\src\libraries\skinDynLib\include\iCub\skinDynLib\common.h(272,27): warning C4267: 'argument': conversion from 'size_t' to '_Ty', possible loss of data [C:\Software\robotology
-superbuild\build\src\ICUB\src\libraries\skinDynLib\skinDynLib.vcxproj]
C:\Software\robotology-superbuild\src\ICUB\src\libraries\skinDynLib\include\iCub\skinDynLib\common.h(272,27): warning C4267:         with [C:\Software\robotology-superbuild\build\src\ICUB\src\libraries\skinDynLib\skin
DynLib.vcxproj]
C:\Software\robotology-superbuild\src\ICUB\src\libraries\skinDynLib\include\iCub\skinDynLib\common.h(272,27): warning C4267:         [ [C:\Software\robotology-superbuild\build\src\ICUB\src\libraries\skinDynLib\skinDyn
Lib.vcxproj]
C:\Software\robotology-superbuild\src\ICUB\src\libraries\skinDynLib\include\iCub\skinDynLib\common.h(272,27): warning C4267:             _Ty=unsigned int [C:\Software\robotology-superbuild\build\src\ICUB\src\libraries
\skinDynLib\skinDynLib.vcxproj]
C:\Software\robotology-superbuild\src\ICUB\src\libraries\skinDynLib\include\iCub\skinDynLib\common.h(272,27): warning C4267:         ] [C:\Software\robotology-superbuild\build\src\ICUB\src\libraries\skinDynLib\skinDyn
Lib.vcxproj]
  (compiling source file '../../../../../../src/ICUB/src/libraries/skinDynLib/src/common.cpp')

C:\Software\robotology-superbuild\src\ICUB\src\libraries\skinDynLib\src\common.cpp(83,27): warning C4267: 'argument': conversion from 'size_t' to 'int', possible loss of data [C:\Software\robotology-superbuild\build\s
rc\ICUB\src\libraries\skinDynLib\skinDynLib.vcxproj]
  dynContact.cpp
C:\Software\robotology-superbuild\src\ICUB\src\libraries\skinDynLib\include\iCub\skinDynLib\common.h(272,27): warning C4267: 'argument': conversion from 'size_t' to '_Ty', possible loss of data [C:\Software\robotology
-superbuild\build\src\ICUB\src\libraries\skinDynLib\skinDynLib.vcxproj]
C:\Software\robotology-superbuild\src\ICUB\src\libraries\skinDynLib\include\iCub\skinDynLib\common.h(272,27): warning C4267:         with [C:\Software\robotology-superbuild\build\src\ICUB\src\libraries\skinDynLib\skin
DynLib.vcxproj]
C:\Software\robotology-superbuild\src\ICUB\src\libraries\skinDynLib\include\iCub\skinDynLib\common.h(272,27): warning C4267:         [ [C:\Software\robotology-superbuild\build\src\ICUB\src\libraries\skinDynLib\skinDyn
Lib.vcxproj]
C:\Software\robotology-superbuild\src\ICUB\src\libraries\skinDynLib\include\iCub\skinDynLib\common.h(272,27): warning C4267:             _Ty=unsigned int [C:\Software\robotology-superbuild\build\src\ICUB\src\libraries
\skinDynLib\skinDynLib.vcxproj]
C:\Software\robotology-superbuild\src\ICUB\src\libraries\skinDynLib\include\iCub\skinDynLib\common.h(272,27): warning C4267:         ] [C:\Software\robotology-superbuild\build\src\ICUB\src\libraries\skinDynLib\skinDyn
Lib.vcxproj]
  (compiling source file '../../../../../../src/ICUB/src/libraries/skinDynLib/src/dynContact.cpp')

  dynContactList.cpp
C:\Software\robotology-superbuild\src\ICUB\src\libraries\skinDynLib\include\iCub\skinDynLib\common.h(272,27): warning C4267: 'argument': conversion from 'size_t' to '_Ty', possible loss of data [C:\Software\robotology
-superbuild\build\src\ICUB\src\libraries\skinDynLib\skinDynLib.vcxproj]
C:\Software\robotology-superbuild\src\ICUB\src\libraries\skinDynLib\include\iCub\skinDynLib\common.h(272,27): warning C4267:         with [C:\Software\robotology-superbuild\build\src\ICUB\src\libraries\skinDynLib\skin
DynLib.vcxproj]
[... 237 lines cut ...]
C:\Software\robotology-superbuild\build\src\ICUB\bindings\CMakeFiles\_icub_python.dir\icubPYTHON_wrap.cxx(53629,94): error C2039: 'RIGHT_FOOT': is not a member of 'iCub::skinDynLib' [C:\Software\robotology-superbuild\
build\src\ICUB\bindings\_icub_python.vcxproj]
  C:\Software\robotology-superbuild\src\ICUB\src\libraries\skinDynLib\include\iCub\skinDynLib\skinContactList.h(36,11):
  see declaration of 'iCub::skinDynLib'

C:\Software\robotology-superbuild\build\src\ICUB\bindings\CMakeFiles\_icub_python.dir\icubPYTHON_wrap.cxx(53629,94): error C2065: 'RIGHT_FOOT': undeclared identifier [C:\Software\robotology-superbuild\build\src\ICUB\b
indings\_icub_python.vcxproj]
C:\Software\robotology-superbuild\build\src\ICUB\bindings\CMakeFiles\_icub_python.dir\icubPYTHON_wrap.cxx(53629,3): error C2660: 'SWIG_Python_SetConstant': function does not take 2 arguments [C:\Software\robotology-su
perbuild\build\src\ICUB\bindings\_icub_python.vcxproj]
  C:\Software\robotology-superbuild\build\src\ICUB\bindings\CMakeFiles\_icub_python.dir\icubPYTHON_wrap.cxx(1168,1):
  see declaration of 'SWIG_Python_SetConstant'
  C:\Software\robotology-superbuild\build\src\ICUB\bindings\CMakeFiles\_icub_python.dir\icubPYTHON_wrap.cxx(53629,3):
  while trying to match the argument list '(PyObject *, const char [11])'

  cartesiancontrollerclient.vcxproj -> C:\Software\robotology-superbuild\build\src\ICUB\lib\iCub\Release\cartesiancontrollerclient.dll
  cartesiancontrollerserver.vcxproj -> C:\Software\robotology-superbuild\build\src\ICUB\lib\iCub\Release\cartesiancontrollerserver.dll
  gazecontrollerclient.vcxproj -> C:\Software\robotology-superbuild\build\src\ICUB\lib\iCub\Release\gazecontrollerclient.dll
  imuFilter.vcxproj -> C:\Software\robotology-superbuild\build\src\ICUB\lib\iCub\Release\imuFilter.dll
```

I am on Windows using the current master version (https://github.com/robotology/icub-main/commit/4595c48eb71a0ce64d61ea309bea329fb0c8bff7)

### Steps to reproduce

Enable ``CREATE_PYTHON`` and compile

traversaro, 2024-02-29:
Which swig version are you using @S-Dafarra ?

S-Dafarra, 2024-02-29:
I updated ``swig`` to 4.2.0, touched the ``icub.i`` file, and now I have a different error
```
(robsub) C:\Software\robotology-superbuild\build\src\ICUB>cmake --build . --config Release --target INSTALL
MSBuild version 17.8.5+b5265ef37 for .NET Framework

  Automatic MOC for target QtICubSkinGuiPlugin
  QtICubSkinGuiPlugin.vcxproj -> C:\Software\robotology-superbuild\build\src\ICUB\lib\qt5\qml\Release\robotology\icub\skingui\QtICubSkinGuiPlugin.dll
  ctrlLib.vcxproj -> C:\Software\robotology-superbuild\build\src\ICUB\Release\ctrlLib.lib
  iKin.vcxproj -> C:\Software\robotology-superbuild\build\src\ICUB\Release\iKin.lib
  skinDynLib.vcxproj -> C:\Software\robotology-superbuild\build\src\ICUB\Release\skinDynLib.lib
  iDyn.vcxproj -> C:\Software\robotology-superbuild\build\src\ICUB\Release\iDyn.lib
  optimization.vcxproj -> C:\Software\robotology-superbuild\build\src\ICUB\Release\optimization.lib
  Swig compile icub.i for python
C:/Software/robotology-superbuild/build/install/share/yarp/bindings\yarp.i(438): warning 206: Unexpected tokens after #endif directive. [C:\Software\robotology-superbuild\build\src\ICUB\bindings\_icub_python.vcxproj]
C:\Software\robotology-superbuild\build\install\include\yarp\os\Contactable.h(46): warning 321: 'open' conflicts with a built-in name in python [C:\Software\robotology-superbuild\build\src\ICUB\bindings\_icub_python.v
cxproj]
C:\Software\robotology-superbuild\build\install\include\yarp\os\Contactable.h(59): warning 321: 'open' conflicts with a built-in name in python [C:\Software\robotology-superbuild\build\src\ICUB\bindings\_icub_python.v
cxproj]
C:\Software\robotology-superbuild\build\install\include\yarp\os\Contactable.h(59): warning 321: 'open' conflicts with a built-in name in python [C:\Software\robotology-superbuild\build\src\ICUB\bindings\_icub_python.v
cxproj]
C:\Software\robotology-superbuild\build\install\include\yarp\os\AbstractContactable.h(38): warning 321: 'open' conflicts with a built-in name in python [C:\Software\robotology-superbuild\build\src\ICUB\bindings\_icub_
python.vcxproj]
C:\Software\robotology-superbuild\build\install\include\yarp\os\AbstractContactable.h(41): warning 321: 'open' conflicts with a built-in name in python [C:\Software\robotology-superbuild\build\src\ICUB\bindings\_icub_
python.vcxproj]
C:\Software\robotology-superbuild\build\install\include\yarp\os\AbstractContactable.h(41): warning 321: 'open' conflicts with a built-in name in python [C:\Software\robotology-superbuild\build\src\ICUB\bindings\_icub_
python.vcxproj]
C:\Software\robotology-superbuild\build\install\include\yarp\os\Log.h(158): warning 325: Nested struct not currently supported (NoLog ignored) [C:\Software\robotology-superbuild\build\src\ICUB\bindings\_icub_python.vc
xproj]
C:\Software\robotology-superbuild\build\install\include\yarp\sig\Image.h(42): error : Type error. Expecting an integral type [C:\Software\robotology-superbuild\build\src\ICUB\bindings\_icub_python.vcxproj]
C:\Software\robotology-superbuild\build\install\include\yarp\sig\Image.h(43): error : Type error. Expecting an integral type [C:\Software\robotology-superbuild\build\src\ICUB\bindings\_icub_python.vcxproj]
C:\Software\robotology-superbuild\build\install\include\yarp\sig\Image.h(44): error : Type error. Expecting an integral type [C:\Software\robotology-superbuild\build\src\ICUB\bindings\_icub_python.vcxproj]
C:\Software\robotology-superbuild\build\install\include\yarp\sig\Image.h(45): error : Type error. Expecting an integral type [C:\Software\robotology-superbuild\build\src\ICUB\bindings\_icub_python.vcxproj]
C:\Software\robotology-superbuild\build\install\include\yarp\sig\Image.h(46): error : Type error. Expecting an integral type [C:\Software\robotology-superbuild\build\src\ICUB\bindings\_icub_python.vcxproj]
C:\Software\robotology-superbuild\build\install\include\yarp\sig\Image.h(47): error : Type error. Expecting an integral type [C:\Software\robotology-superbuild\build\src\ICUB\bindings\_icub_python.vcxproj]
C:\Software\robotology-superbuild\build\install\include\yarp\sig\Image.h(48): error : Type error. Expecting an integral type [C:\Software\robotology-superbuild\build\src\ICUB\bindings\_icub_python.vcxproj]
C:\Software\robotology-superbuild\build\install\include\yarp\sig\Image.h(49): error : Type error. Expecting an integral type [C:\Software\robotology-superbuild\build\src\ICUB\bindings\_icub_python.vcxproj]
C:\Software\robotology-superbuild\build\install\include\yarp\sig\Image.h(50): error : Type error. Expecting an integral type [C:\Software\robotology-superbuild\build\src\ICUB\bindings\_icub_python.vcxproj]
C:\Software\robotology-superbuild\build\install\include\yarp\sig\Image.h(51): error : Type error. Expecting an integral type [C:\Software\robotology-superbuild\build\src\ICUB\bindings\_icub_python.vcxproj]
C:\Software\robotology-superbuild\build\install\include\yarp\sig\Image.h(52): error : Type error. Expecting an integral type [C:\Software\robotology-superbuild\build\src\ICUB\bindings\_icub_python.vcxproj]
C:\Software\robotology-superbuild\build\install\include\yarp\sig\Image.h(53): error : Type error. Expecting an integral type [C:\Software\robotology-superbuild\build\src\ICUB\bindings\_icub_python.vcxproj]
C:\Software\robotology-superbuild\build\install\include\yarp\sig\Image.h(54): error : Type error. Expecting an integral type [C:\Software\robotology-superbuild\build\src\ICUB\bindings\_icub_python.vcxproj]
[... cut ...]
  cartesiancontrollerclient.vcxproj -> C:\Software\robotology-superbuild\build\src\ICUB\lib\iCub\Release\cartesiancontrollerclient.dll
  cartesiancontrollerserver.vcxproj -> C:\Software\robotology-superbuild\build\src\ICUB\lib\iCub\Release\cartesiancontrollerserver.dll
  gazecontrollerclient.vcxproj -> C:\Software\robotology-superbuild\build\src\ICUB\lib\iCub\Release\gazecontrollerclient.dll
  imuFilter.vcxproj -> C:\Software\robotology-superbuild\build\src\ICUB\lib\iCub\Release\imuFilter.dll
```

traversaro, 2024-02-29:
The second error is a known problem with swig 4.2.0 that has been solved in 4.2.1, see https://github.com/robotology/robotology-superbuild/pull/1607 and linked builds. Instead I am surprised 4.1.0 does not work, we are testing that in the robotology-superbuild CI.

traversaro, 2024-02-29:
> To be sure, I pinned `swig` to 4.1.0 and touched again the `icub.i` file but it still fails with the same error of the OP

Just to be sure, have you tried directly a clean build?

Nicogene, 2024-02-29:
On my WSL i managed to compile the bindings on latest master
```
[100%] Built target icub_python_swig_compilation
Consolidate compiler generated dependencies of target _icub_python
[100%] Building CXX object bindings/CMakeFiles/_icub_python.dir/CMakeFiles/_icub_python.dir/icubPYTHON_wrap.cxx.o
/home/ngenesio/robotology/robotology-superbuild/build/src/ICUB/bindings/CMakeFiles/_icub_python.dir/icubPYTHON_wrap.cxx: In function ‘PyObject* PyInit__icub()’:
/home/ngenesio/robotology/robotology-superbuild/build/src/ICUB/bindings/CMakeFiles/_icub_python.dir/icubPYTHON_wrap.cxx:982:64: warning: ‘void PyEval_InitThreads()’ is deprecated [-Wdeprecated-declarations]
  982 | #     define SWIG_PYTHON_INITIALIZE_THREADS  PyEval_InitThreads()
      |                                              ~~~~~~~~~~~~~~~~~~^~
/home/ngenesio/robotology/robotology-superbuild/build/src/ICUB/bindings/CMakeFiles/_icub_python.dir/icubPYTHON_wrap.cxx:53632:3: note: in expansion of macro ‘SWIG_PYTHON_INITIALIZE_THREADS’
53632 |   SWIG_PYTHON_INITIALIZE_THREADS;
      |   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /usr/include/python3.10/Python.h:130,
                 from /home/ngenesio/robotology/robotology-superbuild/build/src/ICUB/bindings/CMakeFiles/_icub_python.dir/icubPYTHON_wrap.cxx:180:
/usr/include/python3.10/ceval.h:122:37: note: declared here
  122 | Py_DEPRECATED(3.9) PyAPI_FUNC(void) PyEval_InitThreads(void);
      |                                     ^~~~~~~~~~~~~~~~~~
[100%] Linking CXX shared module ../lib/_icub.so
[100%] Built target _icub_python
```

@S-Dafarra I had to first do `make uninstall` of icub-main because swig it was taking the old headers installed

cc @pattacini

closed 2024-02-29 by pattacini