robotology/icub-main #949: Improve Motion Control Generic Error handling

MSECode, 2024-03-06:
### Is your feature request related to a problem?

CAN generic error and general generic error are handled by different cases, which is a bit cumbersome since the former is a subpart of the latter and moreover the output of the CAN generic error management it is not totally clear.

### The solution you would like to have available

I'm gonna improve the management of all generic errors sent from the fw side to the sw side.
The specific handling of the generic error is done on the sw side, where based on the set bits of the error mask we are generating in output a error string containing all the error strings correlated to the risen bits of the mask

MSECode, 2024-03-06:
Here the PRs related to this improvements:
- https://github.com/robotology/icub-firmware/pull/472
- https://github.com/robotology/icub-main/pull/942

They can be moved in `ready for review`

valegagge, 2024-03-07:
Hi @MSECode ,
I think both PRs aere ok. 
Did you do some tests?
please add here their results.

Thanks

MSECode, 2024-03-07:
Here a screenshot of the test sone:

[image]

 Currently, I'm simulating on the ems side some generic errors, as you can see from the screenshot

MSECode, 2024-03-07:
One question, @valegagge, is it ok to hardcode the dimension of the array on the Diagnostic code  or do we have to parametrize it based on the struct defined in `Motor.hid`

valegagge, 2024-03-07:
Hi @MSECode ,
sorry, but I don't understand which dimension you refer to. :(

MSECode, 2024-03-07:
> Hi @MSECode , sorry, but I don't understand which dimension you refer to. :(

I'm referring to the array at this line: https://github.com/robotology/icub-main/blob/a3b9c8cc8cd0b70fee392f721681b386c7dd5410/src/libraries/icubmod/embObjLib/diagnosticInfoParsers.cpp#L538

I would suggest to move the `typedef union MotorFaultState` that contains the error bitmask correlated to our array to a header file that is seen by the sw part too, such as `EoMOtionControl.h` and use the embot constexpr `embot::core::binary::bitcountU32(motor_status)` to get the bit dimension of the motor status, which is in our case 32, so that the code is more robust and if it's going to change (even if I doubt) it is still valid and we do not need to modify code in multiple parts

What do u think?

valegagge, 2024-03-08:
You idea is good.
As we discussed F2F, you can move the bitmask definition in the shared files and update the fw and icub-manin consequentially. This solution is possible because the motor-status struct already contains the field (unit32_t) related to this error.

MSECode, 2024-03-11:
Code update with the discussed changes. All the changes are pushed and can be seen in the diff of the PR. Here the output of a fast test on the setup

[image]

pattacini, 2024-03-14:
PR merged:
- https://github.com/orgs/robotology/discussions/691

pattacini, 2024-03-14:
Closing this issue as completed.
Feel free to open it up again if required.

closed 2024-03-14 by pattacini